luxonis/depthai-core #525: Code shared when called from dlopen()ed shared library

nathanieltagg, 2022-07-15:
Hi, I'm trying to read an Oak-D with the depthai-core from within a class in a shared library that is part of my project.

I'm at my wit's end.

I'm trying this code fragment:
```
   dai::Pipeline pipeline;

    // Define sources and outputs
    auto monoLeft = pipeline.create<dai::node::MonoCamera>();
    auto monoRight = pipeline.create<dai::node::MonoCamera>();
    auto stereo = pipeline.create<dai::node::StereoDepth>();
    auto xoutDepth = pipeline.create<dai::node::XLinkOut>();

    xoutDepth->setStreamName("depth");

    // Properties
    monoLeft->setResolution(dai::MonoCameraProperties::SensorResolution::THE_720_P);
    monoLeft->setBoardSocket(dai::CameraBoardSocket::LEFT);
    monoRight->setResolution(dai::MonoCameraProperties::SensorResolution::THE_720_P);
    monoRight->setBoardSocket(dai::CameraBoardSocket::RIGHT);

    stereo->setDefaultProfilePreset(dai::node::StereoDepth::PresetMode::HIGH_DENSITY);
    stereo->setRectifyEdgeFillColor(0);  // black, to better see the cutout
    // stereo->setInputResolution(1280, 720);
    stereo->initialConfig.setMedianFilter(dai::MedianFilter::KERNEL_5x5);
    stereo->setLeftRightCheck(true);
    stereo->setExtendedDisparity(false);
    stereo->setSubpixel(false);

    // Linking
    monoLeft->out.link(stereo->left);
    monoRight->out.link(stereo->right);

    stereo->depth.link(xoutDepth->input);

    // Connect to device and start pipeline
    dai::Device device(pipeline);
```

If I put this code in the main exectuable, it runs correctly, and I can stream the stereo output.

If I put this code into my shared library, it fails at the device creation, with:
```
[184430108107940F00] [1.2] [71.912] [system] [critical] Error executing RPC call: [json.exception.type_error.304] cannot use at() with null
terminate called after throwing an instance of 'dai::XLinkReadError'
  what():  Couldn't read data from stream: '__rpc_main' (X_LINK_ERROR)
```

Turning on the debug, I see that the configuration being uploaded is incomplete, namely the nodes are null in this case:
```
{"connections":[{"node1Id":2,"node1Output":"depth","node1OutputGroup":"","node2Id":3,"node2Input":"in","node2InputGroup":""},{"node1Id":1,"node1Output":"out","node1OutputGroup":"","node2Id":2,"node2Input":"right","node2InputGroup":""},{"node1Id":0,"node1Output":"out","node1OutputGroup":"","node2Id":2,"node2Input":"left","node2InputGroup":""}],"globalProperties":{"calibData":null,"cameraTuningBlobSize":null,"cameraTuningBlobUri":"","leonCssFrequencyHz":700000000.0,"leonMssFrequencyHz":700000000.0,"pipelineName":null,"pipelineVersion":null,"xlinkChunkSize":-1},"nodes":[[0,null],[1,null],[2,null],[3,null]]}
```
This indicates there is some initialization being peformed in the example code that is not being performed when I run my own code.

Any suggestions for how to debug this?  It is clearly NOT the code, since I'm running the same exact lines, but something about the environment.

nathanieltagg, 2022-07-15:
I thought that this might be related to https://github.com/luxonis/depthai-core/issues/265, since I also use nlohmann in my code (specificaly v3.10

closed 2022-07-15 by nathanieltagg

nathanieltagg, 2022-07-15:
I see the problem. I have another bit of code that initializes before main runs, and which uses nlhomann::json.   If that class is included in the build, the sensor fails to initialize.  If it is not, it initializes.

It appears that the only stable thing is to not use nhlomann::json.

Luxonis-Brandon, 2022-07-16:
Sorry about the trouble.  Asking internally for advice and we'll circle back likely on Monday.

themarpe, 2022-07-18:
@nathanieltagg thanks for submitting the issue on this topic.

Do you mind creating a PoC project where this can easily be reproduced - It'll be easier for us to see the crux of the issue and fix it.

If you have a separate nlohmann library, one thing to try out is to remove the nlohmann from being supplied by hunter (hunter_add_package) and use your nlohmann library instead. For both build of depthai-core as well as your program.
Still, if you are using same nlohmann version as core is then you shouldn't have issues