luxonis/depthai-core #531: depthai builds failing for armv61 (raspberry pi zero). Last successful build was 2.13

bgro82, 2022-07-21:
Raspberry pi zero builds of depthai (armv61) last successfully built with 2.13 (about 7-8 months ago). Will it not be supported for future builds or is there some other reason for this?

Luxonis-Brandon, 2022-07-21:
Great question.  Will have to check with the team.  I'm not immediately sure what we use as the runner for the Pi Zero.  Will circle back.

bgro82, 2022-07-27:
@themarpe Thank you! I just checked for the new release (https://www.piwheels.org/project/depthai/) and looks like it still didn't build.

themarpe, 2022-07-28:
@bgro82 sorry for the issue - looks like we'll have to add our own runners to cover this case...

bgro82, 2022-08-22:
@themarpe thanks, do you know when this will be fixed?

themarpe, 2022-08-28:
@bgro82 unfortunatelly piwheels builds still have issues with latest release.

We'll likely have to run this on our own runners - currently we only build for ARMv7. Preferably we'd build for ARMv6 on a stronger RPi4B, but we've yet to implement this.

Any pointers are appreciated!

bgro82, 2023-01-28:
@themarpe any ideas on what may be still causing the builds to fail? I primarily bought these cameras to use with the rpi zero but can't use a recent version of depthai (which causes other dependency issues, etc)

bgro82, 2023-01-28:
after some more reading, is it possible that 32 bit support was removed or disabled in some way?

also after more reading, it appears the architecture is armv6l (not armv61).

themarpe, 2023-02-01:
@bgro82 

Yeah, the armv6 builds aren't available at the moment.
The issue is also that building natively on the Pi Zero isn't easy due to limited amounts of RAM.

I'll see about spinning up a crossbuild this weekend.

bgro82, 2023-02-08:
great thank you. Is it possible to use more disk swapping to make up for the low RAM (understanding it would be much slower)?

themarpe, 2023-02-10:
@bgro82 yes it would be possible.
In my limited testing i did, i haven't been able to setup a proper cross compilation setup for it, but if you opt for swap, you should be able to compile it as intended

bgro82, 2023-02-20:
I increased my swap file to 1024M (default on the pi zero is 100M) and was able to build this successfully when including "--no-dependencies"

themarpe, 2023-02-21:
> If there is somewhere I can share the wheel file let me know and happy to post it to save others the build time.

That'd actually be great - did you compile 2.20.2? Feel free to share here / link to it.

bgro82, 2023-02-21:
sure thing. Here is the latest depthai and opencv built for armv6l for raspbian bullseye and python 3.9:

depthai-2.20.2.0-cp39-cp39-linux_armv6l.whl: https://drive.google.com/file/d/1HWVVFAy7eKKuxEpd1KGq2tl71bJR4kKQ/view?usp=sharing
opencv_python-4.7.0.68-cp39-cp39-linux_armv6l.whl: https://drive.google.com/file/d/1SONV5cnrZV05fSqLexs-MDXxZyJch0x_/view?usp=sharing

themarpe, 2023-02-21:
Thanks @bgro82  - checked and available over at: https://artifacts.luxonis.com/artifactory/luxonis-python-release-local/

The below command should work as expected
```
python3 -m pip install --extra-index-url https://artifacts.luxonis.com/artifactory/luxonis-python-release-local/ depthai
```

bgro82, 2023-02-24:
Excellent, thank you. Happy to build these and provide from time to time as well if they are useful to others.

maxsitt, 2023-06-18:
Hi @bgro82 @themarpe,

it would be great if you could build and provide armv6l wheels for the latest depthai release ([v2.22.0.0](https://github.com/luxonis/depthai-python/releases/tag/v2.22.0.0)). This would enable users of Raspberry Pi Zero W (v1) to use the latest features.

Thank you!

themarpe, 2023-06-20:
@maxsitt 

Do you mind doing the following:
 - Download https://artifacts.luxonis.com/artifactory/luxonis-python-release-local/depthai/depthai-2.22.0.0-cp39-cp39-linux_armv7l.whl (bullseye) or https://artifacts.luxonis.com/artifactory/luxonis-python-release-local/depthai/depthai-2.22.0.0-cp37-cp37m-linux_armv7l.whl (buster)
 - rename from `linux_armv7l` to `linux_armv6l`
 - `python3 -m pip install [wheel]`

If that works, I'll add this as part of our releases (cross checked, our runners build with march=armv6 already, so I think this should be ifne)

themarpe, 2023-06-20:
> After you added the (renamed) armv6l wheel it will be possible to install the depthai package on RPi Zero W with a simple python3 -m pip install depthai? Or do we still have to add --extra-index-url https://artifacts.luxonis.com/artifactory/luxonis-python-release-local/ ?

According to some pypi discussion, it doesn't allow uploading both armv6l and armv7l wheels, but I'll give that a go. 
Otherwise, it'll be available under our releases. We'll update the docs accordingly.

And thanks for giving this a go!

maxsitt, 2023-06-21:
Thanks a lot @themarpe !

When running `python3 -m pip install depthai==2.22.0.0` on RPi Zero W, `depthai-2.22.0.0-cp39-cp39-linux_armv7l.linux_armv6l.whl` is downloaded and successfully installed. Everything I tested works perfectly (custom scripts with YOLOv5n model detection & tracking).

themarpe, 2023-07-27:
Closing as addressed now:)

closed 2023-07-27 by themarpe