luxonis/depthai-core #532: Build error (no <filesystem>) on Ubuntu 18

nathanieltagg, 2022-07-22:
I'm trying to use this code from an ubuntu 18 embedded ARM64 system, with the default (gcc-7) compilers.  The DepthAI code compiled just fine, but when I attempt to call the code from my own file:
```
In file included from /home/autoguide/opt/include/depthai/device/CalibrationHandler.hpp:9:0,
                 from /home/autoguide/opt/include/depthai/depthai.hpp:9,
                 from /home/autoguide/perception-sim/oakd/OakDModule.cpp:1:
/home/autoguide/opt/include/depthai/utility/Path.hpp:3:14: fatal error: filesystem: No such file or directory
```
I cannot figure out how to avoid this message.  I am using Cmake, and doing `find_package(depthai CONFIG)` so I should be pulling the correct configuration.  

The problem goes away if I upgrade to gcc-9, but that interferes with other parts of my build (namely CUDA, which cannot abide a different compiler).  

I'm guessing there's a missing define or something that causes my build to choke on the missing file where the hunter build does not, but looking at the code I don't see how; it seems to switch only on compiler version, which should have been the same.  I can manually edit the Path.hpp code, but that seems unsatisfactory.

Luxonis-Brandon, 2022-07-23:
Sorry about the trouble.  I'm not familiar enough with the build system to provide meaningful advice.  So I'm asking the team for help on this one. Thanks for using OAK!

themarpe, 2022-07-23:
Hi @nathanieltagg 
Thanks for the report. Seems like GCC-7, while supporting C++17 standard, does NOT yet implement the `filesystem` fully. Its still under `experimental/filesystem`.

I'll recheck if there is a means of adding a conditional there for C++17 & filesystem support

themarpe, 2022-07-23:
Addressed in #533

closed 2022-07-30 by themarpe