luxonis/depthai-core #538: [POE Camera] Losing connection randomly and not being able to capture the max resolution image by using Still

miknai, 2022-08-01:
I have 3 POE cameras (2 OAK-D-POE and 1 OAK-D-POE-PRO). I have been facing two issues so far. First, the connection between my PC and a POE camera gets suddenly broken off. Here is the error message shown in terminal. 
```
[2022-08-01 15:42:05.617] [warning] Monitor thread (device: 18443010D1F77B0E00 [10.0.0.64]) - ping was missed, closing the device connection
Traceback (most recent call last):
  File "test-full-resolution.py", line 53, in <module>
    inRgb = qRgb.tryGet()  # Non-blocking call, will return a new data that has arrived or None otherwise
RuntimeError: Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'rgb' (X_LINK_ERROR)'
```
I use the latest versions of Depthai (2.17.2.0) and firmware (0.0.20)
[image]

The second issue is that I am not able to get the maximum resolution image by using Still operator. In this example, https://docs.luxonis.com/projects/api/en/latest/samples/VideoEncoder/rgb_full_resolution_saver/ , I can quit the program by sending 'q' keyboard input, but not able to capture a still image with 'c' input. I modified the code like below to see whether I am sending the 'c' input.
```
key = cv2.waitKey(1)
if key == ord('q'):
    break
elif key == ord('c'):
    ctrl = dai.CameraControl()
    ctrl.setCaptureStill(True)
    qControl.send(ctrl)
    print("Sent 'still' event to the camera!")
```
I get the "Sent 'still' event to the camera!" output in terminal whenever I press 'c' in keyboard, but nothing happens with the camera. 

These two issues happen in all my three POE camera. I had a chance to talk about it with Erik in Discord channel and heard from him that this might be FW issue, so I report the issues here. Feel free to let me know if you need more information. Thank you!

miknai, 2022-08-01:
This also happens when I run the multiple device example. (https://github.com/luxonis/depthai-experiments/blob/master/gen2-multiple-devices/main.py) Below is the outputs in terminal.
```
=== Connected to 184430106192F30F00
   >>> MXID: 184430106192F30F00
   >>> Cameras: RGB LEFT RIGHT
   >>> USB speed: UNKNOWN
   >>> Loading pipeline for: OAK-D-POE
=== Connected to 18443010D15EE00F00
   >>> MXID: 18443010D15EE00F00
   >>> Cameras: RGB LEFT RIGHT
   >>> USB speed: UNKNOWN
   >>> Loading pipeline for: OAK-D-POE
[2022-08-01 16:00:45.800] [warning] Monitor thread (device: 18443010D15EE00F00 [10.0.0.135]) - ping was missed, closing the device connection
=== Connected to 18443010D1F77B0E00
   >>> MXID: 18443010D1F77B0E00
   >>> Cameras: RGB LEFT RIGHT
   >>> USB speed: UNKNOWN
   >>> Loading pipeline for: OAK-D-POE
Traceback (most recent call last):
  File "test-multiple-device.py", line 73, in <module>
    in_rgb = q_rgb.tryGet()
RuntimeError: Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'rgb' (X_LINK_ERROR)'
```

Luxonis-Brandon, 2022-08-01:
Sorry about the trouble here.  Asking.

themarpe, 2022-08-02:
Hi @miknai 
Do you mind testing version `2.17.1.0`? There are some temporary stability improvements which we are working on bringing back (caused some other side effects). 

Also after how long did the initial example crash?

miknai, 2022-08-02:
@themarpe Sure. What command do I need to use to go back to previous version? For your second question. It crashes from the very beginning (like 1~5 seconds) after the program starts streaming video.

Erol444, 2022-08-06:
@miknai You can install specific depthai version with eg. `python3 -mpip install depthai==2.17.1.0`.
Thanks, Erik

miknai, 2022-08-16:
So, this issue actually contains two issues. First one is that I lose connection between host pc and cameras. This issue still persists. I will make a separate Github issue for that. The second one is that I am not able to capture a still image. I was able to find a work-around way to do that by using Script node.

closed 2022-08-16 by miknai

miknai, 2022-08-16:
opened another ticket for the disconnection issue

https://github.com/luxonis/depthai-experiments/issues/389