luxonis/depthai-core #540: OAK-POE not accessile

wajeeha77, 2022-08-03:


miknai, 2022-08-03:
Could you elaborate more? I don't work for Luxonis, but I have been playing with their POE cameras for a couple of weeks. I can share what I know.

wajeeha77, 2022-08-03:
yeah please, can't access the oak despite the network being sound, followed all the trouble shooting steps

miknai, 2022-08-03:
I believe you have ran code like below. No output? 
```
import depthai

device_infos = depthai.Device.getAllAvailableDevices()
print(device_infos)
```

Is your POE injector/ switch providing the enough power to the camera?

wajeeha77, 2022-08-04:
> I believe you have ran code like below. No output?
> 
> ```
> import depthai
> 
> device_infos = depthai.Device.getAllAvailableDevices()
> print(device_infos)
> ```
> 
> Is your POE injector/ switch providing enough power to the camera?

I think so, I am using CISCO MERAKI Z3 switch and I believe it supplies enough power to each of it ports
TO make sure that enough power is being supplied I also had other devices disconnected from the switch and then checked, but that didn't work either

wajeeha77, 2022-08-04:
@alex-luxonis please help

themarpe, 2022-08-05:
@wajeeha77 
First I'd check what IP the device is given, just so you know that the device is up and running. You may check that in the DHCP table on your router. You may also set a static IP in range: 169.254.0.0/16 on your PC and access the camera in this way.

Next up, check your firewall settings on your PC. Make sure in/out bound for 11490 (TCP) & 11491 (UDP) are allowed.

If you cannot ping the device and you know the IP belongs to the camera and it looks like its online, then we'll have to debug further

wajeeha77, 2022-08-05:
> @wajeeha77 First I'd check what IP the device is given, just so you know that the device is up and running. You may check that in the DHCP table on your router. You may also set a static IP in [...]

I have gone through all the steps given on the troubleshooting page, set the IP static, and checked the firewall setting they are clear as well. I installed an IP scanner to check if a new IP appears when connecting the camera with the switch, surprisingly a new IP does appear but its not pingable and also not sure its the one or not, (I am attaching the screenshot of the IP scanner ip list)), to the best of my guess, the oak's PoE IP is the seventh one in the list, -> 192.168.0.17, I believe,
Anyone who is successfully using oak PoE can confirm this by scanning through IP scanner and see if the manufacturer's name comes out to be blank or 'Luxonis', this will confirm if the Oak IP was found or not..
[image]

themarpe, 2022-08-07:
@wajeeha77 
Hmm, the seventh entry (192.168.0.17) has incorrect MAC address for a Luxonis PoE device (of up to this moment assigned MAC addresses). The current base is: `44-A9-2C-3x:xx:xx`

Do you mind performing "Factory Reset", reinstalling the bootloader, etc... and recheck it comes online then?

wajeeha77, 2022-08-08:
> @wajeeha77 Hmm, the seventh entry (192.168.0.17) has incorrect MAC address for a Luxonis PoE device (of up to this moment assigned MAC addresses). The current base is: `44-A9-2C-3x:xx:xx`
> 
> Do you mind performing "Factory Reset", reinstalling the bootloader, etc... and recheck it comes online then?

ok I will try and get back to you

closed 2022-08-31 by wajeeha77