luxonis/depthai-core #541: [Feature Request] Marking whether ImgFrames are Keyframes from H264/H265 Video Encoders

gabrielthomsen, 2022-08-03:
I want to be able to check a field in ImgFrame that came from a video encoder to see whether or not it is a keyframe. For h264/h265, that is dependent on whether it is actually a keyframe or not. For MJPEG, every frame is technically a keyframe, so it could always be true in that case. I just have a use-case where I need to be able to tell whether or not I have a keyframe dynamically.

gabrielthomsen, 2022-08-03:
My workaround for now is checking if sequence number % keyframe_rate == 0 is true. This seems to work pretty well, but not ideal for my use-case.

Luxonis-Brandon, 2022-08-04:
Thanks.  I'm not sure whether this is easy to support or not.  Will ask.