luxonis/depthai-core #542: zero_copy branch example code

0y8w1x, 2022-08-03:
I just wanted to ask if there is an example code snippet, on how to use this branch, because I heard I need to use special functions. I got told by @Luxonis-Erik that opening an issue might give me an answer.

Luxonis-Brandon, 2022-08-04:
Thanks.  We'll circle back.  I don't recall the context of this.  Will ask.

moratom, 2022-08-07:
Hi @0y8w1x,
if you checkout the [branch](https://github.com/luxonis/depthai-core/tree/message_zero_copy), all output queues (the return value of the `device.getOutputQueue(...)`) will automatically have the zero-copy functionality.

I'm not sure how tested it is and if there are any known issues with the implementation, so CC: @themarpe  regarding this.

themarpe, 2022-08-07:
@0y8w1x 
Main thing to note there is that the data isn't available through regular `getData` means but through `packet->data` instead. Also, `getFrame` & `getCvFrame` and other message helpers aren't available. 

This branch is experimental / feature testing branch, and it might or might not be upstreamed one day - it also isn't inteded for regular consumption, but for those working at the very edge of compute / latency / overhead boundaries.

0y8w1x, 2022-08-08:
thanks @themarpe and @moratom for clarifying. I guess I need to go through the commits, to create a working example then 👍

closed 2023-02-20 by 0y8w1x