luxonis/depthai-core #544: Add message syncing node to pipeline

wouterio, 2022-08-04:
## Start with the why:

As a developer I may create a depthai pipeline with multiple output streams/queues. But then it's important that messages from those streams are synced. At this point, the pipeline has no standard node/element that guarantees synced messages. Such a node would be helpful.

## Move to the what:

Provide a syncing node receiving messages from multiple streams, and emitting sync messages combining/containing those messages from the streams with the same sequenceNum.

## Move to the how:

#539 adds `.getTimeStamp()` and `.getSequenceNum()` to all depthai messages. But I have no further insight in you're code, so I'll leave this part to you... ;)

wouterio, 2022-08-04:
I'd envision it to work like this in the pipeline:
```
pipeline = dai.Pipeline()

color = pipeline.create(dai.node.ColorCamera)
left = pipeline.create(dai.node.MonoCamera)
right = pipeline.create(dai.node.MonoCamera)
stereo = pipeline.create(dai.node.StereoDepth)
features = pipeline.create(dai.node.FeatureTracker)
sync = pipeline.create(dai.node.MessageSyncer)

depth_out = pipeline.create(dai.node.XLinkOut)
color_out = pipeline.create(dai.node.XLinkOut)
features_out = pipeline.create(dai.node.XLinkOut)
sync_out = pipeline.create(dai.node.XLinkOut)

color.isp.link(color_out.input)
left.out.link(stereo.left)
right.out.link(stereo.right)
stereo.depth.link(depth_out.input)
right.out.link(features.inputImage)
features.outputFeatures.link(features_out.input)

color_out.output.link(sync.input)
depth_out.output.link(sync.input)
features_out.output.link(sync.input)

sync_out.setStreamName(SYNC_STREAM_NAME)
```

wouterio, 2022-08-04:
Or perhaps then we can even omit some steps in the pipeline:

```
pipeline = dai.Pipeline()

color = pipeline.create(dai.node.ColorCamera)
left = pipeline.create(dai.node.MonoCamera)
right = pipeline.create(dai.node.MonoCamera)
stereo = pipeline.create(dai.node.StereoDepth)
features = pipeline.create(dai.node.FeatureTracker)
sync = pipeline.create(dai.node.MessageSyncer)

sync_out = pipeline.create(dai.node.XLinkOut)

color.isp.link(sync_out.input)
left.out.link(stereo.left)
right.out.link(stereo.right)
stereo.depth.link(sync_out.input)
right.out.link(features.inputImage)
features.outputFeatures.link(sync_out.input)

sync_out.setStreamName(SYNC_STREAM_NAME)
```

themarpe, 2022-08-05:
Thanks for the great writeup.

Something along those lines is on our TODO list, but we've been quite busy with underlying FW and backend improvements lately, so didn't yet come to improving DepthAI library in off itself.

I hope to further push on this front in the following months.

Also, we are bringing in `host_pipeline` which might expose potential improvements being done directly on the host side as well, for some cases.