luxonis/depthai-core #546: [184430109158DF0F00] [192.168.1.200] [557.932] [system] [critical] Fatal error. Please report to developers. Log: 'PlgSrcCtrl' '87'

BBwanaz, 2022-08-04:
I am running OAKD-POE inside a docker container. The app has been running smoothly for weeks but recently it started giving this error here:

```
camera       | [184430109158DF0F00] [192.168.1.200] [557.932] [system] [critical] Fatal error. Please report to developers. Log: 'PlgSrcCtrl' '87'
camera       | [2022-08-04 20:33:48.236] [warning] Monitor thread (device: 184430109158DF0F00 [192.168.1.200]) - ping was missed, closing the device connection
camera       | Traceback (most recent call last):
camera       |   File "/computer-vision/training/record.py", line 70, in <module>
camera       |     run_recorder()
camera       |   File "/computer-vision/training/record.py", line 54, in run_recorder
camera       |     inRgb = qRgb.get()
camera       | RuntimeError: Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'rgb' (X_LINK_ERROR)'
```

Furthermore, specifying the IP of the camera using: device_info = dai.DeviceInfo(ip) is returning "no device with such info" however, when I run it to auto find the device on the network, the finds the device and when I print the device info, it shows the same IP address.

themarpe, 2022-08-05:
@BBwanaz 
Do you mind checking the version of base image you are using? 
It seems like an older version is being used in this case. Try using either: v2.17.1 or v2.17.2

BBwanaz, 2022-08-05:
Hi @themarpe 

both versions and I still get the same error.

Luxonis-Brandon, 2022-08-09:
Sorry about the trouble @BBwanaz .  We'll likely need to dig into firmware to see how/why this is happening and what it will take to fix this.

themarpe, 2022-09-19:
@BBwanaz 

The latest v2.17.3 should address the issues when connecting with specific IP. 

>  The app has been running smoothly for weeks but recently it started giving this error here
One question on the issue -