luxonis/depthai-core #551: Code for depth estimation

seong-me, 2022-08-08:
I want to know the detail process about depth estimation in OAK-D device
So I'm really looking for the code that is related with depth estimation in your repository
But only what I found is like "device.getOutputQueue("depth", 4, false);"
In your repository, is there no code for depth estimation?

Luxonis-Brandon, 2022-08-08:
See here for docs on the stereo depth node:

https://docs.luxonis.com/projects/api/en/latest/components/nodes/stereo_depth/

The stereo depth is calculated on a dedicated piece of silicon - so there is not code, other than configuration.

re-this, 2022-08-08:
Thank you, @Luxonis-Brandon

re-this, 2022-08-08:
Hi, @Luxonis-Brandon 
Sorry for reopening, but I have another question 
Did you use any NN in the process of estimating depth in OAK-D?
For example, such as, when finding corresponding points btw left and right images

Luxonis-Brandon, 2022-08-08:
We do support this, yes.  Here are some examples:
https://github.com/luxonis/depthai/issues/173#issuecomment-1134605006

We're working to get more going as well, and to make them faster.  There's a TON of research in it, and we're also planning (if all goes well) to host a competition with OpenCV to push the state of the art on this.  
[image]