luxonis/depthai-core #552: Error during build depthai-core cross-compile with the openWRT toolchain

zulfikaryh, 2022-08-09:
Hello, I'm trying to cross-compile depthai-core using the OpenWRT toolchain : mipsel_24kc_gcc-7.3.0_musl. but It fails with error :

`-- Found Backward: /home/apps/.hunter/_Base/cb0ea1f/0b78939/0fb33dc/Build/Backward/Source
-- Configuring done
-- Generating done
-- Build files have been written to: /home/apps/.hunter/_Base/cb0ea1f/0b78939/0fb33dc/Build/Backward/Build/Backward-Release-prefix/src/Backward-Release-build
[ 75%] Performing build step for 'Backward-Release'
Scanning dependencies of target backward_object
Scanning dependencies of target backward
[ 33%] Building CXX object CMakeFiles/backward_object.dir/backward.cpp.o
[ 66%] Building CXX object CMakeFiles/backward.dir/backward.cpp.o
In file included from /home/apps/.hunter/_Base/cb0ea1f/0b78939/0fb33dc/Build/Backward/Source/backward.cpp:36:0:
/home/apps/.hunter/_Base/cb0ea1f/0b78939/0fb33dc/Build/Backward/Source/backward.hpp:262:10: fatal error: execinfo.h: No such file or directory
 #include <execinfo.h>
          ^~~~~~~~~~~~
compilation terminated.
In file included from /home/apps/.hunter/_Base/cb0ea1f/0b78939/0fb33dc/Build/Backward/Source/backward.cpp:36:0:
/home/apps/.hunter/_Base/cb0ea1f/0b78939/0fb33dc/Build/Backward/Source/backward.hpp:262:10: fatal error: execinfo.h: No such file or directory
 #include <execinfo.h>
          ^~~~~~~~~~~~
compilation terminated.
CMakeFiles/backward_object.dir/build.make:62: recipe for target 'CMakeFiles/backward_object.dir/backward.cpp.o' failed
make[5]: *** [CMakeFiles/backward_object.dir/backward.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/backward_object.dir/all' failed
make[4]: *** [CMakeFiles/backward_object.dir/all] Error 2
make[4]: *** Waiting for unfinished jobs....
CMakeFiles/backward.dir/build.make:62: recipe for target 'CMakeFiles/backward.dir/backward.cpp.o' failed
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/backward.dir/all' failed
make[5]: *** [CMakeFiles/backward.dir/backward.cpp.o] Error 1
make[4]: *** [CMakeFiles/backward.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/Backward-Release.dir/build.make:114: recipe for target 'Backward-Release-prefix/src/Backward-Release-stamp/Backward-Release-build' failed
make[2]: *** [Backward-Release-prefix/src/Backward-Release-stamp/Backward-Release-build] Error 2
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/Backward-Release.dir/all' failed
make[1]: *** [CMakeFiles/Backward-Release.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
 
[hunter ** FATAL ERROR **] Build step failed (dir: /home/apps/.hunter/_Base/cb0ea1f/0b78939/0fb33dc/Build/Backward
[hunter ** FATAL ERROR **] [Directory:/home/apps/zulfikar/depthai-core/cmake]
 
------------------------------ ERROR -----------------------------
    https://hunter.readthedocs.io/en/latest/reference/errors/error.external.build.failed.html
------------------------------------------------------------------
 
CMake Error at /home/apps/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_error_page.cmake:12 (message):
Call Stack (most recent call first):
  /home/apps/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_fatal_error.cmake:20 (hunter_error_page)
  /home/apps/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_download.cmake:623 (hunter_fatal_error)
  /home/apps/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_add_package.cmake:53 (hunter_download)
  cmake/depthaiDependencies.cmake:21 (hunter_add_package)
  CMakeLists.txt:164 (include)
 
 
-- Configuring incomplete, errors occurred!
See also "/home/apps/zulfikar/depthai-core/build/CMakeFiles/CMakeOutput.log".`

I use cmake version 3.10.2, c version mipsel-openwrt-linux-gcc (OpenWrt GCC 7.3.0 r0-6459074) 7.3.0, and c++ version mipsel-openwrt-linux-g++ (OpenWrt GCC 7.3.0 r0-6459074) 7.3.0
here you can check the full log :
-  [https://pastebin.com/hgdQEwda](build log)
-  [https://pastebin.com/q1emijRh](cmake log)

Do you have any suggestions on what to do?

themarpe, 2022-08-09:
@zulfikaryh 

Disable Backward (stack printing library) using `-D DEPTHAI_ENABLE_BACKWARD=OFF`

themarpe, 2022-09-26:
@zulfikaryh 

Did that resolve your issue? Feel free to open it back up if that's not the case.

closed 2022-09-26 by themarpe