luxonis/depthai-core #554: Monitor Thread Issue with Oak d poe cameras

apollollopa, 2022-08-11:
I have tested multiple oak d poe cameras on different poe power sources and have had the same issue.
I want to run this script daily and on 4k resolution. I have tested the script with 1080p and it seems to work way longer but still in the end it fails with the same script.

When running the script with 4k option It tends to last anywhere from 10 seconds to 30 mins.
while running the script in 1080. I see times from 30s - a couple hours. it varies everytime.  

I have updated the depthai version as well as the bootloader. I have also tried factory resetting. I checked with debugging to see if the cpu was maxing out, And that was not the problem.

[image]
[image]
[image]
[image]
[image]

These are the errors I receive.

[2022-08-11 10:39:29.836] [warning] Monitor thread (device: 18443010814EF50F00 [192.168.70.57]) - ping was missed, closing the device connection
RuntimeError: Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'Image' (X_LINK_ERROR)'

Here is the code I was testing on. 

`import depthai as dai
import time
 
pipeline = dai.Pipeline()
 
camera = pipeline.create(dai.node.ColorCamera)
videoEncoder = pipeline.create(dai.node.VideoEncoder)
imageOutput = pipeline.create(dai.node.XLinkOut)
 
imageOutput.setStreamName("Image")
 
camera.setImageOrientation(dai.CameraImageOrientation.ROTATE_180_DEG)
camera.setBoardSocket(dai.CameraBoardSocket.RGB)
camera.setResolution(dai.ColorCameraProperties.SensorResolution.THE_4_K)
videoEncoder.setDefaultProfilePreset(camera.getFps(), dai.VideoEncoderProperties.Profile.MJPEG)
 
camera.video.link(videoEncoder.input)
videoEncoder.bitstream.link(imageOutput.input)
 
# 
info = dai.DeviceInfo('192.168.70.57')
 
with dai.Device(pipeline, info, maxUsbSpeed=dai.UsbSpeed.SUPER_PLUS) as device:
  imageQueue = device.getOutputQueue(name="Image", maxSize=1, blocking=True)
  device.setLogLevel(dai.LogLevel.DEBUG)
  device.setLogOutputLevel(dai.LogLevel.DEBUG)
 
  while True:
    imageFrame:dai.ImgFrame = imageQueue.get()
    print(imageFrame.getSequenceNum())
    time.sleep(0.001)`

themarpe, 2022-09-19:
@apollollopa 
Addressed in latest `develop` - make sure to update and retry.

Also note that enough bandwidth has to be available especially on bigger frame transfers, for the device to function properly. WiFi connectivity can usually be the culprit here as the speeds can be quite low. (Recheck that you are connecting over ETH instead of WiFi)

apollollopa, 2022-09-19:
I have ran the update. Now I am not getting any debug info from the camera. So not sure if the camera is working or if I need to call debug a different way since the update

themarpe, 2022-09-19:
Odd - you may try `DEPTHAI_LEVEL=debug` env var, but it might be a regression since #587 

Testing now locally

themarpe, 2022-09-19:
@apollollopa 
The behavior seems to match.

The `device.setLogLevel` & `device.setLogOutputLevel` only affect logs coming from the device. 

Only the env var `DEPTHAI_LEVEL` can set the global stdout logging level, at the time being.

closed 2022-09-19 by themarpe