luxonis/depthai-core #559: Sequence number Drifts

codesian, 2022-08-17:
Opening a depth disparity and a RGB 4K stream, makes that frame sequence numbers drift over time.
Lowering FPS setting seems to not affect in sync, higher frame rates increase drift.
Depth settings and RGB align seems to not affect, bahavior its same. 

[C++ example
](https://transfer.sh/FMppKG/fps.zip)

Discussion at [discord](https://discord.com/channels/790680891252932659/924798097660448789/1008791165253992519)

codesian, 2022-08-18:
Sequence numbers are a total disaster, as you can see, drift over time, we let the cameras start to show frame sequence, at 30fps all cameras we can see, source of image and timestamp, for example:

SEQ | SOURCE | TIMESTAMP |  
-- | -- | -- | --
14800 | COLOR | 495108749637 |  
14747 | DISPARITY | 494817001587 |  
14802 | COLOR | 495175183360 |
14749 | DISPARITY | 494883632532 |  
14804 | COLOR | 495241616975 |  
14752 | DISPARITY | 494983573518 |  
14806 | COLOR | 495308050837 |  
14754 | DISPARITY | 495050203940 |  
14809 | COLOR | 495407702637 |  
14756 | DISPARITY | 495116839211 | (COLOR: 14800 - DISP:14756)  (495108749637 - 495116839211)/1M =  8,089574 ms [Sequence difference 44 frames] 
14811 | COLOR | 495474135457 | 
14759 | DISPARITY | 495216777394 | (COLOR: 14804 - DISP:14759)  (495241616975 - 495216777394)/1M =  -24,839581 ms [Sequence difference 45 frames] 
14814 | COLOR | 495573786738 | 
14761 | DISPARITY | 495283407922 | (COLOR: 14806 - DISP:14761)  (495308050837 - 495283407922)/1M =  -24,642915 ms [Sequence difference 45 frames] 

Well now, the question its.. If the images are RGB align, its based on frame sequences or its based on timestamp, what can i find? a 45 frames missaligned images. Don know what happend.

themarpe, 2022-09-25:
CC: @szabi-luxonis 

@codesian seq numbers aren't actively aligned, apart from initial sync, as far as I recall - so any frame drop could throw those off. 
I'd suggest relying on getDeviceTimestamp for syncing the frames.

szabi-luxonis, 2022-09-25:
They indeed aren't, not sure why.
Cc: @alex-luxonis since it's RGB depth alignment related.

jasonm1892, 2022-11-11:
@alex-luxonis was responsible for RGB alignment. Why are you ignoring this one too? The RGB alignment you wrote was crashing (similar to Stereo) until someone else fixed it a year later. History is on github.