luxonis/depthai-core #562: Couldn't read data from steam "__log"&"__timesync" on Raspberry Pi 400

Agent-Birkhoff, 2022-08-22:
[image]

The pipeline will run for about 1-5minutes, then it will pop this.

**Platform**: 32bit Raspberry Pi OS lite on a RPI400, booting from TF card

The power supply is sufficient, for both cam and RPI400

**It seems to work well on a x86 PC.**

**MRE:**
```python
import depthai as dai

with dai.Device(version=dai.OpenVINO.Version.VERSION_2021_4, usb2Mode=True) as device:
    device.setLogLevel(dai.LogLevel.DEBUG)
    device.setLogOutputLevel(dai.LogLevel.DEBUG)

    # Loading blob
    blob = dai.OpenVINO.Blob("./640_360_(10_10).blob")
    INPUT_SHAPE = blob.networkInputs["rgb"].dims[:2]  # Auto INPUT_SHAPE

    # Start defining a pipeline
    pipeline = dai.Pipeline()

    # RGB camera
    cam = pipeline.create(dai.node.ColorCamera)
    cam.setResolution(dai.ColorCameraProperties.SensorResolution.THE_1080_P)
    cam.setIspScale((1, 3), (1, 3))  # RGB->640x360
    cam.setBoardSocket(dai.CameraBoardSocket.RGB)
    cam.setPreviewSize(*INPUT_SHAPE)

    # Neural network
    segmentation_nn = pipeline.create(dai.node.NeuralNetwork)
    segmentation_nn.setBlob(blob)
    segmentation_nn.input.setBlocking(False)
    segmentation_nn.setNumInferenceThreads(0)  # 0 for auto
    cam.preview.link(segmentation_nn.inputs["rgb"])

    # Left mono camera
    left = pipeline.create(dai.node.MonoCamera)
    left.setResolution(dai.MonoCameraProperties.SensorResolution.THE_400_P)
    left.setBoardSocket(dai.CameraBoardSocket.LEFT)
    # Right mono camera
    right = pipeline.create(dai.node.MonoCamera)
    right.setResolution(dai.MonoCameraProperties.SensorResolution.THE_400_P)
    right.setBoardSocket(dai.CameraBoardSocket.RIGHT)

    # Create a node that will produce the depth map
    stereo = pipeline.create(dai.node.StereoDepth)
    stereo.setNumFramesPool(25)  # TODO may be related to the FPS
    left.out.link(stereo.left)
    right.out.link(stereo.right)

    # Depth output linked to NN
    stereo.depth.link(segmentation_nn.inputs["depth"])

    # NN output linked to XLinkOut
    xout_nn = pipeline.create(dai.node.XLinkOut)
    xout_nn.setStreamName("nn")
    segmentation_nn.out.link(xout_nn.input)

    # Start pipeline
    device.startPipeline(pipeline)
    q_nn = device.getOutputQueue(name="nn", maxSize=4, blocking=False)  # type: ignore

    while True:
        msgs = q_nn.get()
        print("Messages received")
```

The blob used:
[blobs.zip](https://github.com/luxonis/depthai-core/files/9395500/blobs.zip)

themarpe, 2022-08-23:
@Agent-Birkhoff 
We've tried reproducing the above issue but can't seem to get to the same error.
Do you mind sharing which library version you've used? And do you mind either:
 - Using latest available DepthAI library `python3 -m pip install depthai -U`
 - Compiling latest DepthAI locally: `git checkout main; git submodule update --init --recursive; python3 -m pip install .`

Let me know if any of these fix the issue for you

Agent-Birkhoff, 2022-08-23:
OK, I'll have a try tomorrow.

BTW I tried the official pre-configured Image (POE-CM4-V8) and have the same bug...

Agent-Birkhoff, 2022-08-24:
Same bug...

On the official pre-configured Image (POE-CM4-V8), depthai version 2.17.3.1.dev0+ee4d2fd5718e4efa045def8fde9fb836df969f82

Agent-Birkhoff, 2022-08-31:
New MRE: (This version of blob below has only one input)
`import depthai as dai

with dai.Device(version=dai.OpenVINO.Version.VERSION_2021_4, usb2Mode=True) as device:
    device.setLogLevel(dai.LogLevel.DEBUG)
    device.setLogOutputLevel(dai.LogLevel.DEBUG)

    # Loading blob
    blob = dai.OpenVINO.Blob("../../RGB.blob")
    INPUT_SHAPE = blob.networkInputs["rgb"].dims[:2]  # Auto INPUT_SHAPE

    # Start defining a pipeline
    pipeline = dai.Pipeline()

    # RGB camera
    cam = pipeline.create(dai.node.ColorCamera)
    cam.setResolution(dai.ColorCameraProperties.SensorResolution.THE_1080_P)
    cam.setIspScale((1, 3), (1, 3))  # RGB->640x360
    cam.setBoardSocket(dai.CameraBoardSocket.RGB)
    cam.setPreviewSize(*INPUT_SHAPE)

    # Neural network
    segmentation_nn = pipeline.create(dai.node.NeuralNetwork)
    segmentation_nn.setBlob(blob)
    segmentation_nn.input.setBlocking(False)
    segmentation_nn.setNumInferenceThreads(0)  # 0 for auto
    cam.preview.link(segmentation_nn.inputs["rgb"])

    # Left mono camera
    left = pipeline.create(dai.node.MonoCamera)
    left.setResolution(dai.MonoCameraProperties.SensorResolution.THE_400_P)
    left.setBoardSocket(dai.CameraBoardSocket.LEFT)
    # Right mono camera
    right = pipeline.create(dai.node.MonoCamera)
    right.setResolution(dai.MonoCameraProperties.SensorResolution.THE_400_P)
    right.setBoardSocket(dai.CameraBoardSocket.RIGHT)

    # Create a node that will produce the depth map
    stereo = pipeline.create(dai.node.StereoDepth)
    stereo.setNumFramesPool(25)  # TODO may be related to the FPS
    left.out.link(stereo.left)
    right.out.link(stereo.right)

    # Depth output linked to XLinkOut
    xout_depth = pipeline.create(dai.node.XLinkOut)
    xout_depth.setStreamName("depth")
    stereo.depth.link(xout_depth.input)

    # NN output linked to XLinkOut
    xout_nn = pipeline.create(dai.node.XLinkOut)
    xout_nn.setStreamName("nn")
    segmentation_nn.out.link(xout_nn.input)

    # Start pipeline
    device.startPipeline(pipeline)
    q_nn = device.getOutputQueue(name="nn", maxSize=4, blocking=False)  # type: ignore
    q_depth = device.getOutputQueue(name="depth", maxSize=4, blocking=False)  # type: ignore

    while True:
        msgs = q_nn.get()
        print("NN!!!")

        depth = q_depth.get()
        print("DEPTH!!!")
`
[RGB.zip](https://github.com/luxonis/depthai-core/files/9459734/RGB.zip)

Same error on RPI and still works fine on PC

@themarpe

Agent-Birkhoff, 2022-08-31:
I tested the model with one input because I tried this: [gen2-deeplabv3_person demo](https://github.com/luxonis/depthai-experiments/tree/master/gen2-deeplabv3_person)

It is also a segmentation model but works fine on RPI for over 30 minutes... So I tried a new model with only one input, just like the demo, but the bug still happened.

themarpe, 2022-08-31:
Thanks, testing it out on our end now.

themarpe, 2022-09-05:
@Agent-Birkhoff

I've been running the above MRE on OAK-CM4 for ~5days (112h) and its still running.

We'll explore more in hopes we catch this issue.

Chenghao-Tan, 2022-09-06:
I have finally fixed the problem. It turns out that my power supply is not sufficient.

I plugged in the DC jack and USB at the same time, to a 5V adapter and the RPI400 individually, but most of the current was still coming from the USB port. Recently I did some measurements and found that DC jack only provided about 10mA, although its voltage was exactly 5V. 

My camera is OAK-D-IOT-75.

Chenghao-Tan, 2022-09-06:
The [gen2-deeplabv3_person demo](https://github.com/luxonis/depthai-experiments/tree/master/gen2-deeplabv3_person) demo may consume less power, so it worked fine while mine didn't.

closed 2022-09-06 by Chenghao-Tan

themarpe, 2022-09-06:
Thanks for circling back on this!