luxonis/depthai-core #564: ImageManip with input from XLinkIn causes Pipeline to freeze

gabrielthomsen, 2022-08-25:
With the below pipeline, I am able to get one image manipulated and outputted from a Movidius board before there is no more output from the board. This seems to only be an issue with the ImageManip node receiving input from an XLinkIn node. In other configurations where the ImageManip input is from a ColorCamera, I've not experienced any freezing. 

```
import depthai as dai
import cv2

def to_planar(arr: np.ndarray, shape: tuple) -> np.ndarray:
    return cv2.resize(arr, shape).transpose(2, 0, 1).flatten()

pipeline = dai.Pipeline()
in = pipeline.create(dai.node.XLinkIn)
in.setMaxDataSize(1280 * 800 * 3)
in.setNumFrames(1)
in.setStreamName('in')

manip_yolo = pipeline.create(dai.node.ImageManip)
manip_yolo.initialConfig.setKeepAspectRatio(True)
manip_yolo.initialConfig.setResize(416, 416)
manip_yolo.initialConfig.setFrameType(dai.ImgFrame.Type.BGR888p)
manip_yolo.inputImage.setBlocking(False)
manip_yolo.inputImage.setQueueSize(1)

right_in.out.link(manip_yolo.inputImage)

out = pipeline.create(dai.node.XLinkOut)
out.setStreamName('out')
out.input.setBlocking(False)
out.input.setQueueSize(1)
manip_yolo.out.link(out.input)

device = dai.Device(pipeline)

# get some image somehow
# I personally am ripping from recorded video

in_q = device.getInputQueue('in', blocking=False, maxSize=1)
out_q = device.getOutputQueue('out', blocking=False, maxSize=1)

msg = dai.ImgFrame()
msg.setType(dai.ImgFrame.Type.BGR888p)
msg.setData(to_planar(cv2.resize(image, (1280, 800)), (1280, 800)))
msg.setTimestamp(1)
msg.setWidth(1280)
msg.setHeight(800)

in_q.send(msg)
img = out_q.get()
```

themarpe, 2022-08-25:
Thanks for the report @gabrielthomsen - we'll look into it

moratom, 2022-08-25:
Hi @gabrielthomsen, the issue is resolved if you comment out the line `in.setNumFrames(1)`, or set more the parameter to something more than `1` here. 

Here is a working example:
```
import depthai as dai
import numpy as np
import cv2

IMG_WIDTH = 1280
IMG_HEIGHT = 800

IMG_PATH = "/home/matevz/Pictures/example.jpeg" # Replace this with a local image

def to_planar(arr: np.ndarray, shape: tuple) -> np.ndarray:
    return cv2.resize(arr, shape).transpose(2, 0, 1).flatten()

pipeline = dai.Pipeline()
in1 = pipeline.create(dai.node.XLinkIn)
in1.setMaxDataSize(IMG_WIDTH * IMG_HEIGHT * 3)
# in1.setNumFrames(2) // Leave the default.
in1.setStreamName('in')

manip_yolo = pipeline.create(dai.node.ImageManip)
manip_yolo.initialConfig.setKeepAspectRatio(True)
manip_yolo.initialConfig.setResize(150, 150)
manip_yolo.initialConfig.setFrameType(dai.ImgFrame.Type.BGR888p)
manip_yolo.inputImage.setBlocking(False)
manip_yolo.inputImage.setQueueSize(1)

in1.out.link(manip_yolo.inputImage)

out = pipeline.create(dai.node.XLinkOut)
out.setStreamName('out')
out.input.setBlocking(False)
out.input.setQueueSize(1)
manip_yolo.out.link(out.input)

device = dai.Device(pipeline)

in_q = device.getInputQueue('in', blocking=False, maxSize=1)
out_q = device.getOutputQueue('out', blocking=False, maxSize=1)

image = cv2.imread(IMG_PATH)

msg = dai.ImgFrame()
msg.setType(dai.ImgFrame.Type.BGR888p)
msg.setData(to_planar(cv2.resize(image, (IMG_WIDTH, IMG_HEIGHT)), (IMG_WIDTH, IMG_HEIGHT)))
msg.setWidth(IMG_WIDTH)
msg.setHeight(IMG_HEIGHT)

while True:
    in_q.send(msg)
    img = out_q.get()
    img_frame = img.getCvFrame()
    cv2.imshow("Output", img_frame)
    if cv2.waitKey(1) == ord('q'):
        break
```

gabrielthomsen, 2022-08-27:
That does not resolve it for me. I think actually added that line in my process of debugging

moratom, 2022-08-27:
Does the script I sent also not work? Otherwise, if the script works, please send a [MRE](https://docs.luxonis.com/en/latest/pages/support/?highlight=MRE#creating-minimal-reproducible-example) for the issue that persists.

gabrielthomsen, 2022-08-29:
The script you sent also does not work

gabrielthomsen, 2022-08-29:
Updating to 2.17 resolved the problem, so it looks like it has already been fixed. My apologies

closed 2022-08-29 by gabrielthomsen

moratom, 2022-08-29:
That's great to hear, I'm glad that resolved the issue :)