luxonis/depthai-core #565: OAK-D from kickstarter campaign: Error message: There is no Intrinsic matrix available for the the requested cameraID

cyberbeat, 2022-08-25:
This is my eeprom dump:

```
Is EEPROM available: 1
User calibration: {
    "batchName": "",
    "batchTime": 0,
    "boardConf": "",
    "boardCustom": "",
    "boardName": "BW1098OBC",
    "boardOptions": 0,
    "boardRev": "R0M0E0",
    "cameraData": [
        [
            2,
            {
                "cameraType": 0,
                "distortionCoeff": [
                    -5.801184177398682,
                    19.16950798034668,
                    0.00119276181794703,
                    0.0004464223748072982,
                    -21.091075897216797,
                    -5.8492302894592285,
                    19.370147705078125,
                    -21.298423767089844,
                    0.0,
                    0.0,
                    0.0,
                    0.0,
                    0.0,
                    0.0
                ],
                "extrinsics": {
                    "rotationMatrix": [
                        [
                            0.0,
                            0.0,
                            0.0
                        ],
                        [
                            0.0,
                            0.0,
                            0.0
                        ],
                        [
                            0.0,
                            0.0,
                            0.0
                        ]
                    ],
                    "specTranslation": {
                        "x": 3.750000238418579,
                        "y": 0.0,
                        "z": 0.0
                    },
                    "toCameraSocket": 0,
                    "translation": {
                        "x": -0.0,
                        "y": -0.0,
                        "z": -0.0
                    }
                },
                "height": 800,
                "intrinsicMatrix": [
                    [
                        858.3536376953125,
                        0.0,
                        640.1364135742188
                    ],
[... 188 lines cut ...]
            ],
            [
                -0.010951868258416653,
                0.9999397993087769,
                -0.0006345618749037385
            ],
            [
                0.01030136737972498,
                0.0007473922451026738,
                0.999946653842926
            ]
        ],
        "rightCameraSocket": 2
    },
    "version": 5
}

No factory calibration: No or invalid EEPROM configuration flashed, error: EEPROM_INVALID_DATA
User calibration raw: [5,0,170,85,122,255,127,63,207,44,128,59,53,40,95,58,188,49,128,187,124,255,127,63,87,180,52,58,206,114,94,186,119,147,53,186,246,255,127,63,82,184,143,66,154,153,153,61,154,153,25,61,1,109,150,137,66,66,87,49,48,57,56,79,66,67,0,0,0,0,0,0,0,82,48,77,48,69,48,0,0,0,0,1,152,248,127,63,153,81,51,60,204,230,40,188,119,111,51,188,14,252,127,63,186,88,38,186,17,199,40,60,165,236,67,58,129,252,127,63,122,77,86,68,0,0,0,0,147,252,31,68,0,0,0,0,85,135,86,68,93,235,198,67,0,0,0,0,0,0,0,0,0,0,128,63,162,150,86,68,0,0,0,0,187,8,32,68,0,0,0,0,189,189,86,68,142,65,198,67,0,0,0,0,0,0,0,0,0,0,128,63,77,250,127,63,246,240,230,187,124,153,54,60,96,109,230,59,80,254,127,63,182,135,189,58,6,195,54,188,192,61,179,186,220,251,127,63,231,188,238,192,4,63,167,189,194,135,157,61,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,205,247,119,192,139,128,42,65,52,3,20,59,120,18,36,185,122,211,8,193,57,198,123,192,83,64,46,65,99,106,12,193,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,77,163,185,192,39,91,153,65,114,86,156,58,204,13,234,57,134,186,168,193,229,44,187,192,16,246,154,65,44,99,170,193,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255, [... cut ...] ,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255]

```

What could I do?

themarpe, 2022-08-25:
I'll let @saching13  / @daxoft  take a look at this.

saching13, 2022-08-25:
I think you received a early device where we didn't do RGB calibration. Or it might have been over written by user calibration. 
We can try looking up the data if we have backup using mxid. Or else you can recalibrate it from depthai package.

cyberbeat, 2022-08-25:
I never did anything with calibration before, and only dumped it with the tool from the examples. It would be nice to at least have some factory default values, that I could flash to the device. Please notice, that there is NO factory eeprom data available in the device.

cyberbeat, 2022-08-25:
No I made a successful manual calibration (with your calibration.py tool), but that seems to have written an invalid eeprom:

```
Is EEPROM available: 1
User calibration: No user calibration: [json.exception.type_error.316] invalid UTF-8 byte at index 2: 0xA7
No factory calibration: No or invalid EEPROM configuration flashed, error: EEPROM_INVALID_DATA
User calibration raw: [5,0,170,85,3,0,0,0,0,66,87,49,48,57,56,79,66,67,0,0,0,0,0,0,0,82,48,77,48,69,48,0,0,0,0,118,253,127,63,183,163,231,59,100,206,171,187,23,133,231,187,89,254,127,63,53,77,59,58,167,247,171,59,81,149,49,186,21,255,127,63,179,234,127,63,128,115,79,60,79,71,181,188,23,180,79,188,185,250,127,63,239,174,17,186,209,52,181,60,231,44,91,58,241,239,127,63,1,2,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,255,56,4,128,7,135,31,131,188,68,0,0,0,0,187,226,109,68,0,0,0,0,17,70,188,68,220,177,2,68,0,0,0,0,0,0,0,0,0,0,128,63,165,188,117,192,24,65,138,66,104,11,63,187,60,86,152,187,18,165,213,194,154,181,131,192,212,141,140,66,197,40,218,194,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,109,150,137,66,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,255,32,3,0,5,0,63,237,89,68,0,0,0,0,42,232,29,68,0,0,0,0,30,49,90,68,34,129,194,67,0,0,0,0,0,0,0,0,0,0,128,63,248,218,202,65,13,252,135,194,168,202,114,187,201,73,162,187,187,174,217,67,150,14,204,65,133,111,142,194,64,196,220,67,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,82,184,143,66,0,2,22,252,127,63,215,70,174,59,245,128,28,188,115,85,177,187,234,251,127,63,70,8,32,188,146,164,27,60,166,222,32,60,226,249,127,63,126,154,112,64,118,88,174,188,51,112,178,190,0,0,112,64,0,0,0,0,0,0,0,0,0,32,3,0,5,0,83,108,89,68,0,0,0,0,127,36,31,68,0,0,0,0,46,78,89,68,199,249,194,67,0,0,0,0,0,0,0,0,0,0,128,63,65,22,166,192,178,172,149,65,51,10,22,187,25,239,177,187,144,255,193,193,24,251,167,192,86,49,151,65,55,41,195,193,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,82,184,143,66,0,1,163,245,127,63,194,82,184,187,37,49,138,60,96,135,183,59,230,254,127,63,37,135,194,58,15,66,138,188,254,28,182,186,154,246,127,63,186,198,240,192,144,45,195,189,196,141,42,62,0,0,240,192,0,0,0,0,0,0,0,0,2,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,25 [... cut ...]
[14442C10C12B4AD200] [3.2] [12.024] [system] [critical] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 09, address: 8028F158' '0'
[CRITICAL] [0:00:12.024538] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 09, address: 8028F158' '0'
  File "depthai_demo.py", line 655, in run
    self.instance.run_all(self.conf)
  File "depthai_demo.py", line 132, in run_all
    self.run()
  File "depthai_demo.py", line 271, in run
    self._device.startPipeline(self._pm.pipeline)
XLinkReadError: Couldn't read data from stream: '__rpc_main' (X_LINK_ERROR)

```

cyberbeat, 2022-08-25:
I now tried to flash the calibration-generated json with the calibration_flash tool from examples, now I get:

`Failed flashing calibration: No PROTECTED permissions to override protected EEPROM fields
`

cyberbeat, 2022-08-25:
I found out about the "magic" enviroment variables, but successful flash does not help, same result.

themarpe, 2022-08-26:
@cyberbeat thanks for the test here, and sorry for the issue - we've seen this happen on one more occasion but haven't been able to reproduce. I'll be taking a look at this to see why the corruption occurs. I'll get back to you soon.

themarpe, 2022-08-26:
@cyberbeat I've reproduced and fixed a part of the issue. Creating an update that will address this shortly.

cyberbeat, 2022-08-26:
Thanks, I picked up your commit on develop, and flashing works now with newest firmware.

closed 2022-10-04 by themarpe

cyberbeat, 2022-10-05:
For which release version this is fixed?

themarpe, 2022-11-14:
@cyberbeat latest has it in:)

Lxycherryup, 2025-07-12:
How to solve this problem:Failed flashing calibration: No PROTECTED permissions to override protected EEPROM fields ?