luxonis/depthai-core #568: Wish for using other features for tracking

cyberbeat, 2022-08-28:
For now you offer HARRIS or SHI_THOMASI features.
But for applications, where later registration is required, it would be nice to use something like SIFT, which could be followed with optical flow (get long trails, and export descriptors to some sort of database on host).
The simplest way would be to allow keypoints as input for feature-tracker (to run some things like feature detection / descriptor building on host). Even more usable would be to implement additional sift detection (would that be possible with opencl?) on device.

themarpe, 2022-08-29:
Thanks @cyberbeat for the feature request - I think such more custom capabilities will land in later revisions, most likely RVC3 or beyond. As RVC2 performs a lot with optimized hardware, its usually not as extensible.

cyberbeat, 2022-08-29:
I did not know anything about RVC3 yet. Mhm, at least it should be no such great effort to allow another input channel for the feature-tracking-node, which would take some keypoints (x,y coords) to track instead of own found corners?

Btw: why is the device-software closed source? It would be nice to be able to adapt the sdk to own needs, doing some openvino programming/extensions..perhaps the community could add nice easy to use on-device features besides from neural nets, and you could decide, which to adopt for offical sdk?

themarpe, 2022-08-29:
@cyberbeat 

The reason that device side is closed is two fold - one the SDK itself for the chip is proprietary and access&license has to be acquired to be able to access it. Second is that some control over IP has to remain for the actual ecosystem, to be able to continue developing and improve things (and not leave everything to be done by just community)

cyberbeat, 2022-08-30:
Thanks for clarifying. What would be the starting point to develop and upload complete own firmware blobs to the device? Is there some documentation about that somewhere already? And would do think about my other idea from above ("to allow another input channel for the feature-tracking-node, which would take some keypoints (x,y coords) to track instead of own found corners")?

themarpe, 2022-08-30:
@cyberbeat 

> What would be the starting point to develop and upload complete own firmware blobs to the device? Is there some documentation about that somewhere already? 

Proprietary tooling & SDK by SoC vendor is required for this - no starting point for it unfortunately.

> And would do think about my other idea from above ("to allow another input channel for the feature-tracking-node, which would take some keypoints (x,y coords) to track instead of own found corners")?

CC: @szabi-luxonis on the above

szabi-luxonis, 2022-09-02:
> And would do think about my other idea from above ("to allow another input channel for the feature-tracking-node, which would take some keypoints (x,y coords) to track instead of own found corners")?

Doable. Added to the nice-to-have task list. 
In case it's urgent, please reach out to priority support @cyberbeat