luxonis/depthai-core #569: `stereo sync - missing frames` Error when attempt to bring up multiple ImgManip and StereoDepth

chengguizi, 2022-08-29:
### Describe the bug
I'm trying to use multiple instance of StereoDepth node (with some ImageManip nodes attached infront). However i seem to run into performance issue.

I tried using to StereoDepth node, but one of them always get stuck in giving out disparity xout. I tried to turn on the debug output it shows this:
```
[18443010F139BE0F00] [1.2] [68.271] [system] [info] PRINT:LeonMss: stereo sync - missing frames!
1 (1280,800)
2 (1280,800)
d1 (1280,800)
[18443010F139BE0F00] [1.2] [68.337] [system] [info] PRINT:LeonMss: stereo sync - missing frames!
1 (1280,800)
2 (1280,800)
d1 (1280,800)
[18443010F139BE0F00] [1.2] [68.403] [system] [info] PRINT:LeonMss: stereo sync - missing frames!
1 (1280,800)
2 (1280,800)
d1 (1280,800)
[18443010F139BE0F00] [1.2] [68.469] [system] [info] PRINT:LeonMss: stereo sync - missing frames!
1 (1280,800)
2 (1280,800)
d1 (1280,800)
[18443010F139BE0F00] [1.2] [68.535] [system] [info] PRINT:LeonMss: stereo sync - missing frames!
1 (1280,800)
2 (1280,800)
d1 (1280,800)
[18443010F139BE0F00] [1.2] [68.601] [system] [info] PRINT:LeonMss: stereo sync - missing frames!
1 (1280,800)
2 (1280,800)
d1 (1280,800)
```

### To Reproduce
1.  I am on `multi_cam_support` branch with FFC type board
2.  I manage to generate a reproducible code in python. with OV9281 type sensors connected at port A,B,C. Turning on enable_2nd_stereo should reproduce the error

Link to the code https://cdn.discordapp.com/attachments/924798503270625290/1012931344235499600/manip_stereo_test_20220827.py

### Expected behaviour
There should be no stereo sync issue, and both streams of stereo disparity xlink output should work smoothly.

### Addtional context
https://discord.com/channels/790680891252932659/924798503270625290/1012396405429633054

chengguizi, 2022-08-29:
@Erol444 @szabi-luxonis Created this issue as instructed. Do let me know if this is reproducible at Luxonis side

szabi-luxonis, 2022-08-29:
@chengguizi 
1. if you want to perform rectification, no need to use ImageManip, can be done inside stereo node, by providing the mesh manually using `loadMeshData` API. By using multiple ImageManips you don't know in which order they are executed, unless there is a script node scheduling them.
2. Make sure all 3 cameras are synced between them. Preferably HW sync. 

`stereo sync - missing frames` means that received frames are not synchronized.

chengguizi, 2022-08-29:
An important note is that if I have 
```
enable_fsin = True
enable_2nd_manip = True
enable_2nd_stereo = False # setting this to true will make time sync warning appears
```
The arrival time looks very normal

```
d1 (1280,800) 2391
cam3 (1280,800) 2392 5859.085069
cam2 (1280,800) 2394 5859.085058
cam1 (1280,800) 2394 5859.085041
d1 (1280,800) 2392
cam3 (1280,800) 2393 5859.118393
cam2 (1280,800) 2395 5859.118379
cam1 (1280,800) 2395 5859.118358
d1 (1280,800) 2393
cam3 (1280,800) 2394 5859.151697
cam2 (1280,800) 2396 5859.151687
cam1 (1280,800) 2396 5859.151671
d1 (1280,800) 2394
cam3 (1280,800) 2395 5859.185009
cam2 (1280,800) 2397 5859.184999
cam1 (1280,800) 2397 5859.184986

```

chengguizi, 2022-08-29:
Another finding, if I crop the images to 400p, then I would not have a issue with the missing frame
`stereo.setInputResolution(640,400)`

Does this suggest performance issue with running 2 StereoDepth nodes concurrently?

szabi-luxonis, 2022-08-29:
Check stereo docs:
https://docs.luxonis.com/projects/api/en/latest/components/nodes/stereo_depth/#stereo-depth-fps
Subpixel + LR-check performance for 1280x720 is 34 fps.
Stereo nodes are time-sliced.

chengguizi, 2022-08-29:
> Check stereo docs: https://docs.luxonis.com/projects/api/en/latest/components/nodes/stereo_depth/#stereo-depth-fps Subpixel + LR-check performance for 1280x720 is 34 fps. Stereo nodes are time-sliced.

Ah thanks @szabi-luxonis ! Didn't think of the fps drop of subpixel + lr, had an assumption that 720p should work with 60fps, so sharing with 2 nodes would give 30fps.

I guess then this `missing frame` error is an expected behaviour? 

Two suggestions:
1. Would it be possible to make it a more direct warning, say `incoming stream FPS is after than StereoDepth node's capacity`?
2. In the case of capacity saturation, i think a better load sharing mechanism could be implemented. Currently one of the shared StereoDepth node effectively stops working. The natural degradation should be, both nodes runs at a reduced frequency.

szabi-luxonis, 2022-08-29:
> I guess then this missing frame error is an expected behaviour?

Yes. Frames are dropped, because they are not processed in-time. You may set input queue size 1 and non-blocking for both Stereo nodes.

Thanks for the suggestions. Each node is independent from each other. A separate balancer/governor can be implemented using script node.

chengguizi, 2022-08-30:
Hi @szabi-luxonis ,

Thanks for the recommendation, I think this is no longer a bug issue then. I shall close this thread!

closed 2022-08-30 by chengguizi