luxonis/depthai-core #572: Possible bug in `CalibrationHandler.cpp`

chengguizi, 2022-08-30:
### Code

`getImuToCameraExtrinsics` function and `computeExtrinsicMatrix` function in `CalibrationHandler.cpp`

https://github.com/luxonis/depthai-core/blob/41de567ea52443368224145b9e54a4f858abaeb8/src/device/CalibrationHandler.cpp#L436

https://github.com/luxonis/depthai-core/blob/41de567ea52443368224145b9e54a4f858abaeb8/src/device/CalibrationHandler.cpp#L522

### Expected Code

The arguments of `matMul` in this two lines should swap place. For example in `computeExtrinsicMatrix` function, `futureTransformationMatrix` will evenetually carries the information of transformation to `dstCamera`, so it should be on the left of the multifplication.

themarpe, 2022-08-30:
Thanks @chengguizi for the report - do you mind opening a PR against `develop` branch for it?

saching13, 2022-08-30:
So what this does is gives is the transformation Matrix which brings points in the first Coordinate system to the Second Coordinate system. This also means that this gets the transformation matrix which gives the IMU position in the given CameraID frame. This happens due to duality of rotation matrix here. Prefer to keep it that way since it aligns with the OpenCV stereo API's and Camera-to-Camera Extrinsics. 

Please let me know if I'm wrong here @chengguizi 
https://docs.opencv.org/3.4/d9/d0c/group__calib3d.html#ga91018d80e2a93ade37539f01e6f07de5

chengguizi, 2022-09-08:
Hi @saching13 , sorry for the late reply. Yes, understand it is hard to wrap around in mind. Let me try to explain my way of thinking the code.

Take for example `CalibrationHandler::getImuToCameraExtrinsics(CameraBoardSocket cameraId, bool useSpecTranslation) `. I understand this function is designed to give us a transformation matrix, that bring us corrdinates in IMU frame to the target Camera frame. In other word, change basis to camera, from IMU.

This is confirmed by how this line writes:
```
if(eepromData.imuExtrinsics.toCameraSocket == cameraId) {
        return transformationMatrix;
}
```

With that understanding, the following logic is problematic:
```
else {
        std::vector<std::vector<float>> localTransformationMatrix = getCameraExtrinsics(eepromData.imuExtrinsics.toCameraSocket, cameraId, useSpecTranslation);
        return matMul(transformationMatrix, localTransformationMatrix);
    }
```

`localTransformationMatrix` from the definition, gives us a transformation of basis from imuExtrinsics.toCameraSocket to cameraId (which is the camera id we want at the end); `transformationMatrix` is  change of basis from IMU to imuExtrinsics.toCameraSocket. By rules of multiplying between transformation matrix, to obtain a change of basis from IMU to cameraId, we should do `localTransformationMatrix * transformationMatrix` instead.

Am I speaking correctly?

chengguizi, 2022-09-17:
Hi @saching13, hope you can take a look into this when you can! Thanks.

chengguizi, 2022-10-18:
@saching13 Any chance to look into this yet? :)

chengguizi, 2023-11-01:
@Erol444  I think the bug has been fixed in latest v2.23.0, is it correct?

saching13, 2023-11-01:
Yes. this has been fixed. Sorry, I missed your tags before.

closed 2023-11-01 by saching13