luxonis/depthai-core #585: [BUG] Imu sequence dropped

dongxuanlb, 2022-09-14:
Because the accelerometer and gyroscope of the IMU work at different frequencies. Therefore, under normal circumstances, according to the current API design, at the same time, there is a high probability that there should be only one data in a package.

For example, I set the accelerometer to work at 125Hz and the gyroscope to work at 100Hz. At this time, the accelerometer should output every 8 milliseconds, and the gyroscope should output every 10 milliseconds.

packet0: accel(seq0) 0       gyro(seq0) 0
packet1: accel(seq1)  8
packet2:                                gyro(seq1) 10
packet3: accel(seq2) 16
packet4:                                gyro(seq2) 20
packet5: accel(seq3) 24
packet6:                                gyro(seq3) 30
packet7: accel(seq4) 32
packet8: accel(seq5) 40     gyro(seq4) 40

Assuming that the accelerometer and the gyroscope are the same clock signal, according to my understanding, the data should be read out at the same time in the 8th packet. Most of the time, a packet should only have data from one of the sensors.

But in fact, I found that the data output by the current API is that each package will contain the data of two sensors.

[2022-09-14 09:59:32.394] [info] s: 1, t:79:12.962, 43:12.967
[2022-09-14 09:59:32.406] [info] s: 1, t:80:12.97, 44:12.977
[2022-09-14 09:59:32.414] [info] s: 1, t:82:12.986, 45:12.987// accel dropped seq 81
[2022-09-14 09:59:32.424] [info] s: 1, t:83:12.994, 46:12.997
[2022-09-14 09:59:32.439] [info] s: 1, t:84:13.002, 47:13.007
[2022-09-14 09:59:32.444] [info] s: 1, t:86:13.017, 48:13.017// accel dropped seq 85
[2022-09-14 09:59:32.454] [info] s: 1, t:87:13.025, 49:13.027
[2022-09-14 09:59:32.464] [info] s: 1, t:88:13.033, 50:13.037
[2022-09-14 09:59:32.474] [info] s: 1, t:89:13.041, 51:13.047
[2022-09-14 09:59:32.483] [info] s: 1, t:91:13.057, 52:13.057// accel dropped seq 90
[2022-09-14 09:59:32.493] [info] s: 1, t:92:13.064, 53:13.067
[2022-09-14 09:59:32.508] [info] s: 1, t:93:13.072, 54:13.077

The code snippet to output the log is as follows:

void OAKD::daiCallback(std::string name, std::shared_ptr<dai::ADatatype> data) {
    if (!publish) return;
    if (name == "imu") {
        auto imuDataPtr = std::dynamic_pointer_cast<dai::IMUData>(data);
        for(int i = 0; i < imuDataPtr->packets.size(); ++i) {
                auto accel = imuDataPtr->packets[i].acceleroMeter;
                auto gyro = imuDataPtr->packets[i].gyroscope;
            
                double timeA = duration_cast<milliseconds>(accel.timestamp.get() - _baseTime).count() / 1000.;
                double timeG = duration_cast<milliseconds>(gyro.timestamp.get() - _baseTime).count() / 1000.;
                info("s: {}, t:{}:{}, {}:{}", imuDataPtr->packets.size(), accel.sequence, timeA, gyro.sequence, timeG);
        }

The configuration code snippet for the IMU is as follows:
        IMU_node->enableFirmwareUpdate(false);
        // 125, 250, 500
        IMU_node->enableIMUSensor(dai::IMUSensor::ACCELEROMETER_RAW, _options.fps_accel);
        // 100, 200, 400
        IMU_node->enableIMUSensor(dai::IMUSensor::GYROSCOPE_RAW, _options.fps_gyro);
        
        // above this threshold packets will be sent in batch of X, if the host is not blocked and USB bandwidth is available
        IMU_node->setBatchReportThreshold(1);
        // maximum number of IMU packets in a batch, if it's reached device will block sending until host can receive it
        // if lower or equal to batchReportThreshold then the sending is always blocking on device
        // useful to reduce device's CPU load and number of lost packets, if CPU load is high on device side due to multiple nodes
        IMU_node->setMaxBatchReports(10);
        IMU_node->out.link(xoutIMU->input);

It can be clearly seen that the logic of the API output is now:
1. Which sensor has a lower frequency and output according to this sensor, so that the sequence of this sensor will not be lost.
2. According to the sequence of fast sensors, there will be a lot of drops.

Please help, is there something wrong with the way I write the code, or is the current API really designed that way?

szabi-luxonis, 2022-09-14:
That's correct. The IMU output is synchronized to the lowest sampling rate.
E.g. if accelerometer running at 125hz, gyro at 100, the synchronized rate will be 100.

dongxuanlb, 2022-09-14:
I understand that the current design will lose the data part of the sensor with higher frequency, so that there are acccel and gyro data in each packet.

But this is not really "synchronized" by means of timestamps.

In the example of depthcore-ros, there is a FillImuData_LinearInterpolation method. According to the above description, this function should not work as it should.

So, is there a way to subscribe to accel or gyro separately, and then I pass the FillImuData_LinearInterpolation function to synchronize them according to the gyro timestamp?

szabi-luxonis, 2022-09-14:
>But this is not really "synchronized" by means of timestamps.

They are synchronized to the closest matching timestamp between samples, but not interpolated.

>So, is there a way to subscribe to accel or gyro separately, and then I pass the FillImuData_LinearInterpolation function to synchronize them according to the gyro timestamp?

It's not possible at the moment.