luxonis/depthai-core #586: hardware synchronization

yiyuandonghua, 2022-09-14:
Hi, I would like to ask some questions about hardware triggering for synchronization.
1. Can the FSYN hardware pin be used?
2. How much V is the high/low level of the level trigger?
3. Is there a test code for hardware triggering to take pictures?

Looking forward to your reply，thanks!

themarpe, 2022-09-16:
CC: @alex-luxonis @Luxonis-David

Luxonis-David, 2022-09-17:
@yiyuandonghua 

1. You can use the FSYN HW pin in script-node, not sure if this is the only option
2. 1V8
3. IDK, would need @alex-luxonis to comment 

FSYN is normally shorted between all sensors that are located on the device, so note that you will drive FSYNC on all sensors, if that is not wanted, you can change the resistor placement and i.e. disconnect center camera from stereo pair.