luxonis/depthai-core #591: Can TrackedFeatures add information to referenced images?

dongxuanlb, 2022-09-19:
### Start with the `why`:

According to the code of feature_tracker.cpp, I found that the seq and timestamp in TrackedFeatures are "corresponding to" Frame, and there is no order relationship from the log.

    [2022-09-19 20:49:37.006] [info] [imu]-A[29]1.538ms, G[105]1.54ms
    [2022-09-19 20:49:37.006] [info] [imu]-A[31]1.542ms, G[106]1.542ms
    [2022-09-19 20:49:37.006] [info] [imu]-A[32]1.544ms, G[107]1.545ms
    [2022-09-19 20:49:37.010] [info] [img]-L[50]1.518ms, R[50]1.518ms
    [2022-09-19 20:49:37.010] [info] [fet]-L[45]1.352ms, R[45]1.352ms    <------- add refFrameSeq 50
    [2022-09-19 20:49:37.022] [info] [imu]-A[33]1.546ms, G[108]1.547ms
    [2022-09-19 20:49:37.022] [info] [imu]-A[34]1.548ms, G[109]1.55ms
    [2022-09-19 20:49:37.022] [info] [imu]-A[36]1.552ms, G[110]1.552ms
    [2022-09-19 20:49:37.022] [info] [imu]-A[37]1.554ms, G[111]1.555ms
    [2022-09-19 20:49:37.022] [info] [imu]-A[38]1.556ms, G[112]1.557ms
    [2022-09-19 20:49:37.034] [info] [imu]-A[39]1.558ms, G[113]1.56ms
    [2022-09-19 20:49:37.034] [info] [imu]-A[41]1.562ms, G[114]1.562ms
    [2022-09-19 20:49:37.034] [info] [imu]-A[42]1.564ms, G[115]1.565ms
    [2022-09-19 20:49:37.034] [info] [imu]-A[43]1.566ms, G[116]1.567ms
    [2022-09-19 20:49:37.034] [info] [imu]-A[45]1.57ms, G[117]1.57ms
    [2022-09-19 20:49:37.045] [info] [img]-L[51]1.552ms, R[51]1.552ms
    [2022-09-19 20:49:37.045] [info] [fet]-L[46]1.385ms, R[46]1.385ms   <------- add refFrameSeq 51
    [2022-09-19 20:49:37.046] [info] [imu]-A[46]1.572ms, G[118]1.572ms

             ┌─────────────────┐
    inputConfig  │                 │       outputFeatures
    ────────────►│     Feature     ├────────────────────►
    inputImage   │     Tracker     │passthroughInputImage
    ────────────►│-----------------├────────────────────►
             └─────────────────┘

### Move to the `what`:
in TrackedFeatures add reference frame seq information.

szabi-luxonis, 2022-09-19:
I'm not sure what are you referring to @dongxuanlb 
The returned sequence number and timestamp from `TrackedFeatures` is inherited from the input image on which feature tracking was performed. In other word the sequence number and timestamp of `outputFeatures` is the same as `inputImage`/`passthroughInputImage`

dongxuanlb, 2022-09-20:
sorry for the previous description is not clear.

the code snippet as follow， referring to feature_tracker.cpp in example

    while(mRunning) {
        auto left = leftQueue->get<dai::ImgFrame>();
        auto right = rightQueue->get<dai::ImgFrame>();

        if (!(publish&&firstTs)) continue;

        double time0 = getFrameTime(left->getTimestamp());
        double time1 = getFrameTime(right->getTimestamp());
        if (_options.show_frame_timestamp) {
            info("[img]-L[{}]{}ms, R[{}]{}ms", left->getSequenceNum(), time0, right->getSequenceNum(), time1);
        }
        if (_options.get_feature) {
            auto trackedFeaturesLeft = featuresLeftQueue->get<dai::TrackedFeatures>();
            auto trackedFeaturesRight = featuresRightQueue->get<dai::TrackedFeatures>();
            info("[fet]-L[{}]{}ms, R[{}]{}ms", 
            trackedFeaturesLeft->getSequenceNum(), getFrameTime(trackedFeaturesLeft->getTimestamp()), 
            trackedFeaturesRight->getSequenceNum(), getFrameTime(trackedFeaturesRight->getTimestamp()));
        }

Judging by the documentation and the information you mentioned, the sequence of image frames and feature point frames in the log should be the same. But in reality, they are different.

    [2022-09-19 20:49:37.010] [info] [img]-L[50]1.518ms, R[50]1.518ms
    [2022-09-19 20:49:37.010] [info] [fet]-L[45]1.352ms, R[45]1.352ms

    [2022-09-19 20:49:37.045] [info] [img]-L[51]1.552ms, R[51]1.552ms
    [2022-09-19 20:49:37.045] [info] [fet]-L[46]1.385ms, R[46]1.385ms 

They are 5 frames apart. If this is normal, does it mean that the information of tracking frame 45 takes 5*33ms (30fps) on the device, which is about 150ms?

szabi-luxonis, 2022-09-20:
> They are 5 frames apart. If this is normal, does it mean that the information of tracking frame 45 takes 5*33ms (30fps) on the device, which is about 150ms?

There is some queuing happening on the device, if you want the lowest latency set input queue size of feature tracker to 1, non blocking:

```
    featureTracker.inputImage.setBlocking(False)
    featureTracker.inputImage.setQueueSize(1)
```

dongxuanlb, 2022-09-20:
> ```
>     featureTracker.inputImage.setBlocking(False)
>     featureTracker.inputImage.setQueueSize(1)
> ```

it still have 4 frames apart.

    [2022-09-20 09:58:50.650] [info] [img]-L[9]0.166ms, R[9]0.166ms
    [2022-09-20 09:58:50.650] [info] [fet]-L[5]0.032ms, R[5]0.032ms

dongxuanlb, 2022-09-20:
My thinking is that feature point detection and tracking must consume a lot of time. If this time is relatively long, such as more than 5ms, I will use the callback mechanism to subscribe to the feature point information to ensure that the image information will not be blocked.

So, I need to confirm that you mentioned `the sequence number and timestamp of outputFeatures is the same as inputImage/passthroughInputImage`

If yes, please help to confirm whether the feature point tracking really takes about 120ms to 150ms.

szabi-luxonis, 2022-09-20:
Corner detection takes about 4 5 ms at 720p, optical flow around 7 8 ms.

Make sure that you consume the latest arrived packet on host, by either setting host output queue size to 1 or using getAll API

dongxuanlb, 2022-09-20:
> Corner detection takes about 4 5 ms at 720p, optical flow around 7 8 ms.
> 
> Make sure that you consume the latest arrived packet on host, by either setting host output queue size to 1 or using getAll API

@szabi-luxonis I know where my problem is.

In my code logic, the oak device will be started first, but since other components have not been initialized, I prevent data from being sent.

In this state, I forgot that the Track data still needs to be fetched from the queue.

    auto left = leftQueue->get<dai::ImgFrame>();
    auto right = rightQueue->get<dai::ImgFrame>();
    // fix!
    if (_options.get_feature) {
        auto trackedFeaturesLeft = featuresLeftQueue->get<dai::TrackedFeatures>();
        auto trackedFeaturesRight = featuresRightQueue->get<dai::TrackedFeatures>();
    }
    if (!(publish&&firstTs)) continue;

many thanks

closed 2022-09-20 by szabi-luxonis