luxonis/depthai-core #594: Monitor Thread still exists after update on oak d poe

apollollopa, 2022-09-21:
[18443010814EF50F00] [192.168.70.57] [27664.186] [system] [info] Memory Usage - DDR: 162.13 / 340.93 MiB, CMX: 2.26 / 2.50 MiB, LeonOS Heap: 38.89 / 77.57 MiB, LeonRT Heap: 3.79 / 41.36 MiB
[18443010814EF50F00] [192.168.70.57] [27664.186] [system] [info] Temperatures - Average: 30.44 °C, CSS: 32.25 °C, MSS 29.59 °C, UPA: 29.59 °C, DSS: 30.32 °C
[18443010814EF50F00] [192.168.70.57] [27664.186] [system] [info] Cpu Usage - LeonOS 23.98%, LeonRT: 1.09%
829922
829927
829932
829937
[2022-09-21 01:08:09.564] [warning] Monitor thread (device: 18443010814EF50F00 [192.168.70.57]) - ping was missed, closing the device connection
F: [global] [    891565] [EventRead00Thr] tcpipPlatformRead:272 Cannot find file descriptor by key: 56
F: [global] [    891565] [Scheduler00Thr] tcpipPlatformWrite:300  Cannot find file descriptor by key: 56
Traceback (most recent call last):
  File "/home/mackerel/test_depthai.py", line 30, in <module>
    imageFrame:dai.ImgFrame = imageQueue.get()
RuntimeError: Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'Image' (X_LINK_ERROR)'

themarpe, 2022-09-21:
@apollollopa 
Do you mind editing the post with the MRE that leads to this (if its the same as in previous Issue, feel free to copy it over)

Also, what is your networking like? Monitor thread can usually shutdown connection on events when bandwidth and/or network stability isn't adequate. Make sure to double check that you have adequate 1Gbit link between you host and the device (preferably no WiFi in between)

apollollopa, 2022-09-21:
`import depthai as dai
import time
 
pipeline = dai.Pipeline()
 
camera = pipeline.create(dai.node.ColorCamera)
videoEncoder = pipeline.create(dai.node.VideoEncoder)
imageOutput = pipeline.create(dai.node.XLinkOut)
 
imageOutput.setStreamName("Image")
 
camera.setImageOrientation(dai.CameraImageOrientation.ROTATE_180_DEG)
camera.setBoardSocket(dai.CameraBoardSocket.RGB)
camera.setResolution(dai.ColorCameraProperties.SensorResolution.THE_4_K)
videoEncoder.setDefaultProfilePreset(camera.getFps(), dai.VideoEncoderProperties.Profile.MJPEG)
 
camera.video.link(videoEncoder.input)
videoEncoder.bitstream.link(imageOutput.input)
 
# 
info = dai.DeviceInfo('192.168.70.57')
 
with dai.Device(pipeline, info, maxUsbSpeed=dai.UsbSpeed.SUPER_PLUS) as device:
  imageQueue = device.getOutputQueue(name="Image", maxSize=1, blocking=True)
  device.setLogLevel(dai.LogLevel.DEBUG)
  device.setLogOutputLevel(dai.LogLevel.DEBUG)
 
  while True:
    imageFrame:dai.ImgFrame = imageQueue.get()
    print(imageFrame.getSequenceNum())
    time.sleep(0.001)

`

themarpe, 2022-09-25:
@apollollopa 

Do you mind answering the question WRT to you networking as asked above? I wonder if that might be the root cause here.

> Also, what is your networking like? Monitor thread can usually shutdown connection on events when bandwidth and/or network stability isn't adequate. Make sure to double check that you have adequate 1Gbit link between you host and the device (preferably no WiFi in between)