luxonis/depthai-core #599: synchronization between imu measurement and stereo images

SinnedHog, 2022-09-30:
I performed an imu-camera calibration using the open source library Kalibr (https://github.com/ethz-asl/kalibr) to find out the delay between the imu measurements and stereo images. I fixed the camera exposure time to 1ms and the result from the calibration indicated that the camera is 10ms slower than the imu timestamp. I have read in one of the github post (https://github.com/luxonis/depthai-core/issues/516) that even though the imu and stereo cameras are not hardware sync but they are running on the same clock. Therefore, the delay should not be larger than half of the camera exposure time, which in this case is 0.5ms. Is the result from kalibr expected? My experience from using kalibr is that it has been reliably given me accurate estimate. Has anyone encountered the same issue before?

Erol444, 2022-10-17:
Hi @SinnedHog ,
Here's a simple [demo of IMU + frame](https://github.com/luxonis/depthai-experiments/tree/master/gen2-syncing#imu--rgb--depth-timestamp-syncing) syncing based on timestamps. I hope this helps.
Thanks, Erik

SinnedHog, 2022-10-20:
Hi @Erol444,
I think is this not what I am looking for. What I meant is that I am still getting an time delay of 10ms even after aligning them using their timestamp. So the question is if the IMU time or mono images time is actually what it is? By the way, I did this calibration via ROS, not sure if it is the ROS wrapper (depthai-ros) problem. Do you have a script/program in depthai-python that allows me to log IMU measurements and mono images concurrently? I want to repeat the calibration using non ROS wrapper.

Erol444, 2022-10-23:
Hi @SinnedHog ,
I don't think it's possible to hardware sync IMU yet, or to reduce drift from the camera stream via firmware (yet). So my naive approach would be to have a higher IMU frequency, so max delay would be lower (eg at 250Hz it would be max 2ms).
Thanks, Erik

dongxuanlb, 2022-11-01:
Hi @SinnedHog 

According to my recent experience, OAKD's left and right monocular and imu do use the same clock.
The reason for the feeling of delay in programming is mainly to do the "calibration" of the basic time. Here I refer to the imu example in the example directory.

     if (name == IMU_STREAM_NAME) {
        auto imuDataPtr = std::dynamic_pointer_cast<dai::IMUData>(data);
        auto imuPackets = imuDataPtr->packets;

        if(!firstTs) {
            for(auto& imuPacket : imuPackets) {
                _baseTime = std::min(imuPacket.acceleroMeter.timestamp.get(), imuPacket.gyroscope.timestamp.get());
                firstTs = true;
                break;
            }
        }

Then all other frame times can be obtained by the following helper functions：

    double OAKD::getFrameTime(time_point<steady_clock, steady_clock::duration> time) {
        return duration_cast<milliseconds>(time - _baseTime).count() / 1000.;
    }

For example in my code：

        auto rgbPtr = std::dynamic_pointer_cast<dai::ImgFrame>(data);
        auto seqId = rgbPtr->getSequenceNum();
        _syncFrameMutex.lock();
        SyncFramePtr& sync = _preallocatedSyncFramePtrs[seqId % preallocated_item_num];
        sync->rgbImg = rgbPtr->getFrame();
        sync->rgbSeqId = seqId;
        sync->rgbT = **getFrameTime(rgbPtr->getTimestamp())**;
        publishSyncFrame(sync);

[2022-11-01 19:49:12.074] [info] [OAK-D]Publish Sync Frame
[2022-11-01 19:49:12.078] [info] [OAK-D][LImg][0]0ms, [RImg][0]2.466ms
[2022-11-01 19:49:12.080] [info] [OAK-D][RGB ][0]0ms
[2022-11-01 19:49:12.080] [info] [OAK-D][LPts][0]0ms, [RPts][0]0ms
[2022-11-01 19:49:12.120] [info] [OAK-D]Publish Sync Frame
[2022-11-01 19:49:12.123] [info] [OAK-D][LImg][0]2.466ms, [RImg][0]2.466ms
[2022-11-01 19:49:12.126] [info] [OAK-D][RGB ][0]2.467ms
[2022-11-01 19:49:12.130] [info] [OAK-D][LPts][0]2.466ms, [RPts][0]2.466ms
[2022-11-01 19:49:12.151] [info] [OAK-D]Publish Sync Frame
[2022-11-01 19:49:12.153] [info] [OAK-D][LImg][1]2.5ms, [RImg][1]2.5ms
[2022-11-01 19:49:12.155] [info] [OAK-D][RGB ][1]2.501ms
[2022-11-01 19:49:12.155] [info] [OAK-D][LPts][1]2.5ms, [RPts][1]2.5ms
[2022-11-01 19:49:12.192] [info] [OAK-D]Publish Sync Frame
[2022-11-01 19:49:12.196] [info] [OAK-D][LImg][2]2.533ms, [RImg][2]2.533ms
[2022-11-01 19:49:12.198] [info] [OAK-D][RGB ][2]2.534ms
[2022-11-01 19:49:12.200] [info] [OAK-D][LPts][2]2.533ms, [RPts][2]2.533ms

SinnedHog, 2022-11-03:
Hi @dongxuanlb, thanks for your response. I understand that according to the explanation given in #516 (which is asked by you), both imu timestamp and mono image timestamp are gotten from the same clock. in that case, by just comparing their timestamp, we can align them together. However, my question is really if there is any delay when assigning the timestamp to the imu measuremnt and the mono image measurement. Have you tried to perform a camera-imu calibration such as using the Kalibr (https://github.com/ethz-asl/kalibr) to find out whether the imu and camera are truly synchronized? 
After performing the camera-imu calibration, the result shows that there is a 7-10ms delay between these two sources which shouldn't be the case since my IMU rate is around 180Hz, therefore the delay should be less than 1.0/180/2 = 3ms. Therefore, I was wondering could there be any delay in assigning the timestamp to either imu or mono image when they are read in by the board.

saching13, 2022-11-16:
@SinnedHog  there is no delay in assigning the timestamp from our debugging. But it looks like there is some delay is getting the image frames out of USB. 
[image]
In the above image the plots are plotted where X axis is the time at which messages are published. And Y axis represents the header timestamp. Form this you can see that the header timestamp of image comes later relative to the IMU which is batched for better throughput.
But if we look at it form the header.timestamp being the X axis. they overlap as shown in the image below. Which means that there if you create a queue of IMU messages you can look-up for the closest IMU sample to the current image received for better sync here.
[image]

Thoughts ?