luxonis/depthai-core #600: [Problem with USB2] {Problem with USB2}

AdarshShukla-04, 2022-10-02:


themarpe, 2022-10-02:
@AdarshShukla-04 

Does other USB Devices work with this same USB2 cable? We haven't observed issues with USB2 only cabling yet. 
Its possible that its a charging only USB cable (no USB2 D+/- lines)

brmarkus, 2022-10-03:
@AdarshShukla-04 which exact OAK device do you use?

USB2 versus USB3 is about speed/bandwidth as well as power-transfer.

Do you use a device, which supports an external power-supply (some OAK-device do allow to optionally connect a power-supply)?

Can you try to use an actively powered USB(2)-hub between your machine and your OAK device?

AdarshShukla-04, 2022-10-04:
@brmarkus I am using OAK-1. With USB-2 sometimes it works properly and sometime not like in object detection or while capturing the images.

brmarkus, 2022-10-04:
Do you have a chance to use an actively powered USB-hub between your machine and your OAK-1 device, have you already tried?
Your machine might not provide enough power via the USB2-port - is it powered by battery or by a power-supply?