luxonis/depthai-core #605: [Feature-Request] Adding submodule init and update step in CMake

hardesh-peppermint, 2022-10-10:
Hi,

I wanted to add GIT submodule init and update step in CMake directly. Each time after updating the repository I have to use those commands manually. It'll be better if we add it directly in the main CMakeLists.txt file of the project. 

Sharing a reference doc: https://cliutils.gitlab.io/modern-cmake/chapters/projects/submodule.html

I can also open a PR to add this snippet in CMake. Let me know the next steps. 

Thanks,
Harshal

hardesh-peppermint, 2022-10-10:
FYI, the end goal is to build depthai-core using colcon directly.

themarpe, 2022-10-10:
Hi @hardesh-peppermint 

You may add the init command, but update recursive implicitly won't fly as it'll bork development, switching from development branches back to specific commit.

Recursive submodule init on the other hand could be done implicitly in section where shared detection is done IMO

hardesh-peppermint, 2022-10-11:
Okay, got it. What do you mean by shared detection? Should I add a separate cmake flag to update the submodules?

themarpe, 2022-11-08:
@hardesh-peppermint 

`depthai-core\shared\depthai-shared.cmake` in the section: `message(FATAL_ERROR "Submodule 'depthai-shared' not initialized/updated. Run 'git submodule update --init --recursive' first")`

And likewise for `depthai-bootloader-shared`

But in that case each should IMO initialize only itself: `git submodule update --init --recursive shared/depthai-shared` and `git submodule update --init --recursive shared/depthai-bootloader-shared`

hardesh-peppermint, 2022-11-10:
Sure. That should also work. Any plans on adding this in the next version or should I open a PR?

themarpe, 2022-11-11:
@hardesh-peppermint feel free to open a PR :)