luxonis/depthai-core #608: LNK2001 Error

Nicodenhertog101, 2022-10-12:
I'm Stuggeling with i think the CMake, this was the first time using it for me, i have googled every thing about the error and im learning lots but im not getting any further

Windows 10 VS2022 C++ (Win32)

no error's shown until i Build or Run the Local Debugger

the 19 LNK2001 errors i get
`
1>CAM.obj : error LNK2001: unresolved external symbol "public: int __thiscall dai::node::ColorCamera::getVideoHeight(void)const " (?getVideoHeight@ColorCamera@node@dai@@QBEHXZ)
1>CAM.obj : error LNK2001: unresolved external symbol "public: int __thiscall dai::node::ColorCamera::getVideoWidth(void)const " (?getVideoWidth@ColorCamera@node@dai@@QBEHXZ)
1>CAM.obj : error LNK2001: unresolved external symbol "public: void __thiscall dai::node::ColorCamera::setResolution(enum dai::ColorCameraProperties::SensorResolution)" (?setResolution@ColorCamera@node@dai@@QAEXW4SensorResolution@ColorCameraProperties@3@@Z)
1>CAM.obj : error LNK2001: unresolved external symbol "public: void __thiscall dai::node::ColorCamera::setBoardSocket(enum dai::CameraBoardSocket)" (?setBoardSocket@ColorCamera@node@dai@@QAEXW4CameraBoardSocket@3@@Z)
1>CAM.obj : error LNK2001: unresolved external symbol "public: __thiscall dai::node::ColorCamera::ColorCamera(class std::shared_ptr<class dai::PipelineImpl> const &,__int64)" (??0ColorCamera@node@dai@@QAE@ABV?$shared_ptr@VPipelineImpl@dai@@@std@@_J@Z)
1>CAM.obj : error LNK2001: unresolved external symbol "public: void __thiscall dai::node::MonoCamera::setResolution(enum dai::MonoCameraProperties::SensorResolution)" (?setResolution@MonoCamera@node@dai@@QAEXW4SensorResolution@MonoCameraProperties@3@@Z)
1>CAM.obj : error LNK2001: unresolved external symbol "public: void __thiscall dai::node::MonoCamera::setBoardSocket(enum dai::CameraBoardSocket)" (?setBoardSocket@MonoCamera@node@dai@@QAEXW4CameraBoardSocket@3@@Z)
1>CAM.obj : error LNK2001: unresolved external symbol "public: __thiscall dai::node::MonoCamera::MonoCamera(class std::shared_ptr<class dai::PipelineImpl> const &,__int64)" (??0MonoCamera@node@dai@@QAE@ABV?$shared_ptr@VPipelineImpl@dai@@@std@@_J@Z)
1>CAM.obj : error LNK2001: unresolved external symbol "public: void __thiscall dai::node::XLinkIn::setStreamName(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &)" (?setStreamName@XLinkIn@node@dai@@QAEXABV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@Z)
1>CAM.obj : error LNK2001: unresolved external symbol "public: __thiscall dai::node::XLinkIn::XLinkIn(class std::shared_ptr<class dai::PipelineImpl> const &,__int64)" (??0XLinkIn@node@dai@@QAE@ABV?$shared_ptr@VPipelineImpl@dai@@@std@@_J@Z)
1>CAM.obj : error LNK2001: unresolved external symbol "public: void __thiscall dai::node::EdgeDetector::setMaxOutputFrameSize(int)" (?setMaxOutputFrameSize@EdgeDetector@node@dai@@QAEXH@Z)
1>CAM.obj : error LNK2001: unresolved external symbol "public: __thiscall dai::node::EdgeDetector::EdgeDetector(class std::shared_ptr<class dai::PipelineImpl> const &,__int64)" (??0EdgeDetector@node@dai@@QAE@ABV?$shared_ptr@VPipelineImpl@dai@@@std@@_J@Z)
1>CAM.obj : error LNK2001: unresolved external symbol "public: void __thiscall dai::node::XLinkOut::setStreamName(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &)" (?setStreamName@XLinkOut@node@dai@@QAEXABV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@Z)
1>CAM.obj : error LNK2001: unresolved external symbol "public: __thiscall dai::node::XLinkOut::XLinkOut(class std::shared_ptr<class dai::PipelineImpl> const &,__int64)" (??0XLinkOut@node@dai@@QAE@ABV?$shared_ptr@VPipelineImpl@dai@@@std@@_J@Z)
1>CAM.obj : error LNK2001: unresolved external symbol "private: __int64 __thiscall dai::PipelineImpl::getNextUniqueId(void)" (?getNextUniqueId@PipelineImpl@dai@@AAE_JXZ)
1>CAM.obj : error LNK2001: unresolved external symbol "public: void __thiscall dai::Node::Output::link(class dai::Node::Input const &)" (?link@Output@Node@dai@@QAEXABVInput@23@@Z)
1>CAM.obj : error LNK2001: unresolved external symbol "public: __thiscall dai::Pipeline::Pipeline(void)" (??0Pipeline@dai@@QAE@XZ)
1>CAM.obj : error LNK2001: unresolved external symbol "public: virtual __thiscall dai::Device::~Device(void)" (??1Device@dai@@UAE@XZ)
1>CAM.obj : error LNK2001: unresolved external symbol "public: __thiscall dai::Device::Device(class dai::Pipeline const &)" (??0Device@dai@@QAE@ABVPipeline@1@@Z)`

themarpe, 2022-10-14:
@Nicodenhertog101 

Did you link with depthai-core.lib?

There is an example VS integration available on this branch: `msvc_template_libusb_dll_fix`

itberrios, 2023-01-25:
I'm having similar issues using Visual Studio, I'm seeing "unresolved external symbol..." messages. I have made sure to link with depthai-core.lib, but I'm not sure what's wrong.