luxonis/depthai-core #610: is there any way to project rgb to left or right img?

di-dii, 2022-10-15:
is there any way to project img of mid rgb camera onto left or right img?

besides. can the left or right camera output rgb img?

Thanks a lot.

Erol444, 2022-10-16:
Hi @di-dii ,
No, you can only align depth to any of the sensor. For projecting one stream to another you need to know the depth, and objects closer to the camera will have to be "projected"/moved more pixels. DepthAI doesn't support this feature yet. What would be the use-case of such feature?
Left/right cameras on all current OAK devices are monochrome, so they can't output color image.
Thanks, Erik

di-dii, 2022-10-18:
> 

in my case, I need to do some segmentation on color image, then I need to project the mask on left and right image,  so i can get some matched key point pairs betweend left and right in the ROI.  this will became part of my slam work. 
I know this kind of projection will cause some margin lost, but it's okay.