luxonis/depthai-core #612: [Feature-Request] Merge branches to mainline

jasonm1892, 2022-10-17:
Welcome back after a long pause (forced)!

I raised an issue half a year ago that features are all around the place, on different branches.

I see that multi cam support is still not merged. Image manip refactor somehow got merged.
I did some research, see here: https://github.com/luxonis/depthai-python/issues/550#issuecomment-1147376920

> Any update? @Luxonis-Brandon It's been 2 months+.
> 
> I can't get a status update on `multi_cam_support` branches either, or known issues. [luxonis/depthai#666 (comment)](https://github.com/luxonis/depthai/issues/666#issuecomment-1127881705) On discord `Luxonis-Erik` mentioned that `"I am not sure and I'm too afraid to ask:p It's been on the stale branch for quite a while now..."`, so I guess you don't know either **inside** the company? If it's done why it's not merged, if it's not done why can't we get a status update after asking multiple times, multiple people, for months and months?
> 
> I went back and checked older PRs, took me a long time to go through each .. You market your product as DepthAI = Depth + AI. Depth PR was opened on 23rd of november 2021, merged on 10th of december 2021. [luxonis/depthai-core#23](https://github.com/luxonis/depthai-core/pull/23) It was crashing, in some cases in seconds. There were/are a lot of reports like: `Fatal error. Please report to developers. Log: 'StereoSipp` There were attempts (?), with 0 success on making it stable: [luxonis/depthai-core#82](https://github.com/luxonis/depthai-core/pull/82) Opened on 9th of march, merged on 28th of april but didn't fix it. [luxonis/depthai-core#139](https://github.com/luxonis/depthai-core/pull/139)
> 
> Then it was finally fixed in this pr: [luxonis/depthai-core#200](https://github.com/luxonis/depthai-core/pull/200) on 22nd of september. **10 months** after it was introduced. Similar pattern as in case of ImageManip, work was done on it which failed, then got reassigned to another guy. From 23rd of november to 22nd of september Stereo node was unstable, multiple people reported it. **DepthAI didn't have depth.** I understand that individual people are not your target, but big companies, but how many potential customers turned away because **DepthAI couldn't provide a stable Depth** ? This goes back to **No one wants an unstable product**
> 
> In your demos you just recently enabled RGB alignment, previously the detected bounding boxes weren't mapped to the right depth values, so the returned X,Y,Z were off by a few meters in some cases. [luxonis/depthai-core#457](https://github.com/luxonis/depthai-core/pull/457) It was first mentioned in this issue: [luxonis/depthai#284 (comment)](https://github.com/luxonis/depthai/issues/284#issuecomment-797820055) (took 2 months to mainline) Then this PR: [luxonis/depthai-core#427](https://github.com/luxonis/depthai-core/pull/427) on 16th of march fixed it (?). Apparently it looks like it wasn't enabled because it was crashing? Again, no known issues except googling: `Fatal error. Please report to developers. Log: 'PlgWarpHW'`.
> 
> This results in a horrible user experience, checking out branches, not knowing what known issues are around, spending time debugging it with no chance of fixing it as user etc. and ultimately frustration.

No status on it, no replies, except next week/month.
This happened previously with StereoDepth, RGB alignment etc (see above, github history is still available). The multi-cam-support guy worked on those too for almost a year with no results.
Could you get someone who merges this feature into mainline? It's been a "open PR" for more than 1 year, from commits it looks like in development for 1.5 years. Could you just reassign the task, like you did with Stereo and Image manip and others so users can use it? 

You have products that you are selling with no out of the box support.

Also it looks like you didn't address stability issues at all. 2.17.1 worked fine for me, but you reverted in 2.17.2 so it's crashing again. 
Are you planning to fix your products?

jasonm1892, 2022-10-18:
Does anyone from Luxonis think that this is "good enough"? Delays of 1+ year for critical features? Stereo was crashing for more than a year, then ImageManip, then "multi cam support". 2 of 3 was handled by @alex-luxonis 
He is active on discord, but he answers only what he wants, ignoring the rest. Did the same thing with stereo crashes, or rgb alignment, until someone else did it. From user POV I don't care who does it. This is not a one time issue, it's been repetitive.

> I went back and checked older PRs, took me a long time to go through each .. You market your product as DepthAI = Depth + AI. Depth PR was opened on 23rd of november 2021, merged on 10th of december 2021. https://github.com/luxonis/depthai-core/pull/23 It was crashing, in some cases in seconds. There were/are a lot of reports like: Fatal error. Please report to developers. Log: 'StereoSipp There were attempts (?), with 0 success on making it stable: https://github.com/luxonis/depthai-core/pull/82 Opened on 9th of march, merged on 28th of april but didn't fix it. https://github.com/luxonis/depthai-core/pull/139
> 
> Then it was finally fixed in this pr: https://github.com/luxonis/depthai-core/pull/200 on 22nd of september. 10 months after it was introduced. Similar pattern as in case of ImageManip, work was done on it which failed, then got reassigned to another guy. From 23rd of november to 22nd of september Stereo node was unstable, multiple people reported it. DepthAI didn't have depth. I understand that individual people are not your target, but big companies, but how many potential customers turned away because DepthAI couldn't provide a stable Depth ? This goes back to No one wants an unstable product
> 
> In your demos you just recently enabled RGB alignment, previously the detected bounding boxes weren't mapped to the right depth values, so the returned X,Y,Z were off by a few meters in some cases. https://github.com/luxonis/depthai-core/pull/457 It was first mentioned in this issue: https://github.com/luxonis/depthai/issues/284#issuecomment-797820055 (took 2 months to mainline) Then this PR: https://github.com/luxonis/depthai-core/pull/427 on 16th of march fixed it (?). Apparently it looks like it wasn't enabled because it was crashing? Again, no known issues except googling: Fatal error. Please report to developers. Log: 'PlgWarpHW'.
> 
> This results in a horrible user experience, checking out branches, not knowing what known issues are around, spending time debugging it with no chance of fixing it as user etc. and ultimately frustration.

Can someone look at the above summary, and tell me that this is normal for Luxonis?
@Luxonis-Brandon @alex-luxonis @szabi-luxonis @themarpe @OanaMariaVatavu @Erol444 

@szabi-luxonis looks like you were the lucky one to take over stereo fix and others, do you think it's normal? Will you merge `multi cam support` too?

I just want 1+ year features to get into release :(

themarpe, 2022-10-20:
> I just want 1+ year features to get into release :(

Done :) 

---

Same summary as I have given the last time - the main reason for work on separate branches is to quickly accommodate custom needs. Is it normal? Its not - its a swift platform that moves, and instead of getting a SW drop once, we are on a continuous mission of improving our devices capabilities. Depending on whether or not you are on the side where the quick turnaround can be business critical and a great improvement for you OR on the other case where a certain efforts doesn't concern you at all, it can make a lot or very little sense. (and might seem like we are not looking out for the other case, which of course isn't true, if it has to be explicitly said)

The problems that pertain are usually hard and in some cases quick spinups yield in much longer proper integration which are the reasons for a large "branching factor N". We've been reducing it and have focused on bringing the integratable parts of various branches into mainline (as seen in our efforts in the past months/weeks).

Constructive feedback is always welcome and specific requests formalized and broken down into a good GitHub Issue are even better for us to take a deeper look into and attack as time permits.

Regards

closed 2022-10-20 by themarpe