luxonis/depthai-core #615: Oak-d lite runtime error in c++, but works fine with python API

koostosh, 2022-10-21:
I have an Oak-D lite camera running with python demo and other code samples, as long as I use python API.

I tried running multiple examples from code samples, each of them fails the same way. Example output from https://docs.luxonis.com/projects/api/en/latest/samples/MonoCamera/mono_preview/

```[2022-10-21 18:22:28.040] [trace] RPC: [1,1,5804304869041345055,[1.0]]
[2022-10-21 18:22:28.043] [debug] Log thread exception caught: Invalid Member Count
[2022-10-21 18:22:28.043] [trace] RPC: [1,1,17425566508637143278,null]
[2022-10-21 18:22:28.045] [trace] RPC: [1,1,16527326580805871264,[{"connections":[{"node1Id":1,"node1Output":"out","node1OutputGroup":"","node2Id":3,"node2Input":"in","node2InputGroup":""},{"node1Id":0,"node1Output":"out","node1OutputGroup":"","node2Id":2,"node2Input":"in","node2InputGroup":""}],"globalProperties":{"calibData":null,"cameraTuningBlobSize":null,"cameraTuningBlobUri":"","leonCssFrequencyHz":700000000.0,"leonMssFrequencyHz":700000000.0,"pipelineName":null,"pipelineVersion":null,"xlinkChunkSize":-1},"nodes":[[0,{"id":0,"ioInfo":[[["","inputControl"],{"blocking":true,"group":"","name":"inputControl","queueSize":8,"type":3,"waitForMessage":false}],[["","out"],{"blocking":false,"group":"","name":"out","queueSize":8,"type":0,"waitForMessage":false}],[["","raw"],{"blocking":false,"group":"","name":"raw","queueSize":8,"type":0,"waitForMessage":false}]],"name":"MonoCamera","properties":[185,0]}],[1,{"id":1,"ioInfo":[[["","inputControl"],{"blocking":true,"group":"","name":"inputControl","queueSize":8,"type":3,"waitForMessage":false}],[["","out"],{"blocking":false,"group":"","name":"out","queueSize":8,"type":0,"waitForMessage":false}],[["","raw"],{"blocking":false,"group":"","name":"raw","queueSize":8,"type":0,"waitForMessage":false}]],"name":"MonoCamera","properties":[185,0]}],[2,{"id":2,"ioInfo":[[["","in"],{"blocking":true,"group":"","name":"in","queueSize":8,"type":3,"waitForMessage":true}]],"name":"XLinkOut","properties":[185,0]}],[3,{"id":3,"ioInfo":[[["","in"],{"blocking":true,"group":"","name":"in","queueSize":8,"type":3,"waitForMessage":true}]],"name":"XLinkOut","properties":[185,0]}]]}]]
[2022-10-21 18:22:28.054] [trace] RPC: [1,1,10180360702496156555,null]
[2022-10-21 18:22:28.055] [trace] RPC: [1,1,14047900442330284907,null]
[2022-10-21 18:22:28.182] [debug] DataOutputQueue (right) closed
[2022-10-21 18:22:28.182] [debug] DataOutputQueue (left) closed
[2022-10-21 18:22:28.182] [debug] Device about to be closed...
[2022-10-21 18:22:28.182] [debug] Timesync thread exception caught: Couldn't read data from stream: '__timesync' (X_LINK_ERROR)
[2022-10-21 18:22:28.184] [debug] XLinkResetRemoteTimeout returned: X_LINK_ERROR
[2022-10-21 18:22:28.628] [debug] XLinkResetRemote of linkId: (0)
[2022-10-21 18:22:29.551] [debug] Device closed, 1368```

As it is similar to #265, I verified I am using newest nlohman-json 3.11.2 to compile depthai-core.
Disabling watchdog does not change anything, camera is connected via usb 3.0 cable.

Any suggestions?

alex-luxonis, 2022-10-21:
@koostosh The error below might come from some local changes in the work tree (e.g different device firmware binary), or perhaps the submodules not being updated (`depthai-shared` mainly):
`[debug] Log thread exception caught: Invalid Member Count`

Could you try with a fresh clone, and after cloning run: `git submodule update --init --recursive` , then build.

themarpe, 2022-11-08:
@koostosh 

Sorry for the delay on this - do you mind pointing the nlohmann json library to be 3.9.1 instead of later versions? 
On the other hand, do you mind expanding on how you've checked that both your app and depthai use the same library version?

koostosh, 2022-11-08:
I already tried downgrading json to 3.9.1 but it does not help.

I compiled both c++ and python api from scratch, so i just checked the code they used was the same revision. To be clear: I am not using hunter for dependecies.

I understand that best guess is that device firmware and my localy compiled library have some mismatch in dependencies. I will try downgrading all dependencies to exact versions used by hunter.

themarpe, 2022-11-08:
@koostosh 

Does libnop match as well?

koostosh, 2022-12-05:
I went through Hunter config and made sure my buildsystem uses exactly the same versions of:
nlohmann-json, XLink, Bzip2, libarchive, fp16, libnop, zlib and libusb. Error remains the same

themarpe, 2022-12-05:
@koostosh 
I'd suggest trying on latest `main` making sure all submodules are checked out at the exact specified commits as specified and that all the dependencies are also at the exact same commits

koostosh, 2022-12-05:
fixed by downgrading spdlog and letting it run on its own fmt version

closed 2022-12-05 by koostosh