luxonis/depthai-core #618: [BUG] {OAK-D-PRO-W POE C++ Fatal Error on MSS CPU}

samialperen, 2022-10-24:
After the recent merge from develop branch to main, I am not able to connect to OAK-D-PRO-W POE in ROS. I am getting the following error
```
[19443010116AFC1200] [192.168.120.20] [86.609] [system] [critical] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 09, address: 801F6FC8' '0'
```

Before the merge, everything was working fine. I am pretty sure develop branch introduced this bug because before the merge, I wanted to check develop branch and got the same error. Can someone please help me out?

themarpe, 2022-10-24:
Hi @samialperen 

Thanks for the report. We'll check it out!

Do you mind running `calibration_dump` example (either C++ or Python variant) and posting the output?

Also, do you observe the error in any other scenario as well (eg some example in this repository)?

samialperen, 2022-10-24:
[calibration.txt](https://github.com/luxonis/depthai-core/files/9853875/calibration.txt)
Above is the output for calibration dump

I managed to use cam_test.py under depthai-python. I can also use rgb_stereo_node.launch or rgb_publisher.launch in ROS. Seems the error happens with stereo_inertial_node.launch

themarpe, 2022-10-24:
@samialperen 
Do you mind testing with `stability_test` branch as an additional data point?
Otherwise will check this out tomorrow

samialperen, 2022-10-24:
here you go @themarpe 
[calibration-stable_test.txt](https://github.com/luxonis/depthai-core/files/9854923/calibration-stable_test.txt)

themarpe, 2022-10-24:
@samialperen 
I meant against the ROS example or any other that is failing for you

samialperen, 2022-10-24:
@themarpe same error occurs again on depthai-core stability_test branch with ROS example stereo_inertial_node.launch
```
[19443010116AFC1200] [192.168.120.20] [1351.541] [system] [critical] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 09, address: 801F6FAC' '0'
```

themarpe, 2022-10-24:
Thanks, I'll test this out tomorrow morning. Will get back to you as soon as I have findings

themarpe, 2022-10-25:
@samialperen 
We've found some potentially relevant fix on this issue - we'll be doing an update in `develop` soon and a release afterwards.

alex-luxonis, 2022-10-25:
`develop` branch updated with commit https://github.com/luxonis/depthai-core/commit/c877409:
> Update FW:
> - fix for high system load crash, especially on PoE devices
> - fix for OV9282 as MonoCamera on RGB/CAM_A socket (issue was black image)

(`develop` on depthai-python also updated)

samialperen, 2022-10-26:
@alex-luxonis @themarpe Thank you very much guys! I will verify the fix today.

samialperen, 2022-10-27:
@themarpe @alex-luxonis OAK-D-W-PRO POE works fine with stereo_inertial_publisher.launch in ROS with high FPS now. Thanks a lot for the fix :+1: . I am closing the issue.

closed 2022-10-27 by samialperen