luxonis/depthai-core #621: [BUG] {OAK-D  Fatal error on MSS CPU: trap: 09, address: 801F6FC8' '0'}

friend045, 2022-10-26:
**Describe the bug**

depthai-core version : 2.18.0

I updated the version to 2.18.0.
When I run the program, I get the following error.

```
[14442C10D17960D700] [2.2.1.3] [206.603] [system] [critical] Fatal error. 
Please report to developers. Log: 'Fatal error on MSS CPU: trap: 09, address: 801F6FC8' '0'
F: [global] [         0] [ThreadN] dispatcherResponseServe:925  no request for this response: XLINK_WRITE_RESP 1
```

But version 2.15.3 can be executed normally.

themarpe, 2022-10-26:
@friend045 

Thanks for the report, and sorry for the issues.

Do you mind sharing the script/program that caused this error? Also after how long did it occur?

We are working on ironing these out as well as releasing an updated v2.19.0 soon

alex-luxonis, 2022-10-26:
@friend045 Do you mind giving a try also to the `develop` branch, we fixed a potential crash that could have occurred under higher system load (especially higher camera FPS configured).
Latest commit, that has the fix: https://github.com/luxonis/depthai-core/commit/c87740908330100112bf12ea7f5aaf016fda89b2

friend045, 2022-10-26:
> @friend045
> 
> Thanks for the report, and sorry for the issues.
> 
> Do you mind sharing the script/program that caused this error? Also after how long did it occur?
> 
> We are working on ironing these out as well as releasing an updated v2.19.0 soon

@themarpe ,

Because it is a company project, the code may not be shared.
The error will appear within 20 seconds of the program execution.

friend045, 2022-10-26:
> @friend045 Do you mind giving a try also to the `develop` branch, we fixed a potential crash that could have occurred under higher system load (especially higher camera FPS configured). Latest commit, that has the fix: [c877409](https://github.com/luxonis/depthai-core/commit/c87740908330100112bf12ea7f5aaf016fda89b2)

@alex-luxonis ,
No problem, I'll try it .

alex-luxonis, 2022-10-27:
@friend045 Any specific error encountered?

Few things, in case not already tried:
- update git submodules: https://github.com/luxonis/depthai-core#building
- remove build directory in depthai-core: `rm -r build`
- clear Hunter package manager cache: https://github.com/luxonis/depthai-core#hunter

friend045, 2022-10-27:
> @friend045遇到任何具體錯誤？
> 
> 幾件事，以防尚未嘗試：
> 
> * 更新 git 子模塊：[https ://github.com/luxonis/depthai-core#building](https://github.com/luxonis/depthai-core#building)
> * 刪除 depthai-core 中的構建目錄：`rm -r build`
> * 清除 Hunter 包管理器緩存：[https ://github.com/luxonis/depthai-core#hunter](https://github.com/luxonis/depthai-core#hunter)

OS: Windows 
Compiler: Visual Studio 2019

```
depthai-resources.lib : fatal error LNK1107: invalid or corrupt file: cannot read at 0x360
```

themarpe, 2022-11-08:
@friend045 

We regularly prebuild shared libraries for Windows. You can find them here: https://github.com/luxonis/depthai-core/releases/tag/v2.19.0

Do you mind giving it a try with those? (latest 2.19.0)

friend045, 2022-11-08:
Currently I tried v2.19.0 on Windows 10 and it works fine. Apart from the relatively long startup time, I haven't encountered any problems for the time being.
But windows11 will encounter normal execution, but no output information.

themarpe, 2022-11-08:
@friend045 Does it fail with no output information or it still works as expected just that no output is received? That is using the prebuilt library, correct?

friend045, 2022-12-06:
Correct !
Use prebuilt libraries. The program continues to run, but does not output x, y, z coordinates.