luxonis/depthai-core #623: [Feature-Request] ImageManip to allow warp image into a different resolution

chengguizi, 2022-10-28:
### Start with the `why`:
Currently, the warp mesh function in depthai core assumes the input and output image size remains the same. For the purpose of undistortion for large field of view / fisheye cameras, we typically would like to undistort them into a few pinhole camera images of different views, as a single pinhole image only has the best details up to ~120 degree. However, currently, the `setWarpMesh` function forces us to warp into the original resolution. This is un-desirable in two ways:
- If we want to represent the original fisheye camera into say 4 pinhole images at half its resolution, we are still having to use ImageManip to warp each image at original resolution, which is a waste of warping computational power (by x4 times)
- As we need each pinhole image at a smaller resolution (as it is a partial view of the fisheye image, higher resolution will not give more info), currently we still need to cascade another ImangeManip node just to perform resize or croping. This would add the pipeline latency

### Move to the `what`:

Right now the API

`void setWarpMesh(const std::vector<Point2f> &meshData, int width, int height)`

assumes that
width = originalImageWidth / meshStep + 1
height = originalImageHeight / meshStep + 1

Otherwise an error is emitted at runtime by the firmware.

There is no way to specify output image size. So we have to do a warping of width x height image everytime. It is not possible if we want to produce a warped image of half size or other resolution, to reduce the computation cost.

### Move to the `how`:

https://docs.opencv.org/3.4/da/d54/group__imgproc__transform.html#gab75ef31ce5cdfb5c44b6da5f3b908ea4

Take reference to cv::remap() API, we should be able to use the dimensions of the meshData to infer the appropriate output image size.

So we if do something like

`setMeshStep(10)` and `setWarpMesh(meshData, 64 + 1, 40 + 1)`, the firmware should be able to infer that the desired output image size is 640*400.

themarpe, 2022-10-29:
@chengguizi 

Did you check setCrop&setResize function yet? I think this is what you are looking for (example should be also using it if I'm not mistaken)

chengguizi, 2022-10-30:
Yes i tested it. If I didnt remember wrongly, setWarpMesh would shadows `setCropRect` settings.

Example would be like below, in this case `setCropRect` has no effect.
```
# Warp preview frame 2
manip2 = pipeline.create(dai.node.ImageManip);
# Create a custom warp mesh
mesh2 = [
    (20, 20), (250, 100), (1460, 20),
    (100, 250), (250, 250), (1400, 250),
    (20, 1080), (250, 1000), (1460,1080)
]

manip2.setWarpMesh(mesh2, 3, 3);
manip2.setMaxOutputFrameSize(maxFrameSize);
manip2.initialConfig.setCropRect(0,0,0.5,0.5)
manip2.initialConfig.setResize(800, 600)
```

As for the use of `setResize` and `setWarpMesh` together, it seems to work. But I am afraid that it is implemented with two step process, where a full resolution warped image is processed and a rescale happens. That would be pretty inefficient if the original image is large (1MP/2MP), and we do want to work with a smaller resolution at the end (VGA resolution).

@themarpe  Hope i explained myself well :)

themarpe, 2022-11-08:
@chengguizi 

Sorry for the delay - setResize & setWarpMesh work in conjunction. Only a single operation is performed. setResize just notifies the underlying Warp engine on the output size it should warp into. Other steps are skipped when manually providing the warp mesh. I assume a cleaner API could be made by having the size be specified alongside `setWarpMesh` optionally as well.

Hopefully the current state works to address your needs - let me know:)

closed 2022-11-08 by themarpe

chengguizi, 2022-12-18:
@themarpe 
Thanks! I will test it soon on the v2.19 version and report back my findings!

chengguizi, 2023-01-14:
@themarpe Yes, indeed, I could get setWarpMesh and setResize working as the same ImageManip node!

A follow up question on this:
- If I do something like this `manip->initialConfig.setFrameType(dai::ImgFrame::Type::RAW8);` while using `setWarpMesh` / `setResize`. Which operation is performed first? color conversion or warping?
- It would be more efficient if the color-to-mono conversion takes place first, am I correct?

themarpe, 2023-01-14:
@chengguizi 

Great! 

One the questions:
 - the flow something along the lines of warp/crop/... -> resize -> format (the resize is done as part of of first step in current updated ImageManip impl)
 - Correct, though for ease of creating a generic implementation that isn't riddled with edge cases, the flow is fixed.

chengguizi, 2023-01-15:
Thanks! Clarifies a lot :)