luxonis/depthai-core #624: [BUG] {depthai.node.script KeyError: ('event',) develop branch}

samialperen, 2022-10-28:
I am trying to separate mono frames with dot projector then mono frames with flood light. Here is the code I am trying to replicate: https://github.com/luxonis/depthai-python/blob/develop/examples/MonoCamera/mono_preview_alternate_pro.py

Here is the related part in my code (C++)
```
        auto script = pipeline.create<dai::node::Script>(); 
        script->setProcessor(dai::ProcessorType::LEON_CSS);
        script->setScript(R"(
            dotBright = 500  # Note: recommended to not exceed 765, for max duty cycle
            floodBright = 200
            LOGGING = False  # Set `True` for latency/timings debugging
            node.warn(f'IR drivers detected: {str(Device.getIrDrivers())}')
            flagDot = False
            while True:
                # Wait first for a frame event, received at MIPI start-of-frame
                event = node.io['event'].get()
                if LOGGING: tEvent = Clock.now()
                # Immediately reconfigure the IR driver.
                # Note the logic is inverted, as it applies for next frame
                Device.setIrLaserDotProjectorBrightness(0 if flagDot else dotBright)
                Device.setIrFloodLightBrightness(floodBright if flagDot else 0)
                if LOGGING: tIrSet = Clock.now()
                # Wait for the actual frames (after MIPI capture and ISP proc is done)
                frameL = node.io['frameL'].get()
                if LOGGING: tLeft = Clock.now()
                frameR = node.io['frameR'].get()
                if LOGGING: tRight = Clock.now()
                if LOGGING:
                    latIR      = (tIrSet - tEvent               ).total_seconds() * 1000
                    latEv      = (tEvent - event.getTimestamp() ).total_seconds() * 1000
                    latProcL   = (tLeft  - event.getTimestamp() ).total_seconds() * 1000
                    diffRecvRL = (tRight - tLeft                ).total_seconds() * 1000
                    node.warn(f'T[ms] latEv:{latEv:5.3f} latIR:{latIR:5.3f} latProcL:{latProcL:6.3f} '
                            + f' diffRecvRL:{diffRecvRL:5.3f}')
                # Sync checks
                diffSeq = frameL.getSequenceNum() - event.getSequenceNum()
                diffTsEv = (frameL.getTimestamp() - event.getTimestamp()).total_seconds() * 1000
                diffTsRL = (frameR.getTimestamp() - frameL.getTimestamp()).total_seconds() * 1000
                if diffSeq or diffTsEv or (abs(diffTsRL) > 0.8):
                    node.error(f'frame/event desync! Fr-Ev: {diffSeq} frames,'
                            + f' {diffTsEv:.3f} ms; R-L: {diffTsRL:.3f} ms')
                # Route the frames to their respective outputs
                node.io['dotL' if flagDot else 'floodL'].send(frameL)
                node.io['dotR' if flagDot else 'floodR'].send(frameR)
                flagDot = not flagDot
        )");
```

I am using the depthai-core develop branch. Here is the error I am getting:
```
[18443010F1CFC51200] [192.168.123.20] [87230.329] [Script(13)] [warning] IR drivers detected: [('LM3644', 2, 99)]
[18443010F1CFC51200] [192.168.123.20] [87230.329] [Script(13)] [critical] KeyError: ('event',)

At:
  <script>(9): <module>
```

alex-luxonis, 2022-10-28:
@samialperen That error suggests an error in defining the pipeline, could you check if the camera `frameEvent` output is linked to a Script input named `event` (the name can be customized, but needs to match with the one used in the Python script).

https://github.com/luxonis/depthai-python/blob/4517dc8/examples/MonoCamera/mono_preview_alternate_pro.py#L94
```py
monoL.frameEvent.link(script.inputs['event'])
```

samialperen, 2022-10-28:
@alex-luxonis You are right, I missed that part. Yeeey! Now I have a new error:
```
[18443010F1CFC51200] [192.168.123.20] [311.246] [Script(13)] [warning] IR drivers detected: [('LM3644', 2, 99)]
[18443010F1CFC51200] [192.168.123.20] [311.442] [Script(13)] [error] frame/event desync! Fr-Ev: 21 frames, 524.505 ms; R-L: 0.011 ms
[18443010F1CFC51200] [192.168.123.20] [311.502] [Script(13)] [error] frame/event desync! Fr-Ev: 21 frames, 524.503 ms; R-L: 24.989 ms
[18443010F1CFC51200] [192.168.123.20] [311.529] [Script(13)] [error] frame/event desync! Fr-Ev: 23 frames, 574.455 ms; R-L: 24.994 ms
[18443010F1CFC51200] [192.168.123.20] [311.592] [Script(13)] [error] frame/event desync! Fr-Ev: 27 frames, 674.361 ms; R-L: -49.942 ms
[18443010F1CFC51200] [192.168.123.20] [315.633] [system] [critical] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 00, address: 00000000' '0'
```

Just in case, here is the part of my code related to the issue:
```
        // Stereo imges
        auto xoutLeft = pipeline.create<dai::node::XLinkOut>();
        auto xoutRight = pipeline.create<dai::node::XLinkOut>();

        // Alp
        auto xoutDotLeft = pipeline.create<dai::node::XLinkOut>();
        auto xoutDotRight = pipeline.create<dai::node::XLinkOut>();
        auto xoutFloodLeft = pipeline.create<dai::node::XLinkOut>();
        auto xoutFloodRight = pipeline.create<dai::node::XLinkOut>();

        xoutDotLeft->setStreamName("dot-left");
        xoutDotRight->setStreamName("dot-right");
        xoutFloodLeft->setStreamName("flood-left");
        xoutFloodRight->setStreamName("flood-right");

        auto script = pipeline.create<dai::node::Script>(); 
        script->setProcessor(dai::ProcessorType::LEON_CSS);
        script->setScript(R"(
            dotBright = 500  # Note: recommended to not exceed 765, for max duty cycle
            floodBright = 200
            LOGGING = False  # Set `True` for latency/timings debugging
            node.warn(f'IR drivers detected: {str(Device.getIrDrivers())}')
            flagDot = False
            while True:
                # Wait first for a frame event, received at MIPI start-of-frame
                event = node.io['event'].get()
                if LOGGING: tEvent = Clock.now()
                # Immediately reconfigure the IR driver.
                # Note the logic is inverted, as it applies for next frame
                Device.setIrLaserDotProjectorBrightness(0 if flagDot else dotBright)
                Device.setIrFloodLightBrightness(floodBright if flagDot else 0)
                if LOGGING: tIrSet = Clock.now()
                # Wait for the actual frames (after MIPI capture and ISP proc is done)
                frameL = node.io['frameL'].get()
                if LOGGING: tLeft = Clock.now()
                frameR = node.io['frameR'].get()
                if LOGGING: tRight = Clock.now()
                if LOGGING:
                    latIR      = (tIrSet - tEvent               ).total_seconds() * 1000
                    latEv      = (tEvent - event.getTimestamp() ).total_seconds() * 1000
                    latProcL   = (tLeft  - event.getTimestamp() ).total_seconds() * 1000
                    diffRecvRL = (tRight - tLeft                ).total_seconds() * 1000
                    node.warn(f'T[ms] latEv:{latEv:5.3f} latIR:{latIR:5.3f} latProcL:{latProcL:6.3f} '
                            + f' diffRecvRL:{diffRecvRL:5.3f}')
                # Sync checks
                diffSeq = frameL.getSequenceNum() - event.getSequenceNum()
                diffTsEv = (frameL.getTimestamp() - event.getTimestamp()).total_seconds() * 1000
                diffTsRL = (frameR.getTimestamp() - frameL.getTimestamp()).total_seconds() * 1000
                if diffSeq or diffTsEv or (abs(diffTsRL) > 0.8):
                    node.error(f'frame/event desync! Fr-Ev: {diffSeq} frames,'
                            + f' {diffTsEv:.3f} ms; R-L: {diffTsRL:.3f} ms')
                # Route the frames to their respective outputs
                node.io['dotL' if flagDot else 'floodL'].send(frameL)
                node.io['dotR' if flagDot else 'floodR'].send(frameR)
                flagDot = not flagDot
        )");

        // XLinkOut
        xoutLeft->setStreamName("left");
        xoutRight->setStreamName("right");
        if(rectify) {
            stereo->rectifiedLeft.link(xoutLeft->input);
            stereo->rectifiedRight.link(xoutRight->input);
        } else {
            stereo->syncedLeft.link(xoutLeft->input);
            stereo->syncedRight.link(xoutRight->input);
        }

        // Alp
        monoLeft->frameEvent.link(script->inputs["event"]);
        monoLeft->out.link(script->inputs["frameL"]);
        monoRight->out.link(script->inputs["frameR"]);

        script->outputs["dotL"].link(xoutDotLeft->input);
        script->outputs["dotR"].link(xoutDotRight->input);
        script->outputs["floodL"].link(xoutFloodLeft->input);
        script->outputs["floodR"].link(xoutFloodRight->input);
    }
    
    // Link plugins CAM -> STEREO -> XLINK
    stereo->setRectifyEdgeFillColor(0);
    monoLeft->out.link(stereo->left);
    monoRight->out.link(stereo->right);

    if(enableDepth) {
        stereo->depth.link(xoutDepth->input);
    } else {
        stereo->disparity.link(xoutDepth->input);
    }

    imu->out.link(xoutImu->input);
    std::cout << stereoWidth << " " << stereoHeight << " " << rgbWidth << " " << rgbHeight << std::endl;
    return std::make_tuple(pipeline, stereoWidth, stereoHeight);
```

samialperen, 2022-10-28:
@alex-luxonis  @themarpe 
Based on the other issue: https://github.com/luxonis/depthai-core/issues/618 
I decreased the stereo_fps to 1 and still the same issue exists. Here is the output
```
80 720 0 0
process[nodelet_manager-4]: started with pid [36556]
[ INFO] [1666974435.476537849]: Initializing nodelet with 16 worker threads.
process[depth_image_convertion_nodelet-5]: started with pid [36565]
Listing available devices...
Device Mx ID: 18443010F1CFC51200
process[depth_image_to_pointcloud-6]: started with pid [36589]
[18443010F1CFC51200] [192.168.123.20] [34.915] [Script(13)] [warning] IR drivers detected: [('LM3644', 2, 99)]
[18443010F1CFC51200] [192.168.123.20] [151.582] [system] [critical] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 00, address: 00000000' '0'
```

For debug purposes, here is my full code:
```
#include <camera_info_manager/camera_info_manager.h>
#include <depthai_ros_msgs/SpatialDetectionArray.h>
#include <ros/ros.h>
#include <sensor_msgs/Image.h>
#include <sensor_msgs/Imu.h>
#include <stereo_msgs/DisparityImage.h>

#include <cstdio>
#include <functional>
#include <iostream>
#include <tuple>
#include <cassert> // assert
#include <bits/stdc++.h>

// Inludes common necessary includes for development using depthai library
#include <depthai_bridge/BridgePublisher.hpp>
#include <depthai_bridge/DisparityConverter.hpp>
#include <depthai_bridge/ImageConverter.hpp>
#include <depthai_bridge/ImuConverter.hpp>
#include <depthai_bridge/SpatialDetectionConverter.hpp>

#include "depthai/depthai.hpp"

std::vector<std::string> usbStrings = {"UNKNOWN", "LOW", "FULL", "HIGH", "SUPER", "SUPER_PLUS"};

std::tuple<dai::Pipeline, int, int> createPipeline(bool enableSpatialDetection,
                                                   bool lrcheck,
                                                   bool extended,
                                                   bool subpixel,
                                                   int stereo_fps,
                                                   int confidence,
                                                   int LRchecktresh,
                                                   int detectionClassesCount,
                                                   std::string stereoResolution,
                                                   std::string rgbResolutionStr,
                                                   int rgbScaleNumerator,
                                                   int rgbScaleDinominator,
                                                   int previewWidth, 
                                                   int previewHeight,
                                                   bool syncNN,
                                                   std::string nnPath) {
    dai::Pipeline pipeline;
    pipeline.setXLinkChunkSize(0);
    auto controlIn = pipeline.create<dai::node::XLinkIn>();
    auto monoLeft = pipeline.create<dai::node::MonoCamera>();
    auto monoRight = pipeline.create<dai::node::MonoCamera>();
    
    // Alp
    //auto monoLeftNoDepth = pipeline.create<dai::node::MonoCamera>();
    //auto monoRightNoDepth = pipeline.create<dai::node::MonoCamera>();    

    auto stereo = pipeline.create<dai::node::StereoDepth>();
    auto xoutDepth = pipeline.create<dai::node::XLinkOut>();
[... 350 lines cut ...]

    dai::rosBridge::BridgePublisher<sensor_msgs::Image, dai::ImgFrame> leftFloodPublish(
        leftFloodQueue,
        pnh,
        std::string("leftFlood/image_rect"),
        std::bind(&dai::rosBridge::ImageConverter::toRosMsg, &converter, std::placeholders::_1, std::placeholders::_2),
        30,
        leftCameraInfo,
        "leftFlood");
    leftFloodPublish.addPublisherCallback();                        

    dai::rosBridge::BridgePublisher<sensor_msgs::Image, dai::ImgFrame> rightFloodPublish(
        rightFloodQueue,
        pnh,
        std::string("rightFlood/image_rect"),
        std::bind(&dai::rosBridge::ImageConverter::toRosMsg, &rightconverter, std::placeholders::_1, std::placeholders::_2),
        30,
        rightCameraInfo,
        "rightFlood");                
    rightFloodPublish.addPublisherCallback();

    ros::spin();
    return 0;
}
```