luxonis/depthai-core #626: [BUG] createImageManip fails with 2.18.0

ecmnet, 2022-10-30:
Using JavaCPP as wrapper worked fine up to 2.17.4 creating and running multiple nodes on OAKD Lite. Using 2.18.0, processing stops at the point where an ImageManip node is created without any messages or errors and the device provides no output anymore.

themarpe, 2022-11-01:
Hi @ecmnet 

I would need more information on this issue - do you mind appending DEPTHAI_LEVEL=debug output?
Also if you can create an MRE or pack the whole program with java libs alongside, I can take a look with a debugger (built in Debug configuration preferably)

ecmnet, 2022-11-01:
Hi @themarpe 

providing an example would take some more time. At least here's the debug output:

[2022-11-01 06:14:51.073] [debug] Library information - version: 2.18.0, commit:  from , build: 2022-10-28 12:12:15 +0000
[2022-11-01 06:14:51.090] [debug] Initialize - finished
[2022-11-01 06:14:51.259] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.21.tar.xz' open: 5ms, archive read: 178ms
[2022-11-01 06:14:51.638] [debug] Device - OpenVINO version: 2022.1
[2022-11-01 06:14:51.638] [debug] Device - BoardConfig: {"emmc":null,"gpio":[],"logDevicePrints":null,"logPath":null,"logSizeMax":null,"logVerbosity":null,"network":{"mtu":0,"xlinkTcpNoDelay":true},"pcieInternalClock":null,"sysctl":[],"uart":[],"usb":{"flashBootedPid":63037,"flashBootedVid":999,"maxSpeed":3,"pid":63035,"vid":999},"usb3PhyInternalClock":null,"watchdogInitialDelayMs":null,"watchdogTimeoutMs":null} 
libnop:
0000: b9 0e b9 05 81 e7 03 81 3b f6 81 e7 03 81 3d f6 03 b9 02 00 01 ba 00 be be bb 00 bb 00 be be be
0020: be be be be
[2022-11-01 06:14:51.919] [debug] Resources - Archive 'depthai-device-fwp-a555263a9dbff1166088bcb758ab4306dca9ae1a.tar.xz' open: 4ms, archive read: 839ms
[2022-11-01 06:14:53.083] [debug] Searching for booted device: DeviceInfo(name=20.3, mxid=1844301011B57D1200, X_LINK_BOOTED, X_LINK_USB_VSC, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[1844301011B57D1200] [20.3] [67.961] [system] [info] Memory Usage - DDR: 0.12 / 340.61 MiB, CMX: 2.05 / 2.50 MiB, LeonOS Heap: 7.22 / 77.48 MiB, LeonRT Heap: 2.88 / 41.35 MiB
[1844301011B57D1200] [20.3] [67.961] [system] [info] Temperatures - Average: 29.65 °C, CSS: 31.04 °C, MSS 29.59 °C, UPA: 28.62 °C, DSS: 29.35 °C
[1844301011B57D1200] [20.3] [67.961] [system] [info] Cpu Usage - LeonOS 9.34%, LeonRT: 1.67%
[2022-11-01 06:14:53.903] [debug] Schema dump: {"connections":[{"node1Id":1,"node1Output":"preview","node1OutputGroup":"","node2Id":6,"node2Input":"inputImage","node2InputGroup":""},{"node1Id":2,"node1Output":"depth","node1OutputGroup":"","node2Id":0,"node2Input":"in","node2InputGroup":""},{"node1Id":1,"node1Output":"preview","node1OutputGroup":"","node2Id":0,"node2Input":"in","node2InputGroup":""},{"node1Id":7,"node1Output":"out","node1OutputGroup":"","node2Id":5,"node2Input":"in","node2InputGroup":""},{"node1Id":6,"node1Output":"out","node1OutputGroup":"","node2Id":7,"node2Input":"in","node2InputGroup":""},{"node1Id":4,"node1Output":"out","node1OutputGroup":"","node2Id":2,"node2Input":"right","node2InputGroup":""},{"node1Id":3,"node1Output":"out","node1OutputGroup":"","node2Id":2,"node2Input":"left","node2InputGroup":""}],"globalProperties":{"calibData":null,"cameraTuningBlobSize":null,"cameraTuningBlobUri":"","leonCssFrequencyHz":700000000.0,"leonMssFrequencyHz":700000000.0,"pipelineName":null,"pipelineVersion":null,"xlinkChunkSize":-1},"nodes":[[0,{"id":0,"ioInfo":[[["","in"],{"blocking":true,"group":"","name":"in","queueSize":8,"type":3,"waitForMessage":true}]],"name":"XLinkOut","properties":[185,3,136,0,0,128,191,189,7,112,114,101,118,105,101,119,0]}],[1,{"id":1,"ioInfo":[[["","inputConfig"],{"blocking":false,"group":"","name":"inputConfig","queueSize":8,"type":3,"waitForMessage":false}],[["","frameEvent"],{"blocking":false,"group":"","name":"frameEvent","queueSize":8,"type":0,"waitForMessage":false}],[["","video"],{"blocking":false,"group":"","name":"video","queueSize":8,"type":0,"waitForMessage":false}],[["","raw"],{"blocking":false,"group":"","name":"raw","queueSize":8,"type":0,"waitForMessage":false}],[["","inputControl"],{"blocking":true,"group":"","name":"inputControl","queueSize":8,"type":3,"waitForMessage":false}],[["","isp"],{"blocking":false,"group":"","name":"isp","queueSize":8,"type":0,"waitForMessage":false}],[["","still"],{"blocking":false,"group":"","name":"still","queueSize":8,"type":0,"waitForMessage":false}],[["","preview"],{"blocking":false,"group":"","name":"preview","queueSize":8,"type":0,"waitForMessage":false}]],"name":"ColorCamera","properties":[185,23,185,27,0,3,0,0,0,185,3,0,0,0,185,5,0,0,0,0,0,185,5,0,0,0,0,0,0,0,0,0,0,0,0,185,3,0,0,0,185,3,0,0,0,0,0,0,0,0,0,0,0,0,0,255,255,0,0,0,129,224,1,129,128,2,255,255,255,255,0,136,0,0,112,65,136,0,0,128,191,136,0,0,128,191,1,185,4,0,0,0,0,3,3,4,4,4]}],[2,{"id":2,"ioInfo":[[["","right"],{"blocking":false,"group":"","name":"right","queueSize":8,"type":3,"waitForMessage":true}],[["","debugDispLrCheckIt2"],{"blocking":false,"group":"","name":"debugDispLrCheckIt2","queueSize":8,"type":0,"waitForMessage":false}],[["","rectifiedRight"],{"blocking":false,"group":"","name":"rectifiedRight","queueSize":8,"type":0,"waitForMessage":false}],[["","inputConfig"],{"blocking":false,"group":"","name":"inputConfig","queueSize":4,"type":3,"waitForMessage":false}],[["","debugExtDispLrCheckIt1"],{"blocking":false,"group":"","name":"debugExtDispLrCheckIt1","queueSize":8,"type":0,"waitForMessage":false}],[["","syncedLeft"],{"blocking":false,"group":"","name":"syncedLeft","queueSize":8,"type":0,"waitForMessage":false}],[["","confidenceMap"],{"blocking":false,"group":"","name":"confidenceMap","queueSize":8,"type":0,"waitForMessage":false}],[["","syncedRight"],{"blocking":false,"group":"","name":"syncedRight","queueSize":8,"type":0,"waitForMessage":false}],[["","outConfig"],{"blocking":false,"group":"","name":"outConfig","queueSize":8,"type":0,"waitForMessage":false}],[["","debugExtDispLrCheckIt2"],{"blocking":false,"group":"","name":"debugExtDispLrCheckIt2","queueSize":8,"type":0,"waitForMessage":false}],[[ [... cut ...]
[2022-11-01 06:14:57.396] [debug] DataOutputQueue (preview) closed
[2022-11-01 06:14:57.397] [debug] DataOutputQueue (nn) closed
[2022-11-01 06:14:57.665] [debug] Log thread exception caught: Couldn't read data from stream: '__log' (X_LINK_ERROR)
[2022-11-01 06:14:57.667] [debug] Timesync thread exception caught: Couldn't read data from stream: '__timesync' (X_LINK_ERROR)
[2022-11-01 06:14:57.667] [debug] Watchdog thread exception caught: Couldn't write data to stream: '__watchdog' (X_LINK_ERROR)

ecmnet, 2022-11-01:
Might be related to https://github.com/luxonis/depthai-core/issues/621

themarpe, 2022-11-08:
@ecmnet 

Do you mind testing out with latest 2.19.0? There were some bugfixes along the way, that might be relevant.

ecmnet, 2022-11-08:
Will do as soon as 2.19.0 Is available in javacpp 1.5.9-snapshot

themarpe, 2022-11-21:
Any news on this @ecmnet ?

There was also a v2.19.1 release which addresses some (runtime though) issues

ecmnet, 2022-12-01:
@themarpe According to the debug output with 1.5.9-SNAPSHOT using 2.19.1 the error mentioned above does not occur anymore. However, the code running with 2.17.4 still does not provide any data. As far as I can see up to now, the callback, I am using is not triggered anymore.

`       imageCallback = new OAKDImageCallback();
	 queue.addCallback(imageCallback);`

Callback class:

`private class OAKDImageCallback extends Callback { 

		public void call() {

			try {

				ImgFrame imgFrame = queue.tryGetImgFrame();
				if(imgFrame!=null && !imgFrame.isNull() ) {
					switch(imgFrame.getInstanceNum()) {
					case RGB_FRAME:
						rgb_tms = System.currentTimeMillis();
						if(rgb_mode)
							transfer_rgb.offer(imgFrame);
						else
							imgFrame.close();
						break;
					case MONO_FRAME:
						rgb_tms = System.currentTimeMillis();
						if(!rgb_mode)
							transfer_mono.offer(imgFrame);
						else
							imgFrame.close();  // Shouldn't this be called always??
						break;
					case DEPTH_FRAME:
						depth_tms = System.currentTimeMillis();
						transfer_depth.offer(imgFrame);
						break;
					}

				} 
			} catch (Exception e) {	}
		}
	}`

themarpe, 2022-12-01:
Thanks - does this happen with 2.19.1?

I'm sorry, but I unfortunatelly won't have time to dig into Java port for the time being. You can try checking further with a debugger to see where the native code gets stuck.

By "before stopping" does it close or just nothing comes through? Do you mind try running with DEPTHAI_WATCHDOG=0 (a warning should be printed)?

ecmnet, 2022-12-01:
Yes this is 2.19.1, stopping means just nothing coming through. Other API calls such as reading intrinsics etc are fine.
Will stick to 2.17..4 as a working solution for now.

ecmnet, 2022-12-03:
Tested with Watchdog disabled. Same result - polling stops after 2 or 3 frames. 

```
[2022-12-03 08:32:23.747] [warning] Watchdog disabled! In case of unclean exit, the device needs reset or power-cycle for next run
Connected cameras: 0 1 2  USB speed is HIGH
OAK-D Lite pipeline started

```
Callback does not work at all:

```
OAK-D Lite pipeline started
[2022-12-03 08:35:14.011] [warning] Watchdog disabled! In case of unclean exit, the device needs reset or power-cycle for next run
0 1 2 - HIGH
CameraPinholeRadial{fx=453.11041259765625, fy=453.11041259765625, skew=0.0, cx=331.7960205078125, cy=240.80442810058594, width=640, height=480}
OAK-D Lite depth pipeline 1 started: true
```

closed 2022-12-05 by themarpe

saudet, 2023-04-29:
FYI, seems to be fixed in recent versions: https://github.com/bytedeco/javacpp-presets/issues/1278#issuecomment-1519405288