luxonis/depthai-core #630: Linker Error with nlhomman/json

andrewwetzel23, 2022-11-03:
I am working on building a c++ project in visual studio. I downloaded depthai-core-v2.19.0-win64.zip from the releases page. I then included the necessary directories in my c/c++ additional include directories. Then I added depthai-core.lib and depthai-opencv.lib to my additional dependices. When I true to build I get a linker error dealing with the json package. (I have tried the master and develop branches from nlohmann/json).

LNK2019	unresolved external symbol "private: class nlohmann::json_abi_v3_11_2::basic_json<class std::map,class std::vector,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,bool,__int64,unsigned __int64,double,class std::allocator,struct nlohmann::json_abi_v3_11_2::adl_serializer,class std::vector<unsigned char,class std::allocator<unsigned char> > > __cdecl dai::PipelineImpl::serializeToJson(void)const " (?serializeToJson@PipelineImpl@dai@@AEBA?AV?$basic_json@Vmap@std@@Vvector@2@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@2@_N_J_KNVallocator@2@Uadl_serializer@json_abi_v3_11_2@nlohmann@@V?$vector@EV?$allocator@E@std@@@2@@json_abi_v3_11_2@nlohmann@@XZ) referenced in function "public: class nlohmann::json_abi_v3_11_2::basic_json<class std::map,class std::vector,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,bool,__int64,unsigned __int64,double,class std::allocator,struct nlohmann::json_abi_v3_11_2::adl_serializer,class std::vector<unsigned char,class std::allocator<unsigned char> > > __cdecl dai::Pipeline::serializeToJson(void)const " (?serializeToJson@Pipeline@dai@@QEBA?AV?$basic_json@Vmap@std@@Vvector@2@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@2@_N_J_KNVallocator@2@Uadl_serializer@json_abi_v3_11_2@nlohmann@@V?$vector@EV?$allocator@E@std@@@2@@json_abi_v3_11_2@nlohmann@@XZ)	StereoVision	C:\Users\Andrew\OneDrive - Lion Power, LLC\Documents\GitLab\Computer_Vision\deploy_cpp\StereoVision\Cam.obj	1

themarpe, 2022-11-03:
@andrewwetzel23 

Does you project also rely on nlohmann json? If so, make sure to use the same version. For 2.19.0 that'd be: `nlohmann json v3.9.1`

Or rely on the json provided by DepthAI.

If want another version, then build DepthAI alongside your project and skip using nlohmann from Hunter pkg manager

andrewwetzel23, 2022-11-07:
That fixed it. Thanks