luxonis/depthai-core #632: VC++ Compile Issue

kishordurve, 2022-11-04:
I am trying to get OAK-D to work in a VC++ (VS2015) application. I downloaded the bin, include , and lib folders from the Release V2.17.4 from the Realeases page here - https://github.com/luxonis/depthai-core/releases. However, I am unable to get the project to compile as dai:: Pipeline is not recognized in spite of having 'using namespace dai". Though dai is being recognized as a namespace dai:: Pipeline is not recognized. Can anyone help out, please? The OS is WIN10.
When I posted this issue on the discussion forum I was directed by Eric to try the project at https://github.com/luxonis/depthai-core/tree/mscv_fix/msvc. I did that but this happened:
1. Opened the project at depthai-core-mscv_fix\msvc and tried to build it. It could not find some include files. I copied the contents of the include folder to depthai-shared and then the 'file not found' errors vanished. However, I got 'CameraBoardSocket': is not a member of 'dai' error.
2. Started CMake and ran Configure in the CMake UI. After doing configuration I clicked Generate and got an error saying 'Cannot find source file: depthai-core-mscv_fix/shared/depthai-shared/src/datatype/DatatypeEnum.cpp'. I searched the parent folder but this file does not exist in it.
3. Saw that ALL_BUILD.vcxproj and other prj and sln files had been generated in the build folder by CMake. I opened ALL_BUILD.vcxproj in VS2015, changed the Platform Toolset to V14.0 and tried to build the project but got compile errors saying 'CameraBoardSocket' was not a member of dai
It would help a great deal if a working project with basic functionalities is created and shared with links to include, lib and bin folders.

themarpe, 2022-11-08:
@kishordurve 
I'd suggest checking out `dephtai-core-example` and compiling that instead. 
Note, VS2015 is likely unsupported. Last we've tried was 2019 & 2022 - so do keep that in mind.

kishordurve, 2022-11-10:
I couldn't get CMake to build a project. So I created a console application in VS2019. I took the code from depthai-core-example-main\src\main.cpp and pasted it in my project. After including all necessary .h/.hpp/.lib and dll files I was able to get it to run. However, the camera is not getting detected. I get a message 'USB protocol not available - if running in a container make sure that the following is set: -u dev/bus/USB --device-cgroup-rules=c 189:*rm'. What do I need to do?

themarpe, 2022-11-10:
@kishordurve 
Make sure to copy over the libusb.dll along your application - I think this is what is missing in your configuration

kishordurve, 2022-11-11:
Yes, it is running now. I am able to stream color image. 

Thank you!

themarpe, 2022-11-11:
Great!

closed 2022-11-11 by themarpe