luxonis/depthai-core #634: Windows cross-compilation for RPi 64 (C/C++)

adriaan-mernok, 2022-11-05:
I AM IN DESPERATE NEED OF ASSISTANCE

Hi, has anyone been able to set up a decent cross-compilation workflow? I've been struggling for more than a week now and can't seem to get going.

My current setup:

Development/build machine: Windows 10
IDE: CLion (I can switch, at this point I am fine with letting my subscription run out if it means I can get a decent workflow going).
Compilation toolchain: https://gnutoolchains.com/raspberry64/
Target machine: RPI CM4 running Raspberry OS Lite (64-bit)
Language: C/C++

I've cloned the depthai-core-example project as a starting point after I managed to successfully run a hello-world cross-compiled project.
Compiling the project on the rpi is not an option as it drastically slows development.
I want to stick to C/C++ as we have many ARM devices running on C/C++ & have build up quite a big code base for it.

I've read this discussion[https://discuss.luxonis.com/d/327-cross-compiling-depthai-core-example], but it does not seem like Mike has been very successful.

themarpe, 2022-11-05:
Hi @adriaan-mernok 

That post is a tad stale now - libusb is also handled by Hunter package manager, so there are no more external dependencies to take care of.

I'd suggest writing a good toolchain file for yor cross compiler, then pass it to CMake when building.

You can also try bulding core first with passing the toolchain and option BUILD_SHARED_LIBS=ON , then building the app separately

adriaan-mernok, 2022-11-05:
Hi @themarpe, any chance you could provide me with an example project? As I said, I've been trying to get this to work for about a week now with no success. My knowledge regarding cmake is quite limited, thus I opted for an IDE such as CLion to somewhat ease toolchain creation and management. It is my understanding that I would have to cross-compile opencv as well if I don't wish to manually install it on each device? I prefer compiling as a static library with the project since that simplifies the process of getting firmware on the devices in production - single binary loaded to the device as opposed to adding all the libraries as well as the binary project file.

themarpe, 2022-11-07:
@adriaan-mernok 

Sorry, I'd love to help with a step by step, but currently don't have the bandwidth to do so.

What I'd recommend is installing an Linux OS (perhaps in a VM), then trying https://github.com/dockcross/dockcross as is (targeting ARMv8), following the same compilation steps as described in depthai-core-example, just with dockcross prepended

I think this should work as expected. Let me know.

WRT static compilation and opencv, I'd suggest just relying on manually installed libopencv-dev on the target machine, and pointing the sysroot there.

Cross compilation can get tricky fast, so if not 100% necessary, I'd suggest first just working on the target directly. For that I'd recommend VSCode IDE & Remote SSH to open the target folder on your host directly.