luxonis/depthai-core #635: [BUG] {Poor performance on MacOS}

ZumWoelkchen, 2022-11-07:
**What**
I'm experiencing strange performance issues on MacOS. The rgb_depth_aligned sample runs at about 10 FPS which should run at 30 FPS. And basically all samples that include highish rgb resolutions run either very slow or have a really high latency ~3 Sec.

**How to reproduce:**
Clone the depthai-core repo
Build with this configuration: 
cmake -S. -Bbuild -D’BUILD_SHARED_LIBS=ON’ -D’DEPTHAI_BUILD_EXAMPLES=ON’ -D’CMAKE_INSTALL_PREFIX=/usr/local’
sudo cmake --build build -j16 --target install
run build/examples/rgb_depth_aligned or build/examples/rgb_video

**Expected behavior**
Smooth video and depth with 30 FPS

**Result**
Slow, laggy video with high latency

**Additional context**
I benchmarked frame acquisition times and it seems to be stuck at 100ms (10 FPS).
The same code and all samples work as expected on Ubuntu 20.04 (Stable, expected framerate and no lags at all)

getUsbSpeed returns SUPER in all tests. Verbose logs show no issues.

OAK:
OAK-D Pro W (IMX378 RGB)

Mac:
MacBook Pro 16" 2019 - 2,4 Ghz Intel Core i9 - 16GB RAM - AMD Radeon Pro 5300M

Ubuntu:
Intel Core i7 11800H - 32GB RAM - Nvidia RTX 3070 Laptop GPU

themarpe, 2022-11-08:
Thanks for the report @ZumWoelkchen 

Do you mind cross checking with Python library & examples? I think this boils down to displaying the frames. 
If you remove the `imshow` calls and just log the framerate, is it running as expected?

ZumWoelkchen, 2022-12-22:
Getting back a little late but after some extensive testing i can confirm that the issue lies in single-threaded imshow/OpenCV on MacOS.