luxonis/depthai-core #637: [BUG] Poor performance for transferring mono frames to host, especially on PoE devices

moratom, 2022-11-09:
**Describe the bug**
The OAK-D-PRO-PoE  seems to not perform very well when transfering 640x400 mono frames,
achieving 30 FPS on both sensor which translates to 16.3 Mbps of throughput.

The OAK-D-PRO usb version does 90 FPS with the same example, so about 3x better.
In both cases the bottleneck seems to be the LeonOS CPU as it is maxed out at 100%.

I initially tested with Depthai 2.19.0.0, and then checked with versions since 2.14.0.0 and observed similar behavior, 
so this likely isn't a regression.

Debug log:
```
[2022-11-09 18:33:10.993] [debug] Python bindings - version: 2.19.0.0 from  build: 2022-11-01 23:33:55 +0000
[2022-11-09 18:33:10.993] [debug] Library information - version: 2.19.0, commit: d29af8b0a5bbbc8fedf09d8ccfcd6bac8f30d8bc from 2022-11-01 21:18:27 +0100, build: 2022-11-01 23:12:42 +0000
[2022-11-09 18:33:10.994] [debug] Initialize - finished
[2022-11-09 18:33:10.994] [debug] Environment 'DEPTHAI_PROTOCOL' set to 'tcpip'
[2022-11-09 18:33:11.061] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.21.tar.xz' open: 2ms, archive read: 63ms
[2022-11-09 18:33:11.216] [debug] Device - OpenVINO version: 2022.1
[2022-11-09 18:33:11.216] [debug] Device - BoardConfig: {"emmc":null,"gpio":[],"logDevicePrints":null,"logPath":null,"logSizeMax":null,"logVerbosity":null,"network":{"mtu":0,"xlinkTcpNoDelay":true},"pcieInternalClock":null,"sysctl":[],"uart":[],"usb":{"flashBootedPid":63037,"flashBootedVid":999,"maxSpeed":5,"pid":63035,"vid":999},"usb3PhyInternalClock":null,"watchdogInitialDelayMs":null,"watchdogTimeoutMs":null} 
libnop:
0000: b9 0e b9 05 81 e7 03 81 3b f6 81 e7 03 81 3d f6 05 b9 02 00 01 ba 00 be be bb 00 bb 00 be be be
0020: be be be be
[2022-11-09 18:33:11.327] [debug] Resources - Archive 'depthai-device-fwp-b3aeaf23ff5857fc8f79d412ceefc08da23e7aad.tar.xz' open: 1ms, archive read: 326ms
[2022-11-09 18:33:11.348] [debug] Searching for booted device: DeviceInfo(name=192.168.34.171, mxid=184430105170C51200, X_LINK_BOOTLOADER, X_LINK_TCP_IP, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[2022-11-09 18:33:11.573] [debug] Connected bootloader version 0.0.19
[2022-11-09 18:33:11.574] [info] New bootloader version available. Device has: 0.0.19, available: 0.0.21
[2022-11-09 18:33:12.468] [debug] Booting FW with Bootloader. Version 0.0.19, Time taken: 894ms
[2022-11-09 18:33:12.468] [debug] DeviceBootloader about to be closed...
[2022-11-09 18:33:12.468] [debug] XLinkResetRemote of linkId: (0)
[2022-11-09 18:33:13.716] [debug] DeviceBootloader closed, 1248
[2022-11-09 18:33:13.719] [debug] Searching for booted device: DeviceInfo(name=192.168.34.171, mxid=184430105170C51200, X_LINK_BOOTED, X_LINK_TCP_IP, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[184430105170C51200] [192.168.34.171] [4.903] [system] [info] Memory Usage - DDR: 0.12 / 340.61 MiB, CMX: 2.05 / 2.50 MiB, LeonOS Heap: 25.02 / 77.48 MiB, LeonRT Heap: 2.89 / 41.35 MiB
[184430105170C51200] [192.168.34.171] [4.903] [system] [info] Temperatures - Average: 47.74 °C, CSS: 49.14 °C, MSS 47.13 °C, UPA: 47.13 °C, DSS: 47.58 °C
[184430105170C51200] [192.168.34.171] [4.903] [system] [info] Cpu Usage - LeonOS 42.58%, LeonRT: 0.59%
[2022-11-09 18:33:17.623] [debug] Schema dump: {"connections":[{"node1Id":2,"node1Output":"out","node1OutputGroup":"","node2Id":3,"node2Input":"in","node2InputGroup":""},{"node1Id":0,"node1Output":"out","node1OutputGroup":"","node2Id":1,"node2Input":"in","node2InputGroup":""}],"globalProperties":{"calibData":null,"cameraTuningBlobSize":null,"cameraTuningBlobUri":"","leonCssFrequencyHz":700000000.0,"leonMssFrequencyHz":700000000.0,"pipelineName":null,"pipelineVersion":null,"xlinkChunkSize":-1},"nodes":[[0,{"id":0,"ioInfo":[[["","out"],{"blocking":false,"group":"","name":"out","queueSize":8,"type":0,"waitForMessage":false}],[["","raw"],{"blocking":false,"group":"","name":"raw","queueSize":8,"type":0,"waitForMessage":false}],[["","frameEvent"],{"blocking":false,"group":"","name":"frameEvent","queueSize":8,"type":0,"waitForMessage":false}],[["","inputControl"],{"blocking":true,"group":"","name":"inputControl","queueSize":8,"type":3,"waitForMessage":false}]],"name":"MonoCamera","properties":[185,7,185,27,0,3,0,0,0,185,3,0,0,0,185,5,0,0,0,0,0,185,5,0,0,0,0,0,0,0,0,0,0,0,0,185,3,0,0,0,185,3,0,0,0,0,0,0,0,0,0,0,0,0,0,1,255,2,136,0,0,240,66,3,3]}],[1,{"id":1,"ioInfo":[[["","in"],{"blocking":true,"group":"","name":"in","queueSize":8,"type":3,"waitForMessage":true}]],"name":"XLinkOut","properties":[185,3,136,0,0,128,191,189,4,108,101,102,116,0]}],[2,{"id":2,"ioInfo":[[["","out"],{"blocking":false,"group":"","name":"out","queueSize":8,"type":0,"waitForMessage":false}],[["","raw"],{"blocking":false,"group":"","name":"raw","queueSize":8,"type":0,"waitForMessage":false}],[["","frameEvent"],{"blocking":false,"group":"","name":"frameEvent","queueSize":8,"type":0,"waitForMessage":false}],[["","inputControl"],{"blocking":true,"group":"","name":"inputControl","queueSize":8,"type":3,"waitForMessage":false}]],"name":"MonoCamera","properties":[185,7,185,27,0,3,0,0,0,185,3,0,0,0,185,5,0,0,0,0,0,185,5,0,0,0,0,0,0,0,0,0,0,0,0,185,3,0,0,0,185,3,0,0,0,0,0,0,0,0,0,0,0,0,0,2,255,2,136,0,0,240,66,3,3]}],[3,{"id":3,"ioInfo":[[["","in"],{"blocking":true,"group":"","name":"in","queueSize":8,"type":3,"waitForMessage":true}]],"name":"XLinkOut","properties":[185,3,136,0,0,128,191,189,5,114,105,103,104,116,0]}]]}
[2022-11-09 18:33:17.623] [debug] Asset map dump: {"map":{}}
[184430105170C51200] [192.168.34.171] [4.934] [MonoCamera(2)] [info] Using board socket: 2, id: 2
[184430105170C51200] [192.168.34.171] [4.934] [MonoCamera(0)] [info] Using board socket: 1, id: 1
[184430105170C51200] [192.168.34.171] [4.935] [system] [info] SIPP (Signal Image Processing Pipeline) internal buffer size '16384'B
[184430105170C51200] [192.168.34.171] [4.970] [system] [info] MonoCamera allocated resources: no shaves; cmx slices: [13-15] 

[184430105170C51200] [192.168.34.171] [5.906] [system] [info] Memory Usage - DDR: 3.46 / 340.61 MiB, CMX: 2.08 / 2.50 MiB, LeonOS Heap: 51.56 / 77.48 MiB, LeonRT Heap: 3.71 / 41.35 MiB
[184430105170C51200] [192.168.34.171] [5.906] [system] [info] Temperatures - Average: 49.19 °C, CSS: 50.92 °C, MSS 47.58 °C, UPA: 48.70 °C, DSS: 49.59 °C
[184430105170C51200] [192.168.34.171] [5.906] [system] [info] Cpu Usage - LeonOS 92.26%, LeonRT: 9.86%
[184430105170C51200] [192.168.34.171] [6.906] [system] [info] Memory Usage - DDR: 3.46 / 340.61 MiB, CMX: 2.08 / 2.50 MiB, LeonOS Heap: 51.56 / 77.48 MiB, LeonRT Heap: 3.71 / 41.35 MiB
[184430105170C51200] [192.168.34.171] [6.906] [system] [info] Temperatures - Average: 49.75 °C, CSS: 51.58 °C, MSS 49.14 °C, UPA: 48.70 °C, DSS: 49.59 °C
[184430105170C51200] [192.168.34.171] [6.907] [system] [info] Cpu Usage - LeonOS 99.99%, LeonRT: 6.25%
[184430105170C51200] [192.168.34.171] [7.911] [system] [info] Memory Usage - DDR: 3.46 / 340.61 MiB, CMX: 2.08 / 2.50 MiB, LeonOS Heap: 51.81 / 77.48 MiB, LeonRT Heap: 3.71 / 41.35 MiB
[184430105170C51200] [192.168.34.171] [7.911] [system] [info] Temperatures - Average: 49.59 °C, CSS: 50.70 °C, MSS 48.47 °C, UPA: 49.14 °C, DSS: 50.03 °C
[184430105170C51200] [192.168.34.171] [7.911] [system] [info] Cpu Usage - LeonOS 99.99%, LeonRT: 6.38%
[184430105170C51200] [192.168.34.171] [8.912] [system] [info] Memory Usage - DDR: 3.46 / 340.61 MiB, CMX: 2.08 / 2.50 MiB, LeonOS Heap: 51.81 / 77.48 MiB, LeonRT Heap: 3.71 / 41.35 MiB
[184430105170C51200] [192.168.34.171] [8.913] [system] [info] Temperatures - Average: 49.75 °C, CSS: 50.92 °C, MSS 48.92 °C, UPA: 48.92 °C, DSS: 50.25 °C
[184430105170C51200] [192.168.34.171] [8.913] [system] [info] Cpu Usage - LeonOS 99.99%, LeonRT: 6.31%
[184430105170C51200] [192.168.34.171] [9.914] [system] [info] Memory Usage - DDR: 3.46 / 340.61 MiB, CMX: 2.08 / 2.50 MiB, LeonOS Heap: 51.81 / 77.48 MiB, LeonRT Heap: 3.71 / 41.35 MiB
[184430105170C51200] [192.168.34.171] [9.914] [system] [info] Temperatures - Average: 50.20 °C, CSS: 51.80 °C, MSS 49.14 °C, UPA: 49.14 °C, DSS: 50.70 °C
[184430105170C51200] [192.168.34.171] [9.914] [system] [info] Cpu Usage - LeonOS 100.00%, LeonRT: 6.21%
[184430105170C51200] [192.168.34.171] [10.914] [system] [info] Memory Usage - DDR: 3.46 / 340.61 MiB, CMX: 2.08 / 2.50 MiB, LeonOS Heap: 51.81 / 77.48 MiB, LeonRT Heap: 3.71 / 41.35 MiB
[184430105170C51200] [192.168.34.171] [10.914] [system] [info] Temperatures - Average: 50.20 °C, CSS: 51.80 °C, MSS 49.37 °C, UPA: 49.37 °C, DSS: 50.25 °C
[... cut ...]
[2022-11-09 18:33:34.016] [debug] Log thread exception caught: Couldn't read data from stream: '__log' (X_LINK_ERROR)
[2022-11-09 18:33:34.016] [debug] Timesync thread exception caught: Couldn't read data from stream: '__timesync' (X_LINK_ERROR)
[2022-11-09 18:33:37.606] [debug] Device closed, 3605
Left FPS 32.37516594388514, Right FPS 32.37493902070482

```

**Expected behavior**
Ideally we would get closer theoretical maximum throughput on both USB and PoE devices.

**System information**
- OS: Linux Ubuntu 22.04

diablodale, 2022-12-08:
is this a dup of https://github.com/luxonis/depthai-core/issues/423?

moratom, 2022-12-09:
Indeed it is. Thanks @diablodale for noticing.

closed 2022-12-09 by moratom