luxonis/depthai-core #638: `UseHomographyRectification` API doesn't correct lens distortion on-device with 120 FoV lense.

developer-mayuan, 2022-11-10:
I have a question about `useHomographyRectification` API of StereoDepth node. According to the documentation (as shown in the figure below), the mesh data should be generated same as opencv's `initUndistortRectifyMap` function.
[image]

In my setup, I'm currently on lenses with 120 degree FoV, so a sparse mesh should be generated on the edge by it which corrects the lense distortion. But I actually got different rectification results between using this API and providing my own mesh data generated from a slighted modified version of [depthai's calibration script](https://github.com/luxonis/depthai/blob/main/calibrate.py). Could this caused because I use a different image resolution at runtime (400P) instead of the resolution used during calibration (800P)?

Rectified images using `HomographyRectification(false)` (**wrong result**):
[image]

Rectified images using customized mesh data generated from a modified version of calibration script (**correct result**):
[image]

The modification we made to the calibration script is in the `save_mesh_data()` function as following so that we can calibrate camera at 800P and rectify images at 400P at runtime:
```python
def create_save_mesh(self, decimation=2):  # , output_path):

        curr_path = Path(__file__).parent.resolve()
        print("Mesh path")
        print(curr_path)

        map_x_l, map_y_l = cv2.initUndistortRectifyMap(
            self.M1, self.d1, self.R1, self.M2, self.img_shape, cv2.CV_32FC1
        )
        map_x_r, map_y_r = cv2.initUndistortRectifyMap(
            self.M2, self.d2, self.R2, self.M2, self.img_shape, cv2.CV_32FC1
        )

        print("shape of maps")
        print(map_x_l.shape)
        print(map_y_l.shape)
        print(map_x_r.shape)
        print(map_y_r.shape)

        meshCellSize = 40
        mesh_left = []
        mesh_right = []

        for y in range(0, map_x_r.shape[0] + 1, meshCellSize):
            for x in range(0, map_x_r.shape[1] + 1, meshCellSize):
                x_ind = x
                y_ind = y
                if y == map_x_r.shape[0]:
                    y_ind = y - 1
                if x == map_x_r.shape[1]:
                    x_ind = x - 1
                # mesh.append((map_y[y_ind, x_ind], map_x[y_ind, x_ind]))
                mesh_left.append(
                    (
                        map_y_l[y_ind, x_ind] / decimation,
                        map_x_l[y_ind, x_ind] / decimation,
                    )
                )
                mesh_right.append(
                    (
                        map_y_r[y_ind, x_ind] / decimation,
                        map_x_r[y_ind, x_ind] / decimation,
                    )
                )

        mesh_left = np.array(mesh_left).astype(np.float32)
        mesh_right = np.array(mesh_right).astype(np.float32)
        left_mesh_fpath = str(curr_path) + "/../resources/left_mesh.calib"
        right_mesh_fpath = str(curr_path) + "/../resources/right_mesh.calib"
        mesh_left.tofile(left_mesh_fpath)
        mesh_right.tofile(right_mesh_fpath)
```

alex-luxonis, 2022-11-10:
Can you attach the device calibration data, we can have a look.
This example can be used to retrieve it:
https://github.com/luxonis/depthai-core/blob/main/examples/calibration/calibration_dump.cpp

developer-mayuan, 2022-11-10:
Hi @alex-luxonis Here is the calibration data I believe (we rotate the camera stream before calibration, so the image resolution is 800*1280):
```json
{
    "batchName": "",
    "batchTime": 0,
    "boardConf": "",
    "boardCustom": "",
    "boardName": "MY_BOARD",
    "boardOptions": 0,
    "boardRev": "R2M2E2",
    "cameraData": [
        [
            2,
            {
                "cameraType": 0,
                "distortionCoeff": [
                    -5.180891036987305,
                    8.956652641296387,
                    0.0003124542417936027,
                    1.3796477105643135e-05,
                    1.4717915058135986,
                    -4.844494819641113,
                    7.172787666320801,
                    4.7251362800598145,
                    0.0,
                    0.0,
                    0.0,
                    0.0,
                    0.0,
                    0.0
                ],
                "extrinsics": {
                    "rotationMatrix": [
                        [
                            0.0,
                            0.0,
                            0.0
                        ],
                        [
                            0.0,
                            0.0,
                            0.0
                        ],
                        [
                            0.0,
                            0.0,
                            0.0
                        ]
                    ],
                    "specTranslation": {
                        "x": 0.0,
                        "y": 0.0,
                        "z": 0.0
                    },
                    "toCameraSocket": -1,
                    "translation": {
                        "x": 0.0,
                        "y": 0.0,
                        "z": 0.0
                    }
                },
                "height": 1280,
                "intrinsicMatrix": [
                    [
                        601.024658203125,
                        0.0,
                        407.5490417480469
                    ],
                    [
                        0.0,
[... 137 lines cut ...]
                0.9999833106994629
            ]
        ],
        "rectifiedRotationRight": [
            [
                0.9987216591835022,
                0.009886703453958035,
                -0.04957118630409241
            ],
            [
                -0.009736666455864906,
                0.9999472498893738,
                0.0032672618981450796
            ],
            [
                0.04960087314248085,
                -0.002780427224934101,
                0.998765230178833
            ]
        ],
        "rightCameraSocket": 2
    },
    "version": 7
}
```

szabi-luxonis, 2022-11-11:
@developer-mayuan if you rotate the cameras during calibration, then you have to rotate them prior to sending them to the stereo node, and set input resolution explicitly.

developer-mayuan, 2022-11-11:
Hi @szabi-luxonis: Thanks for your suggestion, and yes, this is exactly what we have done for our runtime pipeline setup. We matched it with the pipeline we setup for calibration.

szabi-luxonis, 2022-11-16:
@developer-mayuan is to possible to share MRE? E.g. your pipeline.