luxonis/depthai-core #644: [BUG] Unable to specify custom toolchain file using relative path

camblomquist, 2022-11-17:
depthai-core uses its own custom toolchain file which is supposed to utilize the user-specified toolchain file if one is given. To do this, it first checks if said toolchain file exists.
```
    if(NOT EXISTS "@CMAKE_TOOLCHAIN_FILE@")
      message(
          FATAL_ERROR "File not found: '@CMAKE_TOOLCHAIN_FILE@'"
      )
    endif()
```
According to the CMake docs, "[EXISTS] is well-defined only for explicit full paths (a leading ~/ is not expanded as a home directory and is considered a relative path)". In my limited experience, the result is that I am unable to pass a relative file at all. `-DCMAKE_TOOLCHAIN_FILE=toolchain.cmake`, `./toolchain.cmake` (relative to project root), and `../../toolchain.cmake` (relative to generated toolchain) all cause CMake to trip on the above error message. If the full path is provided, then the generated toolchain is able to successfully find the user-specified file.

It's a minor inconvenience with a known workaround, but I would still expect to be able to pass a relative path for the toolchain.

Tested on CMake 3.16 and 3.24

themarpe, 2022-11-21:
Thanks for the report @camblomquist 

I've hit that as well, but didn't dig into the why just yet.

If you are aware of a workaround, I'd gladly take a look at a PR if you'd be up for opening one

camblomquist, 2022-11-21:
> If you are aware of a workaround, I'd gladly take a look at a PR if you'd be up for opening one

The user workaround is to specify the full path of the toolchain file.
I've gone ahead and made a PR for a proper fix.

closed 2022-11-22 by themarpe