luxonis/depthai-core #647: [BUG] {OAK-FFC-4P missing video stream from cam_d}

JorgenAndreasMo, 2022-11-23:
** Description**
OAK-FFC-4P set up with two OV9282 mono cameras at ports LEFT and RIGHT, one LCM48 rgb camera at port RGB and one IMX378 rgb camera at port CAM_D.

Using the demo script depthai-python/utilities/cam_test.py

**Minimal Reproducible Example**
OAK-FFC-4P configured with the above mentioned sensors. Running the depthai-python/utilities/cam_test.py demo script.

**Expected behavior**
Expected video stream from all image sensors

**System log **
Running the cam_test.py demo script results in the following behaviour:

First with the LCM48 at port RGB and IMX378 at port CAM_D 
It resulted in video stream from the two mono cameras and nothing from either RGB

`PS C:\depthai-python\utilities> python .\cam_test.py
Enabled cameras:
    rgb : color
   left : mono
  right : mono
   camd : color
DepthAI version: 2.19.0.0
DepthAI path: C:\Users\Lookout2\AppData\Roaming\Python\Python310\site-packages\depthai.cp310-win_amd64.pyd
[194430102198E31200] [1.6] [2.327] [ColorCamera(2)] [warning] LCM48 supported resolutions: 4_K / 4000X3000 / 5312X6000. Defaulting to 4_K
Connected cameras:
 -socket RGB   : LCM48  5312 x 6000 focus:auto  - COLOR
 -socket LEFT  : OV9282 1280 x  800 focus:fixed - COLOR MONO
 -socket RIGHT : OV9282 1280 x  800 focus:fixed - COLOR MONO
 -socket CAM_D : IMX378 4056 x 3040 focus:auto  - COLOR
USB speed: SUPER
IR drivers: []
Cam:      rgb           left          right         camd     [host | capture timestamp]
FPS:   0.00|  0.00  30.46| 30.02  30.46| 30.02   0.00|  0.00Traceback (most recent call last):
  File "C:\depthai-python\utilities\cam_test.py", line 282, in <module>
    key = cv2.waitKey(1)
KeyboardInterrupt`

Secondly with the LCM48 at port CAM_D and IMX378 at RGB 
It resulted in the following runtime error:
`PS C:\depthai-python\utilities> python .\cam_test.py
Enabled cameras:
    rgb : color
   left : mono
  right : mono
   camd : color
DepthAI version: 2.19.0.0
DepthAI path: C:\Users\Lookout2\AppData\Roaming\Python\Python310\site-packages\depthai.cp310-win_amd64.pyd
[194430102198E31200] [1.6] [2.316] [ColorCamera(8)] [warning] LCM48 supported resolutions: 4_K / 4000X3000 / 5312X6000. Defaulting to 4_K
Connected cameras:
 -socket RGB   : IMX378 4056 x 3040 focus:auto  - COLOR
 -socket LEFT  : OV9282 1280 x  800 focus:fixed - COLOR MONO
 -socket RIGHT : OV9282 1280 x  800 focus:fixed - COLOR MONO
 -socket CAM_D : LCM48  5312 x 6000 focus:auto  - COLOR
USB speed: SUPER
IR drivers: []
Cam:      rgb           left          right         camd     [host | capture timestamp]
FPS:   0.00|  0.00  30.40| 30.02  31.20| 30.02   0.00|  0.00[194430102198E31200] [1.6] [4.544] [system] [critical] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 00, address: 00000000' '0'
FPS:   0.00|  0.00  30.40| 30.02  31.20| 30.02   0.00|  0.00←[31mF: [global] [         0] [ThreadN] dispatcherResponseServe:925 no request for this response: XLINK_WRITE_RESP 1
←[0m
Traceback (most recent call last):
  File "C:\depthai-python\utilities\cam_test.py", line 257, in <module>
    pkt = q[c].tryGet()
RuntimeError: Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'rgb' (X_LINK_ERROR)'`

Attempring single camera configurations, both RGB sensors give video stream when connected to port RGB, but not when connected to port CAM_D. Program didn't crash, but I only recieved a video stream from the mono cameras.

I also got the following error the first time running the LCM48 sensor at port CAM_D:
`PS C:\depthai-python\utilities> python .\cam_test.py
Enabled cameras:
    rgb : color
   left : mono
  right : mono
   camd : color
DepthAI version: 2.19.0.0
DepthAI path: C:\Users\Lookout2\AppData\Roaming\Python\Python310\site-packages\depthai.cp310-win_amd64.pyd
[194430102198E31200] [1.6] [2.754] [ColorCamera(8)] [warning] LCM48 supported resolutions: 4_K / 4000X3000 / 5312X6000. Defaulting to 4_K
[194430102198E31200] [1.6] [2.754] [ColorCamera(2)] [error] Camera not detected on socket: 0
Connected cameras:
 -socket LEFT  : OV9282 1280 x  800 focus:fixed - COLOR MONO
 -socket RIGHT : OV9282 1280 x  800 focus:fixed - COLOR MONO
 -socket CAM_D : LCM48  5312 x 6000 focus:auto  - COLOR
USB speed: SUPER
IR drivers: []
Cam:      rgb           left          right         camd     [host | capture timestamp]
FPS:   0.00|  0.00   0.00|  0.00   0.00|  0.00   0.00|  0.00Traceback (most recent call last):
  File "C:\depthai-python\utilities\cam_test.py", line 259, in <module>
    fps_host[c].update()
  File "C:\depthai-python\utilities\cam_test.py", line 134, in update
    if count > 0: self.fps = count / (timestamp - self.dq[0])
ZeroDivisionError: float division by zero`

themarpe, 2022-11-26:
@JorgenAndreasMo 
Thanks for the report - which library version are you using? Do you mind sending the output if you set DEPTHAI_DEBUG=1 env variable?

CC: @alex-luxonis

JorgenAndreasMo, 2022-11-29:
Hi, I just pulled from Github so it should be the newest library. It prints DepthAI version: 2.19.0.0 in the terminal

Please see the attached log file

[depthai-core_647_logFile.txt](https://github.com/luxonis/depthai-core/files/10117365/depthai-core_647_logFile.txt)

themarpe, 2022-12-01:
Thanks. Can reproduce with rgb_preview.py and setting to CAM_D /w 48MP on CAM_A and 378 on CAM_D or vice versa.

CC: @alex-luxonis - do you mind checking this one out?

JorgenAndreasMo, 2023-01-12:
Hi, checking in again. Have you had any success in solving this issue yet? @themarpe

gajena, 2023-01-28:
@themarpe @alex-luxonis any updates on this, or is there an older version where this one works? I am having an issue where I can only run one colour camera on rgb port, other ports are not working with the colour camera.

alex-luxonis, 2023-01-31:
A fix was pushed to `release_2.20.2` (PR https://github.com/luxonis/depthai-core/pull/694). We'll merge that soon for the next release, but for now you can build from there.

gajena, 2023-01-31:
Checked out to this branch, it works now. Thanks for the quick patch.

themarpe, 2023-02-01:
Thanks all, closing this issue now, as v2.20.2 is released :tada:

closed 2023-02-01 by themarpe