luxonis/depthai-core #650: linker error in examples flash_bootloader and flash_user_bootloader with XLink

soeroesg, 2022-11-29:
**Describe the bug**
The targets `flash_bootloader` and `flash_user_bootloader` fail with the following linker error:
```
flash_bootloader.obj : error LNK2019: unresolved external symbol XLinkProtocolToStr referenced in function main [C:\dev\build\depthai-core\examples\flash_bootloader.vcxproj]
flash_user_bootloader.obj : error LNK2019: unresolved external symbol XLinkProtocolToStr referenced in function main [C:\dev\build\depthai-core\examples\flash_user_bootloader.vcxproj]
```
These targets are only built if `DEPTHAI_BUILD_EXAMPLES=ON`. I am using Visual Studio 2022 17.4.1.
I assume that CMake fails to link the XLink library with these examples.

**Minimal Reproducible Example**
```
set SOURCE_DIR="C:\dev\src\depthai-core"
set BUILD_DIR="C:\dev\build\depthai-core"
mkdir %BUILD_DIR%
cd %BUILD_DIR%

cmake.exe %SOURCE_DIR% ^
  -G"Visual Studio 17 2022" ^
  -A"x64" ^
  -DDEPTHAI_OPENCV_SUPPORT=ON ^
  -DDEPTHAI_BUILD_TESTS=OFF ^
  -DDEPTHAI_BUILD_DOCS=OFF ^
  -DDEPTHAI_BUILD_EXAMPLES=ON ^
  -DBUILD_SHARED_LIBS=ON ^
  -DHUNTER_KEEP_PACKAGE_SOURCES=ON ^
  -DHUNTER_GATE_ROOT="%BUILD_DIR%" ^
  -DHUNTER_ROOT="%BUILD_DIR%"
```

**Expected behavior**
All examples should build without errors.

themarpe, 2022-12-01:
Hi @soeroesg .

Thanks for the report. This stems from a mixed usage of "public headers" or XLink and private linking.
Proper solution is to make XLink strictly shared library and then make XLink have PUBLIC linking in CMakelists.txt

Quicker solution (and preferable if we'd like to completely hide/make XLink a private dependecy) is to remove the usage of those two function in downstream examples.

Do you mind opening a PR with the quicker solution in place? It'd be adequate for now and I'll take a look into digging into the public XLink option some time

diablodale, 2022-12-08:
I have this bug on Windows. Worked in 2.17.0 and fails with 2.19.1
My intention is a single depthai-cored.dll (which within it contains XLink) and a single libusb.dll (because of its license).
No separate xlink.dll.

closed 2023-01-21 by themarpe