luxonis/depthai-core #651: Image Manip Node max CPU Fatal error on MSS CPU: trap: 00, address: 00000000' '0'

PeterQFR, 2022-11-30:
Possibly related to https://github.com/luxonis/depthai-python/issues/625

I am running a spatialImageDetector node and as my cameras are mounted 90 degrees to normal I rotate the depth and rgb images 90 degrees using a imagemanip node. when doing so the LeonRT maxes out to 100% and a CPU trap is triggered.

In my debug output from the node:

```
[INFO] [1669770818.307067979] [depthAI_node]: Initialising DepthAI
[INFO] [1669770818.315620593] [depthAI_node]: Comms Setup
[INFO] [1669770818.315639026] [depthAI_node]: Callbacks Setup
[INFO] [1669770818.315662691] [depthAI_node]: Polling Queues Setup
[1944301051BBED1200] [3.1] [2.926] [system] [info] Memory Usage - DDR: 133.67 / 340.61 MiB, CMX: 2.49 / 2.50 MiB, LeonOS Heap: 54.55 / 77.48 MiB, LeonRT Heap: 5.90 / 41.35 MiB
[1944301051BBED1200] [3.1] [2.926] [system] [info] Temperatures - Average: 44.01 °C, CSS: 44.87 °C, MSS 43.96 °C, UPA: 44.64 °C, DSS: 42.58 °C
[1944301051BBED1200] [3.1] [2.926] [system] [info] Cpu Usage - LeonOS 57.11%, LeonRT: 54.64%
[1944301051BBED1200] [3.1] [3.927] [system] [info] Memory Usage - DDR: 133.67 / 340.61 MiB, CMX: 2.49 / 2.50 MiB, LeonOS Heap: 54.55 / 77.48 MiB, LeonRT Heap: 5.90 / 41.35 MiB
[1944301051BBED1200] [3.1] [3.927] [system] [info] Temperatures - Average: 44.47 °C, CSS: 44.87 °C, MSS 44.18 °C, UPA: 44.87 °C, DSS: 43.96 °C
[1944301051BBED1200] [3.1] [3.927] [system] [info] Cpu Usage - LeonOS 6.13%, LeonRT: 100.00%
[1944301051BBED1200] [3.1] [4.928] [system] [info] Memory Usage - DDR: 133.67 / 340.61 MiB, CMX: 2.49 / 2.50 MiB, LeonOS Heap: 54.55 / 77.48 MiB, LeonRT Heap: 5.90 / 41.35 MiB
[1944301051BBED1200] [3.1] [4.928] [system] [info] Temperatures - Average: 44.41 °C, CSS: 45.77 °C, MSS 44.64 °C, UPA: 44.87 °C, DSS: 42.35 °C
[1944301051BBED1200] [3.1] [4.928] [system] [info] Cpu Usage - LeonOS 5.91%, LeonRT: 100.00%
[1944301051BBED1200] [3.1] [5.929] [system] [info] Memory Usage - DDR: 133.67 / 340.61 MiB, CMX: 2.49 / 2.50 MiB, LeonOS Heap: 54.55 / 77.48 MiB, LeonRT Heap: 5.90 / 41.35 MiB
[1944301051BBED1200] [3.1] [5.929] [system] [info] Temperatures - Average: 44.47 °C, CSS: 45.32 °C, MSS 44.64 °C, UPA: 44.64 °C, DSS: 43.27 °C
[1944301051BBED1200] [3.1] [5.929] [system] [info] Cpu Usage - LeonOS 5.76%, LeonRT: 100.00%
[1944301051BBED1200] [3.1] [7.694] [system] [critical] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 00, address: 00000000' '0'
```

If I disable using the ImageManipNodes and just pipe depth and rgb preview directly to the spatial detector I do not get an issue with the CPU.

The code works on earlier commit 02c304803aabaced594e8173152c6dedb260cd06

but this error appears on a newer commit on the main branch 690b6a63

I am use a DepthAI Pro.

Example Code that doesn't work:
NB depth_temp_ is a depth node initialised elsewhere.
```
       xoutRGB_=  pipeline.create<dai::node::XLinkOut>();
        xoutRGB_->setStreamName("rgb_preview");
        xoutRGB_->input.setBlocking(false);
        xoutRGB_->input.setQueueSize(1);
        rgb_ = pipeline.create<dai::node::ColorCamera>();
        rgb_->setFps(1.0f);
        rgb_->setResolution(dai::ColorCameraProperties::SensorResolution::THE_1080_P);
        rgb_->setInterleaved(false);
        rgb_->setColorOrder(dai::ColorCameraProperties::ColorOrder::BGR);
        
        rgb_->setPreviewSize(416, 416);

        xoutNN_ = pipeline.create<dai::node::XLinkOut>();
        xoutNN_->setStreamName("detections");

        xoutBBDepthMapping_ = pipeline.create<dai::node::XLinkOut>();
        xoutBBDepthMapping_->setStreamName("boundingboxes");

        spatialDetector_ = pipeline.create<dai::node::YoloSpatialDetectionNetwork>();

        spatialDetector_->setConfidenceThreshold(CONFIDENCE_THRESHOLD);
        spatialDetector_->input.setBlocking(false);
        spatialDetector_->inputDepth.setBlocking(false);
         
        spatialDetector_->setBoundingBoxScaleFactor(0.5);
        spatialDetector_->setDepthLowerThreshold(100);
        spatialDetector_->setDepthUpperThreshold(20000);
        spatialDetector_->setNumClasses(80);
        spatialDetector_->setCoordinateSize(4);
        spatialDetector_->setAnchors({10, 14, 23, 27,37, 58, 81, 82, 135, 169, 344, 319});
        spatialDetector_->setAnchorMasks({{"side26", {1, 2, 3}}, {"side13", {3, 4, 5}}});
        spatialDetector_->setIouThreshold(0.5f);
        spatialDetector_->setBlobPath(path_);

      //Comments here turn on or off rotation of depth image to see if that affects CPU.    
       // depthRotation_ = pipeline.create<dai::node::ImageManip>();
        colorRotation_ = pipeline.create<dai::node::ImageManip>();

       /* depthRotation_->initialConfig.setRotationRadians(M_PI_2);
        depthRotation_->initialConfig.setFrameType(dai::ImgFrame::Type::RAW16);
        depthRotation_->inputImage.setBlocking(false);
        depthRotation_->inputImage.setQueueSize(1);
*/
        colorRotation_->initialConfig.setRotationRadians(M_PI_2);
        colorRotation_->initialConfig.setFrameType(dai::ImgFrame::Type::BGR888p);
        colorRotation_->inputImage.setBlocking(false);
        colorRotation_->inputImage.setQueueSize(1);

        //depth_temp_->depth.link(depthRotation_->inputImage);
        rgb_->preview.link(colorRotation_->inputImage);

        colorRotation_->out.link(spatialDetector_->input);
        //depthRotation_->out.link(spatialDetector_->inputDepth);
        depth_temp_->depth.link(spatialDetector_->inputDepth);

        spatialDetector_->input.setBlocking(false);
        spatialDetector_->input.setQueueSize(1);
        spatialDetector_->inputDepth.setBlocking(false);
        spatialDetector_->inputDepth.setQueueSize(1); 
        spatialDetector_->boundingBoxMapping.link(xoutBBDepthMapping_->input);
        spatialDetector_->out.link(xoutNN_->input);
        spatialDetector_->passthrough.link(xoutRGB_->input);
```

PeterQFR, 2022-12-01:
@themarpe Testing on the latest develop: 8f0214fa6c50b10dd1982ddaf014803cf088645
and the depthai-bootloader-shared b287ecbacd3b0c963b5dfcf95767123b0c143b57
depthai-shared 3404950cb5944c64daa7899f065d07d8ea8331c7

I can't run the camera board at all:

```
terminate called after throwing an instance of 'dai::XLinkWriteError'
  what():  Couldn't write data to stream: '__rpc_main' (X_LINK_ERROR)
```
exits on initialisation.

PeterQFR, 2022-12-01:
On the same commits if I enable the image manipulation in the pipeline I dont get anything other than 

```
terminate called without an active exception
terminate called recursively
```
or
```
terminate called after throwing an instance of 'dai::XLinkWriteError'
  what():  Couldn't write data to stream: '__stream_asset_storage' (X_LINK_ERROR)terminate called recursively
```

or in rare cases 
```
[1944301051BBED1200] [3.1] [2.075] [MonoCamera(2)] [error] Camera not detected on socket: 1
[1944301051BBED1200] [3.1] [2.075] [ColorCamera(11)] [error] Camera not detected on socket: 0
[1944301051BBED1200] [3.1] [2.076] [SpatialDetectionNetwork(14)] [warning] 'boundingBoxMapping' output is deprecated! Use 'detection.boundingBoxMapping'.
[1944301051BBED1200] [3.1] [2.076] [MonoCamera(3)] [error] Camera not detected on socket: 2
[1944301051BBED1200] [3.1] [2.859] [SpatialDetectionNetwork(14)] [warning] Network compiled for 6 shaves, maximum available 9, compiling for 4 shaves likely will yield in better performance
Completing 
[INFO] [1669937404.336725913] [depthAI_node]: Initialising DepthAI
[INFO] [1669937404.346109201] [depthAI_node]: Comms Setup
[INFO] [1669937404.346129049] [depthAI_node]: Callbacks Setup
[INFO] [1669937404.346155017] [depthAI_node]: Polling Queues Setup
[1944301051BBED1200] [3.1] [3.483] [system] [info] Memory Usage - DDR: 133.67 / 340.61 MiB, CMX: 2.48 / 2.50 MiB, LeonOS Heap: 33.76 / 77.46 MiB, LeonRT Heap: 5.23 / 41.34 MiB
[1944301051BBED1200] [3.1] [3.483] [system] [info] Temperatures - Average: 35.94 °C, CSS: 37.24 °C, MSS 35.35 °C, UPA: 36.53 °C, DSS: 34.64 °C
[1944301051BBED1200] [3.1] [3.483] [system] [info] Cpu Usage - LeonOS 59.21%, LeonRT: 10.49%
[1944301051BBED1200] [3.1] [4.499] [system] [info] Memory Usage - DDR: 133.67 / 340.61 MiB, CMX: 2.48 / 2.50 MiB, LeonOS Heap: 34.15 / 77.46 MiB, LeonRT Heap: 5.23 / 41.34 MiB
[1944301051BBED1200] [3.1] [4.499] [system] [info] Temperatures - Average: 36.18 °C, CSS: 37.24 °C, MSS 35.35 °C, UPA: 36.30 °C, DSS: 35.83 °C
[1944301051BBED1200] [3.1] [4.499] [system] [info] Cpu Usage - LeonOS 100.00%, LeonRT: 0.08%
[1944301051BBED1200] [3.1] [5.522] [system] [info] Memory Usage - DDR: 133.67 / 340.61 MiB, CMX: 2.48 / 2.50 MiB, LeonOS Heap: 34.19 / 77.46 MiB, LeonRT Heap: 5.23 / 41.34 MiB
[1944301051BBED1200] [3.1] [5.522] [system] [info] Temperatures - Average: 36.89 °C, CSS: 37.24 °C, MSS 36.53 °C, UPA: 37.48 °C, DSS: 36.30 °C
[1944301051BBED1200] [3.1] [5.522] [system] [info] Cpu Usage - LeonOS 100.00%, LeonRT: 0.08%
[1944301051BBED1200] [3.1] [6.552] [system] [info] Memory Usage - DDR: 133.67 / 340.61 MiB, CMX: 2.48 / 2.50 MiB, LeonOS Heap: 34.23 / 77.46 MiB, LeonRT Heap: 5.23 / 41.34 MiB

```
I do have logging enabled. It seems somewhat unstable.

PeterQFR, 2022-12-02:
I've deleted a few previous posts of mine, it seems that I was using a defective USB cable., By changing to a known good cable I retested on the develop branch 
@themarpe Testing on the latest develop: 8f0214fa6c50b10dd1982ddaf014803cf088645
and the depthai-bootloader-shared b287ecbacd3b0c963b5dfcf95767123b0c143b57
depthai-shared 3404950cb5944c64daa7899f065d07d8ea8331c7

With a known good USB cable and without imagemanipulation nodes I get cameras unable to be detected:
```
tarting Front Slam Sensor
[1944301051BBED1200] [1.4] [1.554] [MonoCamera(2)] [error] Camera not detected on socket: 1
[1944301051BBED1200] [1.4] [1.554] [SpatialDetectionNetwork(14)] [warning] 'boundingBoxMapping' output is deprecated! Use 'detection.boundingBoxMapping'.
[1944301051BBED1200] [1.4] [1.554] [MonoCamera(3)] [error] Camera not detected on socket: 2
[1944301051BBED1200] [1.4] [1.555] [ColorCamera(11)] [error] Camera not detected on socket: 0
[1944301051BBED1200] [1.4] [2.342] [SpatialDetectionNetwork(14)] [warning] Network compiled for 6 shaves, maximum available 9, compiling for 4 shaves likely will yield in better performance
Completing 
[INFO] [1669940641.843008040] [depthAI_node]: Initialising DepthAI
[INFO] [1669940641.882444183] [depthAI_node]: Comms Setup
[INFO] [1669940641.882492867] [depthAI_node]: Callbacks Setup
[INFO] [1669940641.882535534] [depthAI_node]: Polling Queues Setup
[1944301051BBED1200] [1.4] [3.351] [system] [info] Memory Usage - DDR: 129.67 / 340.61 MiB, CMX: 2.43 / 2.50 MiB, LeonOS Heap: 34.06 / 77.46 MiB, LeonRT Heap: 5.16 / 41.34 MiB
[1944301051BBED1200] [1.4] [3.351] [system] [info] Temperatures - Average: 37.36 °C, CSS: 38.18 °C, MSS 37.01 °C, UPA: 37.94 °C, DSS: 36.30 °C
[1944301051BBED1200] [1.4] [3.351] [system] [info] Cpu Usage - LeonOS 98.34%, LeonRT: 9.35%
[1944301051BBED1200] [1.4] [4.376] [system] [info] Memory Usage - DDR: 129.67 / 340.61 MiB, CMX: 2.43 / 2.50 MiB, LeonOS Heap: 34.15 / 77.46 MiB, LeonRT Heap: 5.16 / 41.34 MiB
[1944301051BBED1200] [1.4] [4.376] [system] [info] Temperatures - Average: 37.71 °C, CSS: 39.35 °C, MSS 37.48 °C, UPA: 37.24 °C, DSS: 36.77 °C
[1944301051BBED1200] [1.4] [4.376] [system] [info] Cpu Usage - LeonOS 100.00%, LeonRT: 9.35%
[1944301051BBED1200] [1.4] [5.414] [system] [info] Memory Usage - DDR: 129.67 / 340.61 MiB, CMX: 2.43 / 2.50 MiB, LeonOS Heap: 34.19 / 77.46 MiB, LeonRT Heap: 5.16 / 41.34 MiB
[1944301051BBED1200] [1.4] [5.414] [system] [info] Temperatures - Average: 37.59 °C, CSS: 37.71 °C, MSS 37.71 °C, UPA: 37.94 °C, DSS: 37.01 °C
[1944301051BBED1200] [1.4] [5.414] [system] [info] Cpu Usage - LeonOS 100.00%, LeonRT: 0.09%

```

If I enable image manipulation nodes:
```
Starting Front Slam Sensor
[1944301051BBED1200] [1.4] [1.552] [ColorCamera(11)] [error] Camera not detected on socket: 0
[1944301051BBED1200] [1.4] [1.553] [MonoCamera(3)] [error] Camera not detected on socket: 2
[1944301051BBED1200] [1.4] [1.553] [SpatialDetectionNetwork(14)] [warning] 'boundingBoxMapping' output is deprecated! Use 'detection.boundingBoxMapping'.
[1944301051BBED1200] [1.4] [1.554] [MonoCamera(2)] [error] Camera not detected on socket: 1
[1944301051BBED1200] [1.4] [2.337] [SpatialDetectionNetwork(14)] [warning] Network compiled for 6 shaves, maximum available 9, compiling for 4 shaves likely will yield in better performance
Completing 
[INFO] [1669940897.375231274] [depthAI_node]: Initialising DepthAI
[INFO] [1669940897.397930092] [depthAI_node]: Comms Setup
[INFO] [1669940897.397996655] [depthAI_node]: Callbacks Setup
[INFO] [1669940897.398026011] [depthAI_node]: Polling Queues Setup
[1944301051BBED1200] [1.4] [3.341] [system] [info] Memory Usage - DDR: 137.67 / 340.61 MiB, CMX: 2.48 / 2.50 MiB, LeonOS Heap: 34.06 / 77.46 MiB, LeonRT Heap: 5.30 / 41.34 MiB
[1944301051BBED1200] [1.4] [3.341] [system] [info] Temperatures - Average: 37.06 °C, CSS: 37.94 °C, MSS 36.53 °C, UPA: 37.24 °C, DSS: 36.53 °C
[1944301051BBED1200] [1.4] [3.341] [system] [info] Cpu Usage - LeonOS 97.42%, LeonRT: 10.51%
[1944301051BBED1200] [1.4] [4.363] [system] [info] Memory Usage - DDR: 137.67 / 340.61 MiB, CMX: 2.48 / 2.50 MiB, LeonOS Heap: 34.15 / 77.46 MiB, LeonRT Heap: 5.30 / 41.34 MiB
[1944301051BBED1200] [1.4] [4.363] [system] [info] Temperatures - Average: 37.59 °C, CSS: 38.88 °C, MSS 37.24 °C, UPA: 37.48 °C, DSS: 36.77 °C
[1944301051BBED1200] [1.4] [4.363] [system] [info] Cpu Usage - LeonOS 100.00%, LeonRT: 0.07%
[1944301051BBED1200] [1.4] [5.399] [system] [info] Memory Usage - DDR: 137.67 / 340.61 MiB, CMX: 2.48 / 2.50 MiB, LeonOS Heap: 34.19 / 77.46 MiB, LeonRT Heap: 5.30 / 41.34 MiB
[1944301051BBED1200] [1.4] [5.399] [system] [info] Temperatures - Average: 38.00 °C, CSS: 39.11 °C, MSS 37.71 °C, UPA: 37.94 °C, DSS: 37.24 °C
[1944301051BBED1200] [1.4] [5.399] [system] [info] Cpu Usage - LeonOS 100.00%, LeonRT: 0.09%
[1944301051BBED1200] [1.4] [6.438] [system] [info] Memory Usage - DDR: 137.67 / 340.61 MiB, CMX: 2.48 / 2.50 MiB, LeonOS Heap: 34.22 / 77.46 MiB, LeonRT Heap: 5.30 / 41.34 MiB
```

themarpe, 2022-12-02:
@PeterQFR 
Do you mind running with DEPTHAI_DEBUG=1 and post the whole output as well as running the `calibration_dump` example.

PeterQFR, 2022-12-02:
Here is running my code with DEPTHAI_DEBUG=1
```
[2022-12-02 20:07:02.005] [info] DEPTHAI_DEBUG enabled, lowered DEPTHAI_LEVEL to 'debug'
[2022-12-02 20:07:02.005] [debug] Library information - version: 2.19.1, commit: 48f0214fa6c50b10dd1982ddaf014803cf088645 from 2022-12-01 17:35:13 +0100, build: 2022-12-02 10:04:02 +0000
[2022-12-02 20:07:02.014] [debug] Initialize - finished
Starting Front Slam Sensor
[2022-12-02 20:07:02.120] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.22.tar.xz' open: 5ms, archive read: 108ms
[2022-12-02 20:07:02.283] [debug] Device - OpenVINO version: 2022.1
[2022-12-02 20:07:02.283] [debug] Device - BoardConfig: {"camera":[],"emmc":null,"gpio":[],"logDevicePrints":true,"logPath":null,"logSizeMax":null,"logVerbosity":null,"network":{"mtu":0,"xlinkTcpNoDelay":true},"pcieInternalClock":null,"sysctl":[],"uart":[],"usb":{"flashBootedPid":63037,"flashBootedVid":999,"maxSpeed":4,"pid":63035,"vid":999},"usb3PhyInternalClock":null,"watchdogInitialDelayMs":null,"watchdogTimeoutMs":null} 
libnop:
0000: b9 0f b9 05 81 e7 03 81 3b f6 81 e7 03 81 3d f6 04 b9 02 00 01 ba 00 be be bb 00 bb 00 be be be
0020: be be be 01 bb 00
[2022-12-02 20:07:02.528] [debug] Resources - Archive 'depthai-device-fwp-c3891fbe461a569a21fd0dc713d2dced535fbd8d.tar.xz' open: 5ms, archive read: 516ms
[2022-12-02 20:07:03.782] [debug] Searching for booted device: DeviceInfo(name=1.4, mxid=1944301051BBED1200, X_LINK_BOOTED, X_LINK_USB_VSC, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[1944301051BBED1200] [1.4] [1.345] [system] [warning] PRINT:LeonCss: BootloaderConfig.options1 checksum doesn't match. Is: 0x10304419 should be: 0xC7758E57
BootloaderConfig.options2 checksum doesn't match. Is: 0x5ABFBF6E should be: 0x0255D960
GPIO boot mode 0x3, interface SPI_MASTER_EFF
Setting aons(0..4) back to boot from flash (offset = 0)
====ENABLE WATCHDOG====1
initial keepalive, countdown: 10
PLL0: 700000 AUX_IO0: 24000 AUX_IO1: 24000 MCFG: 24000 MECFG: 24000
Board init ret 3
eeprom configuration version: 55AA0007
Reading VERSION 7 --- ->
eeprom configuration load from user area, status: 0
Reading VERSION 7 --- ->
Didn't find any boards matching the eeprom data
Board options: 00000004
--> brdInit ...
brdInitAuxDevices: Error: SC = 27: io_initialize expander_cam_gpios_1 [OK]

spi_N25Q_init: Flash JEDEC ID: c2 25 3a
SR CR: 40 07
QUAD mode already enabled. Dummy cfg: 0x00
Opening bus for IR driver: 1
Failed to probe IR driver LM3644
Opening bus for IR driver: 2
LM3644 detected, ID = 0x02
===== IR write bus 2: 0x07 = 0x89
===== IR write bus 2: 0x02 = 0x01
===== IR write bus 2: 0x03 = 0x00
===== IR write bus 2: 0x04 = 0x00
===== IR write bus 2: 0x05 = 0x00
===== IR write bus 2: 0x06 = 0x00
===== IR write bus 2: 0x07 = 0x09
===== IR write bus 2: 0x08 = 0x1a
===== IR write bus 2: 0x09 = 0x08
===== IR write bus 2: 0x01 = 0x24
Opening bus for IR driver: 3
Failed to probe IR driver LM3644
Closing EEPROm!
MyriaX board configuration
pll0 frequency: 700000, ref0 frequency: 24000
Is booted from flash by bootloader: 0
Networking not available...
=== Enumerating on socket: Cam_A / RGB / Center
initial keepalive, countdown: 9
=== Enumerating on socket: Cam_B / Left
Probe failed I2C2 0x63 reg 300a: expected 7750, read 0000
Probe failed I2C2 0x63 reg 300a: expected 7750, read 0000
=== Enumerating on socket: Cam_C / Right
Probe failed I2C2 0x63 reg 300a: expected 7750, read 0000
Probe failed I2C2 0x63 reg 300a: expected 7750, read 0000
Initializing XLink...
UsbPumpVscAppI_Event: 5 VSC2_EVENT_ATTACH
UsbPumpVscAppI_Event: 4 VSC2_EVENT_RESET
UsbPumpVscAppI_Event: 4 VSC2_EVENT_RESET
UsbPumpVscAppI_Event: 0 VSC2_EVENT_INTERFACE_UP
Done!
Usb connection speed: Super - USB 3.0
[... 148 lines cut ...]
[1944301051BBED1200] [1.4] [9.693] [system] [info] Cpu Usage - LeonOS 100.00%, LeonRT: 0.07%
[1944301051BBED1200] [1.4] [10.743] [system] [info] Memory Usage - DDR: 110.21 / 340.61 MiB, CMX: 2.48 / 2.50 MiB, LeonOS Heap: 34.85 / 77.46 MiB, LeonRT Heap: 5.30 / 41.34 MiB
[1944301051BBED1200] [1.4] [10.743] [system] [info] Temperatures - Average: 35.82 °C, CSS: 36.53 °C, MSS 35.35 °C, UPA: 36.06 °C, DSS: 35.35 °C
[1944301051BBED1200] [1.4] [10.743] [system] [info] Cpu Usage - LeonOS 97.09%, LeonRT: 0.09%
[1944301051BBED1200] [1.4] [11.793] [system] [info] Memory Usage - DDR: 110.21 / 340.61 MiB, CMX: 2.48 / 2.50 MiB, LeonOS Heap: 34.85 / 77.46 MiB, LeonRT Heap: 5.30 / 41.34 MiB
[1944301051BBED1200] [1.4] [11.793] [system] [info] Temperatures - Average: 35.47 °C, CSS: 36.53 °C, MSS 34.88 °C, UPA: 36.30 °C, DSS: 34.16 °C
[1944301051BBED1200] [1.4] [11.793] [system] [info] Cpu Usage - LeonOS 32.04%, LeonRT: 0.07%
[1944301051BBED1200] [1.4] [12.864] [system] [info] Memory Usage - DDR: 110.21 / 340.61 MiB, CMX: 2.48 / 2.50 MiB, LeonOS Heap: 34.85 / 77.46 MiB, LeonRT Heap: 5.30 / 41.34 MiB
[1944301051BBED1200] [1.4] [12.864] [system] [info] Temperatures - Average: 35.88 °C, CSS: 36.77 °C, MSS 36.30 °C, UPA: 36.30 °C, DSS: 34.16 °C
[1944301051BBED1200] [1.4] [12.864] [system] [info] Cpu Usage - LeonOS 31.96%, LeonRT: 0.09%
[1944301051BBED1200] [1.4] [13.923] [system] [info] Memory Usage - DDR: 110.21 / 340.61 MiB, CMX: 2.48 / 2.50 MiB, LeonOS Heap: 34.85 / 77.46 MiB, LeonRT Heap: 5.30 / 41.34 MiB
[1944301051BBED1200] [1.4] [13.923] [system] [info] Temperatures - Average: 35.82 °C, CSS: 36.53 °C, MSS 35.59 °C, UPA: 36.30 °C, DSS: 34.88 °C
[1944301051BBED1200] [1.4] [13.923] [system] [info] Cpu Usage - LeonOS 34.67%, LeonRT: 0.07%
[... cut ...]
[1944301051BBED1200] [1.4] [17.093] [system] [info] Memory Usage - DDR: 110.21 / 340.61 MiB, CMX: 2.48 / 2.50 MiB, LeonOS Heap: 34.85 / 77.46 MiB, LeonRT Heap: 5.30 / 41.34 MiB
[1944301051BBED1200] [1.4] [17.093] [system] [info] Temperatures - Average: 35.35 °C, CSS: 37.01 °C, MSS 34.88 °C, UPA: 35.11 °C, DSS: 34.40 °C
[1944301051BBED1200] [1.4] [17.093] [system] [info] Cpu Usage - LeonOS 32.14%, LeonRT: 0.07%
[1944301051BBED1200] [1.4] [18.152] [system] [info] Memory Usage - DDR: 110.21 / 340.61 MiB, CMX: 2.48 / 2.50 MiB, LeonOS Heap: 34.85 / 77.46 MiB, LeonRT Heap: 5.30 / 41.34 MiB

```

PeterQFR, 2022-12-02:
By `calibration_dump` you mean `calibration_reader`?

Here is the output from that
```
./calibration_reader
Intrinsics from defaultIntrinsics function:
[[806.695251, 0.000000, 653.432068]
[0.000000, 806.695251, 390.269470]
[0.000000, 0.000000, 1.000000]]

Width: 1280
Height: 800
Stereo baseline distance: 7.5 cm
Mono FOV from camera specs: 71.86, calculated FOV: 76.6831
Intrinsics from getCameraIntrinsics function full resolution:
[[806.695251, 0.000000, 653.432068]
[0.000000, 806.695251, 390.269470]
[0.000000, 0.000000, 1.000000]]

Intrinsics from getCameraIntrinsics function 1280 x 720:
[[806.695251, 0.000000, 653.432068]
[0.000000, 806.695251, 350.269470]
[0.000000, 0.000000, 1.000000]]

Intrinsics from getCameraIntrinsics function 720 x 450:
[[453.766083, 0.000000, 367.555542]
[0.000000, 453.766083, 219.526581]
[0.000000, 0.000000, 1.000000]]

Intrinsics from getCameraIntrinsics function 600 x 1280:
[[378.138397, 0.000000, 306.296295]
[0.000000, 378.138397, 635.438843]
[0.000000, 0.000000, 1.000000]]

Extrinsics from left->right test:
[[0.999655, -0.004377, -0.025897, -7.542802]
[0.004335, 0.999989, -0.001683, 0.016673]
[0.025904, 0.001570, 0.999663, -0.102877]
[0.000000, 0.000000, 0.000000, 1.000000]]

Extrinsics from right->left test:
[[0.999655, 0.004335, 0.025904, 7.542792]
[-0.004377, 0.999989, 0.001570, -0.049530]
[-0.025897, -0.001683, 0.999663, -0.092465]
[0.000000, 0.000000, 0.000000, 1.000000]]

Extrinsics from right->rgb test:
[[0.999853, 0.006298, 0.015950, 3.724179]
[-0.006351, 0.999974, 0.003262, 0.006275]
[-0.015929, -0.003363, 0.999867, -0.157903]
[0.000000, 0.000000, 0.000000, 1.000000]]

Extrinsics from rgb->right test:
[[0.999853, -0.006351, -0.015929, -3.726107]
[0.006298, 0.999974, -0.003363, -0.030260]
[0.015950, 0.003262, 0.999867, 0.098462]
[0.000000, 0.000000, 0.000000, 1.000000]]

Extrinsics from left->rgb test:
[[0.999948, 0.002005, -0.009964, -3.815846]
[-0.001989, 0.999997, 0.001649, 0.039360]
[0.009967, -0.001629, 0.999949, -0.164246]
[0.000000, 0.000000, 0.000000, 1.000000]]

```

themarpe, 2022-12-02:
@PeterQFR 
Thanks, I'll try to recreate on a Pro device tomorrow - though if you can provide an MRE to go along that'd be great. 
Note that somewhy the LeonOS CPU usage is 100%

PeterQFR, 2022-12-02:
@themarpe Okay I was putting together a MRE, and in doing so, I was going to use the spatial_tiny_yolo_v3 as a basis. After rebuilding that I was able to run that program. I then tried to rerun my program, and voilia it started working. It seems like a firmware update issue. I'm not sure how DepthAI does firmware updates, but I had clean built earlier in the day, with no improvments. When Building the examples I was getting build failures below. 

```
stderr: depthai                                 
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[3]: *** [examples/CMakeFiles/flash_boot_header.dir/build.make:63: examples/CMakeFiles/flash_boot_header.dir/bootloader/flash_boot_header.cpp.o] Error 1
make[2]: *** [CMakeFiles/Makefile2:3363: examples/CMakeFiles/flash_boot_header.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:141: all] Error 2
---
Failed   <<< depthai [48.4s, exited with code 2]
```
what do you think?

PeterQFR, 2022-12-02:
How is the firmware updated?

PeterQFR, 2022-12-02:
I'm pretty sure that because of the above build failure, my code was not being rebuilt from its version that failed.

PeterQFR, 2022-12-06:
Okay, I'll check depth but I think for that example it was working and has worked since. 

However, today I get the cameras not being detected. Output below from DEPTHAI_DEBUG=1

```
[2022-12-05 17:09:48.357] [info] DEPTHAI_DEBUG enabled, lowered DEPTHAI_LEVEL to 'debug'
[2022-12-05 17:09:48.357] [debug] Library information - version: 2.19.1, commit:  from , build: 2022-12-03 11:23:22 +0000
[2022-12-05 17:09:48.359] [debug] Initialize - finished
Starting Front Slam Sensor
[2022-12-05 17:09:48.426] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.22.tar.xz' open: 1ms, archive read: 66ms
[2022-12-05 17:09:48.708] [debug] Resources - Archive 'depthai-device-fwp-c3891fbe461a569a21fd0dc713d2dced535fbd8d.tar.xz' open: 1ms, archive read: 348ms
[2022-12-05 17:09:49.059] [debug] Device - OpenVINO version: 2021.4
[2022-12-05 17:09:49.059] [debug] Device - BoardConfig: {"camera":[],"emmc":null,"gpio":[],"logDevicePrints":true,"logPath":null,"logSizeMax":null,"logVerbosity":null,"network":{"mtu":0,"xlinkTcpNoDelay":true},"pcieInternalClock":null,"sysctl":[],"uart":[],"usb":{"flashBootedPid":63037,"flashBootedVid":999,"maxSpeed":4,"pid":63035,"vid":999},"usb3PhyInternalClock":null,"watchdogInitialDelayMs":null,"watchdogTimeoutMs":null} 
libnop:
0000: b9 0f b9 05 81 e7 03 81 3b f6 81 e7 03 81 3d f6 04 b9 02 00 01 ba 00 be be bb 00 bb 00 be be be
0020: be be be 01 bb 00
[2022-12-05 17:09:50.510] [debug] Searching for booted device: DeviceInfo(name=3.1, mxid=1944301051BBED1200, X_LINK_BOOTED, X_LINK_USB_VSC, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[1944301051BBED1200] [3.1] [1.462] [system] [warning] PRINT:LeonCss: BootloaderConfig.options1 checksum doesn't match. Is: 0x10304419 should be: 0xFEC083B1
BootloaderConfig.options2 checksum doesn't match. Is: 0x00000000 should be: 0x2380F205
GPIO boot mode 0x3, interface SPI_MASTER_EFF
Setting aons(0..4) back to boot from flash (offset = 0)
====ENABLE WATCHDOG====1
initial keepalive, countdown: 10
PLL0: 700000 AUX_IO0: 24000 AUX_IO1: 24000 MCFG: 24000 MECFG: 24000
Board init ret 3
eeprom configuration version: 55AA0007
Reading VERSION 7 --- ->
eeprom configuration load from user area, status: 0
Reading VERSION 7 --- ->
Didn't find any boards matching the eeprom data
Board options: 00000004
--> brdInit ...
brdInitAuxDevices: Error: SC = 27: io_initialize expander_cam_gpios_1 [OK]

spi_N25Q_init: Flash JEDEC ID: c2 25 3a
SR CR: 40 07
QUAD mode already enabled. Dummy cfg: 0x00
Opening bus for IR driver: 1
Failed to probe IR driver LM3644
Opening bus for IR driver: 2
LM3644 detected, ID = 0x02
===== IR write bus 2: 0x07 = 0x89
===== IR write bus 2: 0x02 = 0x01
===== IR write bus 2: 0x03 = 0x00
===== IR write bus 2: 0x04 = 0x00
===== IR write bus 2: 0x05 = 0x00
===== IR write bus 2: 0x06 = 0x00
===== IR write bus 2: 0x07 = 0x09
===== IR write bus 2: 0x08 = 0x1a
===== IR write bus 2: 0x09 = 0x08
===== IR write bus 2: 0x01 = 0x24
Opening bus for IR driver: 3
Failed to probe IR driver LM3644
Closing EEPROm!
MyriaX board configuration
pll0 frequency: 700000, ref0 frequency: 24000
Is booted from flash by bootloader: 0
Networking not available...
=== Enumerating on socket: Cam_A / RGB / Center
initial keepalive, countdown: 9
=== Enumerating on socket: Cam_B / Left
Probe failed I2C2 0x63 reg 300a: expected 7750, read 0000
Probe failed I2C2 0x63 reg 300a: expected 7750, read 0000
=== Enumerating on socket: Cam_C / Right
Probe failed I2C2 0x63 reg 300a: expected 7750, read 0000
Probe failed I2C2 0x63 reg 300a: expected 7750, read 0000
Initializing XLink...
UsbPumpVscAppI_Event: 5 VSC2_EVENT_ATTACH
UsbPumpVscAppI_Event: 2 VSC2_EVENT_SUSPEND
UsbPumpVscAppI_Event: 3 VSC2_EVENT_RESUME
[... 456 lines cut ...]
UsbPumpVscAppI_Event: 2 VSC2_EVENT_SUSPEND
[1944301051BBED1200] [3.1] [4.763] [system] [warning] PRINT:LeonCss: UsbPumpVscAppI_Event: 3 VSC2_EVENT_RESUME
UsbPumpVscAppI_Event: 2 VSC2_EVENT_SUSPEND
[1944301051BBED1200] [3.1] [4.774] [system] [warning] PRINT:LeonCss: UsbPumpVscAppI_Event: 3 VSC2_EVENT_RESUME
UsbPumpVscAppI_Event: 2 VSC2_EVENT_SUSPEND
[1944301051BBED1200] [3.1] [4.785] [system] [warning] PRINT:LeonCss: UsbPumpVscAppI_Event: 3 VSC2_EVENT_RESUME
[1944301051BBED1200] [3.1] [5.494] [system] [info] Memory Usage - DDR: 104.38 / 340.61 MiB, CMX: 2.43 / 2.50 MiB, LeonOS Heap: 34.84 / 77.46 MiB, LeonRT Heap: 5.45 / 41.34 MiB
[1944301051BBED1200] [3.1] [5.494] [system] [info] Temperatures - Average: 40.74 °C, CSS: 42.12 °C, MSS 40.04 °C, UPA: 40.74 °C, DSS: 40.04 °C
[1944301051BBED1200] [3.1] [5.494] [system] [info] Cpu Usage - LeonOS 34.30%, LeonRT: 0.07%
[1944301051BBED1200] [3.1] [6.496] [system] [info] Memory Usage - DDR: 104.38 / 340.61 MiB, CMX: 2.43 / 2.50 MiB, LeonOS Heap: 34.84 / 77.46 MiB, LeonRT Heap: 5.45 / 41.34 MiB
[1944301051BBED1200] [3.1] [6.496] [system] [info] Temperatures - Average: 40.74 °C, CSS: 41.20 °C, MSS 40.74 °C, UPA: 41.20 °C, DSS: 39.81 °C
[1944301051BBED1200] [3.1] [6.496] [system] [info] Cpu Usage - LeonOS 6.03%, LeonRT: 0.09%
[1944301051BBED1200] [3.1] [7.498] [system] [info] Memory Usage - DDR: 104.38 / 340.61 MiB, CMX: 2.43 / 2.50 MiB, LeonOS Heap: 34.84 / 77.46 MiB, LeonRT Heap: 5.45 / 41.34 MiB
[1944301051BBED1200] [3.1] [7.498] [system] [info] Temperatures - Average: 40.97 °C, CSS: 41.89 °C, MSS 40.74 °C, UPA: 41.20 °C, DSS: 40.04 °C
[1944301051BBED1200] [3.1] [7.498] [system] [info] Cpu Usage - LeonOS 5.92%, LeonRT: 0.07%
[... cut ...]
[1944301051BBED1200] [3.1] [8.514] [system] [warning] PRINT:LeonCss: UsbPumpVscAppI_Event: 3 VSC2_EVENT_RESUME
[1944301051BBED1200] [3.1] [9.502] [system] [info] Memory Usage - DDR: 104.38 / 340.61 MiB, CMX: 2.43 / 2.50 MiB, LeonOS Heap: 34.84 / 77.46 MiB, LeonRT Heap: 5.45 / 41.34 MiB
[1944301051BBED1200] [3.1] [9.502] [system] [info] Temperatures - Average: 41.43 °C, CSS: 41.66 °C, MSS 41.20 °C, UPA: 42.35 °C, DSS: 40.51 °C
[1944301051BBED1200] [3.1] [9.502] [system] [info] Cpu Usage - LeonOS 6.44%, LeonRT: 0.09%

```

PeterQFR, 2022-12-06:
If not the firmware, there's at least something off with my build environment. After a clean build it is working correctly. 

The image manipulator seems to be connected given the pipeline graph below:

[image]

It may be the image manipulator is not working correctly with depth images after recent work? My setup is below
```
  depthRotation_ = pipeline.create<dai::node::ImageManip>();
        colorRotation_ = pipeline.create<dai::node::ImageManip>();

        depthRotation_->initialConfig.setRotationRadians(M_PI_2);
        depthRotation_->initialConfig.setFrameType(dai::ImgFrame::Type::RAW16);
        depthRotation_->inputImage.setBlocking(false);
        depthRotation_->inputImage.setQueueSize(1);

        colorRotation_->initialConfig.setRotationRadians(M_PI_2);
        colorRotation_->initialConfig.setFrameType(dai::ImgFrame::Type::BGR888p);
        colorRotation_->inputImage.setBlocking(false);
        colorRotation_->inputImage.setQueueSize(1);

        rgb_->preview.link(colorRotation_->inputImage);
        depth_temp_->depth.link(depthRotation_->inputImage);

        colorRotation_->out.link(spatialDetector_->input);
        depthRotation_->out.link(spatialDetector_->inputDepth);
```

PeterQFR, 2022-12-06:
calibration dump:
 ```
Is EEPROM available: 1
User calibration: {
    "batchName": "Maxwell",
    "batchTime": 1656224163,
    "boardConf": "IR-C00M01-00",
    "boardCustom": "",
    "boardName": "DM9098",
    "boardOptions": 4,
    "boardRev": "R6M2E6",
    "cameraData": [
        [
            2,
            {
                "cameraType": 0,
                "distortionCoeff": [
                    -10.649245262145996,
                    101.31301879882813,
                    0.0013906945241615176,
                    0.0009143123752437532,
                    -80.78216552734375,
                    -10.699222564697266,
                    100.87681579589844,
                    -79.74972534179688,
                    0.0,
                    0.0,
                    0.0,
                    0.0,
                    0.0,
                    0.0
                ],
                "extrinsics": {
                    "rotationMatrix": [
                        [
                            0.9998529553413391,
                            0.006297831889241934,
                            0.015949560329318047
                        ],
                        [
                            -0.006350639741867781,
                            0.9999744892120361,
                            0.003262458834797144
                        ],
                        [
                            -0.015928607434034348,
                            -0.0033632691483944654,
                            0.9998674988746643
                        ]
                    ],
                    "specTranslation": {
                        "x": 3.75,
                        "y": 0.0,
                        "z": 0.0
                    },
                    "toCameraSocket": 0,
                    "translation": {
                        "x": 3.7241790294647217,
                        "y": 0.00627529202029109,
                        "z": -0.15790285170078278
                    }
                },
                "height": 800,
                "intrinsicMatrix": [
                    [
                        806.6952514648438,
                        0.0,
                        653.4320678710938
                    ],
                    [
[... 519 lines cut ...]
                0.9999045729637146,
                -0.0022102377843111753,
                0.013637839816510677
            ],
            [
                0.002198750153183937,
                0.9999971985816956,
                0.0008572729420848191
            ],
            [
                -0.013639695942401886,
                -0.0008272049599327147,
                0.9999066591262817
            ]
        ],
        "rightCameraSocket": 2
    },
    "version": 7
}

User calibration raw: [7,0,170,85,3,0,0,0,0,68,77,57,48,57,56,0,0,0,0,0,0,0,0,0,0,82,54,77,50,69,54,0,0,0,0,170,249,127,63,146,38,215,187,241,211,72,188,241,207,214,59,145,254,127,63,208,84,103,186,31,235,72,60,141,62,82,58,13,251,127,63,191,249,127,63,163,217,16,187,63,113,95,60,232,24,16,59,209,255,127,63,157,186,96,58,8,121,95,188,201,216,88,186,226,249,127,63,1,2,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,255,32,3,0,5,0,28,75,74,68,0,0,0,0,8,215,35,68,0,0,0,0,28,75,74,68,217,200,203,67,0,0,0,0,0,0,0,0,0,0,128,63,250,108,31,193,138,103,212,66,168,75,76,184,57,35,17,59,30,143,51,194,144,39,32,193,199,30,211,66,193,55,45,194,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,82,184,143,66,0,1,100,233,127,63,7,113,143,187,241,37,212,188,171,15,142,59,75,255,127,63,143,135,220,186,205,52,212,60,212,188,205,58,237,233,127,63,162,94,241,192,197,149,136,60,91,177,210,189,0,0,240,192,0,0,0,0,0,0,0,0,2,32,3,0,5,0,127,172,73,68,0,0,0,0,167,91,35,68,0,0,0,0,127,172,73,68,126,34,195,67,0,0,0,0,0,0,0,0,0,0,128,63,79,99,42,193,68,160,202,66,247,71,182,58,119,174,111,58,120,144,161,194,4,48,43,193,238,192,201,66,220,127,159,194,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,82,184,143,66,0,2,93,246,127,63,11,94,206,59,167,168,130,60,7,25,208,187,84,254,127,63,250,206,85,59,182,124,130,188,75,106,92,187,81,247,127,63,243,88,110,64,247,160,205,59,73,177,33,190,0,0,112,64,0,0,0,0,0,0,0,0,0,112,8,0,15,0,234,142,65,69,0,0,0,0,193,28,239,68,0,0,0,0,234,142,65,69,156,7,144,68,0,0,0,0,0,0,0,0,0,0,128,63,28,239,195,64,229,88,232,194,34,58,145,186,34,8,146,186,232,217,181,67,8,217,185,64,126,149,226,194,132,15,178,67,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,109,150,137,66,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,25 [... cut ...] ,0,0,0,0,0,0,0,0,0,73,82,45,67,48,48,77,48,49,45,48,48,0,0,0,0,70,49,45,70,86,48,48,45,66,67,48,48,49,0,0,0,77,97,120,119,101,108,108,0,0,0,0,0,0,0,0,0,163,249,183,98,0,0,0,0,0,0,0,0,0,0,0,0,4,0,0,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255]

```

themarpe, 2022-12-17:
@PeterQFR 

Thanks for keeping us updated on this issue. I think we've reproduced it on an OAK-D Pro AF.
We'll be testing in the next couple of days

themarpe, 2022-12-17:
@PeterQFR addressed in latest `develop`

Do you mind giving it a try?

PeterQFR, 2022-12-18:
@themarpe, The three cameras are detected, so it seems to have alleviated that issue. I posted the debug output below

```
[2022-12-18 04:21:59.619] [info] DEPTHAI_DEBUG enabled, lowered DEPTHAI_LEVEL to 'debug'
[2022-12-18 04:21:59.619] [debug] Library information - version: 2.19.1, commit:  from , build: 2022-12-18 12:10:00 +0000
[2022-12-18 04:21:59.622] [debug] Initialize - finished
Starting Front Slam Sensor
[2022-12-18 04:21:59.696] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.22.tar.xz' open: 2ms, archive read: 73ms
[2022-12-18 04:21:59.981] [debug] Resources - Archive 'depthai-device-fwp-b6040a0abd70095a5737aa1ad6ab58964e417d8a.tar.xz' open: 1ms, archive read: 359ms
[2022-12-18 04:22:00.337] [debug] Device - OpenVINO version: 2021.4
[2022-12-18 04:22:00.337] [debug] Device - BoardConfig: {"camera":[],"emmc":null,"gpio":[],"logDevicePrints":true,"logPath":null,"logSizeMax":null,"logVerbosity":null,"network":{"mtu":0,"xlinkTcpNoDelay":true},"pcieInternalClock":null,"sysctl":[],"uart":[],"usb":{"flashBootedPid":63037,"flashBootedVid":999,"maxSpeed":4,"pid":63035,"vid":999},"usb3PhyInternalClock":null,"watchdogInitialDelayMs":null,"watchdogTimeoutMs":null} 
libnop:
0000: b9 0f b9 05 81 e7 03 81 3b f6 81 e7 03 81 3d f6 04 b9 02 00 01 ba 00 be be bb 00 bb 00 be be be
0020: be be be 01 bb 00
[2022-12-18 04:22:01.775] [debug] Searching for booted device: DeviceInfo(name=3.1, mxid=1944301051BBED1200, X_LINK_BOOTED, X_LINK_USB_VSC, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[1944301051BBED1200] [3.1] [0.859] [system] [warning] PRINT:LeonCss: BootloaderConfig.options1 checksum doesn't match. Is: 0x10304419 should be: 0xFEC083B1
BootloaderConfig.options2 checksum doesn't match. Is: 0x00000000 should be: 0x2380F205
GPIO boot mode 0x3, interface SPI_MASTER_EFF
Setting aons(0..4) back to boot from flash (offset = 0)
====ENABLE WATCHDOG====1
initial keepalive, countdown: 10
PLL0: 700000 AUX_IO0: 24000 AUX_IO1: 24000 MCFG: 24000 MECFG: 24000
Board init ret 3
eeprom configuration version: 55AA0007
Reading VERSION 7 --- ->
eeprom configuration load from user area, status: 0
Reading VERSION 7 --- ->
PMIC read  reg 0x19, val 0xb8
PMIC read  reg 0x23, val 0x10
PMIC read  reg 0x31, val 0x08
PMIC read  reg 0x33, val 0x28
Found a device/board entry matching the eeprom data - Board: OAK-D S2/Pro (R6 - generic) Device: 
Board options: 00000004
--> brdInit ...
brdInitAuxDevices: Error: SC = 27: io_initialize expander_cam_gpios_1 [OK]

spi_N25Q_init: Flash JEDEC ID: c2 25 3a
SR CR: 40 07
QUAD mode already enabled. Dummy cfg: 0x00
Opening bus for IR driver: 1
Failed to probe IR driver LM3644
Opening bus for IR driver: 2
LM3644 detected, ID = 0x02
===== IR write bus 2: 0x07 = 0x89
===== IR write bus 2: 0x02 = 0x01
===== IR write bus 2: 0x03 = 0x00
===== IR write bus 2: 0x04 = 0x00
===== IR write bus 2: 0x05 = 0x00
===== IR write bus 2: 0x06 = 0x00
===== IR write bus 2: 0x07 = 0x09
===== IR write bus 2: 0x08 = 0x1a
===== IR write bus 2: 0x09 = 0x08
===== IR write bus 2: 0x01 = 0x24
Opening bus for IR driver: 3
Failed to probe IR driver LM3644
Closing EEPROm!
MyriaX board configuration
pll0 frequency: 700000, ref0 frequency: 24000
Is booted from flash by bootloader: 0
Networking not available...
=== Enumerating on socket: Cam_A / RGB / Center
Probe failed I2C1 0x1a reg 0016: expected 0214, read 0378
Probe failed I2C1 0x1a reg 0016: expected 0214, read 0378
  >> Registered camera A12N02A (imx378) as /dev/Camera_0
Adding socket 0: cam 2. Sen name: IMX378
=== Enumerating on socket: Cam_B / Left
  >> Registered camera TG161B (ov9282) as /dev/Camera_1
Adding socket 1: cam 5. Sen name: OV9282
=== Enumerating on socket: Cam_C / Right
  >> Registered camera TG161B (ov9282) as /dev/Camera_2
[... 126 lines cut ...]
Starting Guzzi command handling loop...
[1944301051BBED1200] [3.1] [2.047] [system] [warning] PRINT:LeonCss: inc_camera_process set exposure and gain
[E] app_guzzi_command_callback():173: command->id:7
[E] camera_control_ae_auto():706: Auto ae succeeded camera_id = 1.

[E] app_guzzi_command_callback():236: command "7 1" sent

[E] app_guzzi_command_callback():173: command->id:7
[E] camera_control_ae_auto():706: Auto ae succeeded camera_id = 2.

[E] app_guzzi_command_callback():236: command "7 2" sent

[1944301051BBED1200] [3.1] [2.080] [system] [warning] PRINT:LeonMss: finished sippPrePrepareLinesFreeRT
finished sippPrePrepareLinesFreeRT
[1944301051BBED1200] [3.1] [2.091] [system] [warning] PRINT:LeonMss: finished sippPrePrepareLinesFreeRT
[1944301051BBED1200] [3.1] [2.674] [system] [warning] PRINT:LeonCss: 600.000 mA -> 50
dot: 600.00 mA, flood: 0.00 mA, duty = 0.30
===== IR write bus 2: 0x04 = 0x32
===== IR write bus 2: 0x01 = 0x26

```

The spatial detection is still not providing valid spatial coordinates. Has something changed with setting the correct frame for the depth image. It is set to right camera (I haven't got around to changing it yet) instead of the RGB Frame. Has something changed that has made that an invalid input. The pipeline seems to be correct in that depth info should be being piped to the spatialObjectDetector, provided the image manipulation node is working correctly for depth.

The raw original depth is outputting correctly.

themarpe, 2022-12-18:
@PeterQFR thanks for confirming that the one of the issues is resolved. 

WRT Spatial detector, it could be due to the origin of frames. Do you mind opening a new issue and pasting a complete MRE such that we can debug quickly?

closed 2022-12-18 by themarpe

PeterQFR, 2022-12-20:
@themarpe I realigned to rgb socket and it does more readily produce spatial information, with some occasional zero values, so it is possibly a recent change in the spatial detector that fails to produce spatial coordinates if the alignment warning is triggered. I did confirm there was a depth image going to the spatial detector. 

I'm not sure if this is the desired behaviour, as the warning of inaccuracy but still producing spatial coords is probably better.

themarpe, 2022-12-20:
What warnings were printed @PeterQFR ?
Do you mind pasting the whole output? It might be, we've slightly refactored it, but mostly to accomodate pre Stereo image modifications. Which version does differ (eg which two versions were you comparing against?)