luxonis/depthai-core #652: [BUG] {IMU(0) - IMU invalid settings!: ROTATION_VECTOR output is unsupported}

thuongnct, 2022-12-01:
Hi everyone,

I installed the examples of depthai-core for OAKD pro W camera. I tried to run the example of IMU:

`./imu_gyroscope_accelerometer`

and 

`./imu_rotation_vector`

It worked with gyroscope but not with imu_rotation_vector. The error is:

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  IMU(0) - IMU invalid settings!: ROTATION_VECTOR output is unsupported. BMI270 supports only ACCELEROMETER_RAW and/or GYROSCOPE_RAW outputs.
Aborted (core dumped)

```

Does anyone meet this error? Please help me with some idea to get IMU data.

Thank you very much!

themarpe, 2022-12-01:
Hi @thuongnct 

As stated, the BMI270 IMU unit inside the device doesn't support the precomputed ROTATION_VECTOR output. We'll have to tweak the examples to automatically detect the type and act accordingly.

If you modify it to only query ACCELEROMETER_RAW & GYROSCOPE_RAW it should work as expected.

closed 2022-12-01 by themarpe

thuongnct, 2022-12-02:
Thank you very much @themarpe . It worked as your comment.