luxonis/depthai-core #654: [BUG] {The program can be executed, but there is no export message and no error}

friend045, 2022-12-05:
**Describe the bug**

OS : Windows 10 
SDK : 2.19.0

The program can be executed.
We run both models on OAK-D with fps set to 30 and finally output x,y,z to the host.
The strange thing is that x, y, z are not output, and no error message appears.
But the python version works fine.

**Minimal Reproducible Example**
**Pipeline Graph**

Please google mail if you need more information.
I will open cloud space. 
Thanks.

Erol444, 2022-12-08:
Hi @friend045 ,
Could you please provide [full MRE](https://docs.luxonis.com/en/latest/pages/support/)? As there are hardly any useful information in this issue.
Thanks, Erik

friend045, 2022-12-12:
> Hi , Could you please provide [full MRE](https://docs.luxonis.com/en/latest/pages/support/)? As there are hardly any useful information in this issue. Thanks, Erik

Hi @Erol444 ,

Provide full MRE & Pipeline Graph [here](https://drive.google.com/drive/folders/10BgPe-LZUjick2NhnaVadgn3QwHeGFbG?usp=share_link).

Thanks

friend045, 2023-01-30:
Any ideas? Maybe I should try the new version 2.20.1.

friend045, 2023-02-06:
> We'll be looking into this in more depth this Saturday. Do you mind granting the access to the MRE?

Hi @themarpe ,
I opened the access last week.