luxonis/depthai-core #655: [BUG] {Color tuning problems with NFOV OV9782}

stephansturges, 2022-12-07:
**Describe the bug**
I have issues with the NFOV OV9782 RGB sensors giving a VERY STRONG magenta tint in images since summer 2022
[image]

This is a sensor from a series that I ordered on the shop with a "customisation coupon", switching out the main RGB sensor for an NFOV OV9782 in order to have a global shutter on the RGB sensor. I've since received 4 replacement sensors from Luxonis and ALL display this strong magenta tint, rendering the RGB image useless for my task of semantic segmentation.

[image]

I have cameras that I purchased previously (12 of them) that have OV9782 sensors, both in NFOV and Wide variants, and all of these are OK. I've tried switching the sensors on the PCBs and the problem is DEFINITELY with the sensors, not the motherboard / main system.

[image]
(yes I removed the stickers afterwards ;) )

On suggestions from @Erol444 I tested all of the camera tuning blobs from: https://docs.luxonis.com/projects/api/en/latest/tutorials/image_quality/?highlight=ov9782#camera-tuning

All of these produce strong magenta-tint results. 

[image]

**Minimal Reproducible Example**
Using any application that outputs the RGB stream with an NFOV OV9782 will work.

**Expected behavior**
Balanced RGB output 😉 like using this wide-fov OV9782 that I bought early 2022:
[image]

**Screenshots**
See above

**Pipeline Graph**
This occurs with all pipelines that use the RGB sensor with NFOV OV9782 since summer 2022.

**Attach system log**
Not relevant.

**Additional context**
This is a blocking problem for developing part of the tech I am selling and developing with https://github.com/stephansturges/openlander , so I'd *really* appreciate any help here!

stephansturges, 2022-12-08:
Yep, here you go!
[image]

Erol444, 2022-12-08:
Hi @stephansturges ,
We have confirmed tinting effect on OAK-D-SR as well, so we will be fixing this issue before it will get released (hopefully soon). Note that such tuning usually takes a few weeks.
Thank again, Erik

stephansturges, 2022-12-08:
Does this mean that the fix will be tuning blob that can loaded into the pipeline or will I need to replace the actual sensors?
thanks!

alex-luxonis, 2022-12-08:
Thanks @stephansturges .
We narrowed down the issue, it's a camera module problem, apparently a batch of NFOV modules was produced with missing IR filter. The excessive IR light is causing the pink/magenta tint.
As a quick workaround, in case you have external IR filters, placing one over the camera lens should help with this issue.
We should send you replacement modules, but may take some time.