luxonis/depthai-core #664: Linker Errors when depthai is built with a different build configuration from consuming project.

camblomquist, 2022-12-13:
I'm almost certain the fault here is mine but I'm here in case it isn't and I think this is a bit outside of the support level for the Discord.

I'm building depthai as part of a superbuild that works with multiple third-party dependencies to be consumed by other projects.
The relevant snippet is as follows:
```cmake
set(CMAKE_BUILD_TYPE "Release")
set(CMAKE_MSVC_RUNTIME_LIBRARY "MultiThreadedDLL")

ExternalProject_Add(depthai
    SOURCE_DIR ${CMAKE_CURRENT_SOURCE_DIR}/depthai-core
    PREFIX ${_prefix}
    DEPENDS
        ${DEPTHAI_DEPS}
    CMAKE_ARGS
        -DCMAKE_BUILD_TYPE=${CMAKE_BUILD_TYPE}
        -DCMAKE_INSTALL_PREFIX=<INSTALL_DIR>
        -DCMAKE_MSVC_RUNTIME_LIBRARY=${CMAKE_MSVC_RUNTIME_LIBRARY}
        # Only hint OpenCV's path when building OpenCV from source
        $<$<NOT:$<BOOL:${USE_SYSTEM_OPENCV}>>:-DOpenCV_DIR=<INSTALL_DIR>>
        # Hint to OpenCVConfig to search for static libs when building OpenCV from sourec
        -DOpenCV_STATIC=$<NOT:$<BOOL:${USE_SYSTEM_OPENCV}>>
        $<$<BOOL:${CMAKE_TOOLCHAIN_FILE}>:-DCMAKE_TOOLCHAIN_FILE=${CMAKE_TOOLCHAIN_FILE}>  
    BUILD_COMMAND ${CMAKE_COMMAND} --build <BINARY_DIR> --config ${CMAKE_BUILD_TYPE}
    INSTALL_COMMAND ${CMAKE_COMMAND} --install <BINARY_DIR> --config ${CMAKE_BUILD_TYPE}
)
```
And in my main project, which does *not* set the build type and likely defaults to Debug:
```cmake
set(CMAKE_MSVC_RUNTIME_LIBRARY "MultiThreadedDLL")
find_package(depthai REQUIRED CONFIG)
# ...
target_link_libraries(oak PRIVATE depthai::opencv)
```
On Linux, this compiles and links just "fine"
And on Windows, this compiles fine but I get errors come link time. Namely, there seems to be a mismatch between the configuration of the project and depthai's dependencies.
```
XLinkd.lib(ProtocolManager.obj) : error LNK2038: mismatch detected for '_ITERAT
OR_DEBUG_LEVEL': value '2' doesn't match value '0' in nodefactory.obj [Z:\softw
are\oak\Windows\oak.vcxproj]
XLinkd.lib(ProtocolManager.obj) : error LNK2038: mismatch detected for 'Runtime
Library': value 'MDd_DynamicDebug' doesn't match value 'MD_DynamicRelease' in n
odefactory.obj [Z:\software\oak\Windows\oak.vcxproj]
```
NodeFactory.obj just happens to be the first object i tries to link in my code. I get this same errors from `XLinkd.lib(usb_host.obj)`, `XLinkd.lib(tcpip_host.obj)`, `XLinkd.lib(PlatformDeviceFd.obj)`, `XLinkd.lib(usb_mx_id.obj)`, `XLinkd.lib(win_usb_host.obj)`, `spdlogd.lib(fmt.obj)`, `spdlogd.lib(spdlog.obj)`, `spdlogd.lib(color_sinks.obj)`, `spdlogd.lib(cfg.obj)`, `spdlogd.lib(async.obj)`.

It appears that when trying to link depthai, it tries to find its dependencies based on the consuming project's build type/configuration rather than what depthai was built to use.

**Expected behavior**
Honestly not too sure. I don't know if this is CMake business as usual or not. I personally expect that if I build depthai in a given configuration, that it would try to link with its release dependencies in that same configuration. But if that's not how CMake is supposed to work, then I suppose I just have to work around it.

themarpe, 2022-12-13:
@camblomquist on a quick glance, its MSVCs limitation that it can only link together libraries of same Debug level. In your case I think you might be bulding a Release but it tries linking with Debug variants of libraries.

themarpe, 2022-12-13:
As MSVC/VS proejct is a multigenerator based backend, you need to add a --config Release/Debug when building with eg cmake --build to select a specific build type. Unlike with single generators where that is selected via CMAKE_BUILD_TYPE variable

camblomquist, 2022-12-14:
>  In your case I think you might be bulding a Release but it tries linking with Debug variants of libraries.

Almost the opposite problem, actually. I'm building depthai in a release configuration and my project in a debug configuration. I'm explicitly passing the MultiThreadedDLL runtime to depthai instead of the Debug runtime. The issue is *only* with the configuration of depthai's dependencies. I'm attempting to link with the release build of depthai, but depthai is somehow finding the debug builds of its dependencies which then fails to link.

> you need to add a --config Release/Debug when building with eg cmake --build to select a specific build type.

My understanding is that by default, the config will be Debug unless the project specifies otherwise. If I explicitly pass `--config Debug` to my project build, the result is the same. Passing `--config Release` makes this problem disappear. But I do kind of want to be able to debug my project. Maybe this is a misunderstanding of how configuration works but I don't particularly care about debugging depthai unless I think depthai is the cause of the problem, so I was hoping to be able to link the release build of depthai instead of the debug configuration.

themarpe, 2022-12-14:
> Maybe this is a misunderstanding of how configuration works but I don't particularly care about debugging depthai unless I think depthai is the cause of the problem, so I was hoping to be able to link the release build of depthai instead of the debug configuration.

This is exactly the case. You can NOT link against different debug levels. All libraries have to be the same. (For MSVC).
Can cross check with OpenCV and their prebuilt library - it also carries *d.lib/dll files along for such reasons

camblomquist, 2022-12-14:
> You can NOT link against different debug levels

For what it's worth, xlink and spdlog are the only libraries that have had this problem out of the dozen depthai uses and the dozen more that I'm using. Ignoring the MSVC runtime for a moment, if I don't need to step into a library while debugging then it should be perfectly valid to link against the release target.

The issue here appears not to be with depthai directly but with xlink and spdlog, which set the `IMPORTED_CONFIGURATIONS` property. By doing this, they're changing how they're imported depending on my project's configuration. And since I don't have control over they were built, they end up choosing a configuration that is incompatible with how depthai was built (see the first error about '_ITERATOR_DEBUG_LEVEL',) and my own project (see the second error.)

Now if you don't want to support the library being used in this way, then that's perfectly fine and this issue can be closed. Otherwise, I think the solution would be for depthaiDependencies.cmake to set the target property `MAP_IMPORTED_CONFIG_<CONFIG>` on xlink and spdlog to match how depthai was built (i.e mapping Debug to Release if depthai was built for release and vice versa.)

camblomquist, 2022-12-14:
Actually based on a comment from [this CMake issue](https://gitlab.kitware.com/cmake/cmake/-/issues/20440), it's my responsibility to do the mapping rather than depthai's.
>  There is no way a SDK's package configuration file can know what set of configurations the consuming application will choose.  It doesn't even have to be CMake's standard set.  It is up to the consumer to map their set of configurations to those offered by the SDK.

This issue can be closed then.

closed 2022-12-14 by camblomquist