luxonis/depthai-core #665: [BUG] Set brightness to value different from 0 make the video brightness oscillate and not stable

tarekmuallim, 2022-12-19:
Hello,
I am using OAK-1 Light.
When I change the brightness to any value than 0 the brightness starts to oscillate (video link is attached).
https://drive.google.com/file/d/1R2W6qgnjIOF80MZxxsMOeK1RpbXcbVXX/view?usp=share_link

I try to change the brightness using python. It works without any problem.
The problem appears only when I use DepthAi Core and C++.
My code is something like this:
```
dai::CameraControl ctrl;
ctrl.setBrightness(value);
controlQueue->send(ctrl);
```

Erol444, 2022-12-23:
Hi @tarekmuallim , Could you please prepare [a MRE](https://docs.luxonis.com/en/latest/pages/support/#depthai-issue-software-firmware)?
Thanks, Erik

tarekmuallim, 2022-12-26:
Hi @Erol444 
It will take some time to prepare a MRE.
I have tested changing the brightness in simple program and it worked without problem.
I will recheck my program and go back.
I am closing this for now.

closed 2022-12-26 by tarekmuallim